john30/ebusd #289: unable to open /dev/ttyUSB0: ERR: generic device error

jerome686, 2019-08-04:
Hi
when I use   ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0,

2019-08-05 01:15:58.708 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-08-05 01:15:58.709 [bus notice] bus started with own address 31/36
2019-08-05 01:15:58.709 [bus notice] device invalid
2019-08-05 01:16:03.709 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2019-08-05 01:16:03.710 [bus notice] device invalid
...................................

I used a esera usb ebus coupler,    

when I  type    ebusctl  info

version: ebusd 3.3.v3.3-24-g958a9a8
update check: revision v3.3-4-g212b22d available
signal: acquired
symbol rate: 23
max symbol rate: 40
min arbitration micros: 718
max arbitration micros: 718
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 2
messages: 5
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

what's wrong ?

john30, 2019-08-11:
is the device available at all?
check by doing a file listing like this: `ls -la /dev/ttyUSB0`
and does this reveal read+write rights on the device?

samm-git, 2019-08-18:
For me it looks like you already have one instance running and trying to run another one. This causing port access conflict.

jerome686, 2019-08-24:
thank you very much,   a ebusd  instance  was running,   I killed  it,   and  everything is ok

closed 2019-08-24 by jerome686