john30/ebusd #291: ebusd config on Pi Zero Wifi with serial HW interface

dirkdecoppel, 2019-08-11:
hello,
I am trying to install the ebusd on a Pi zero with wifi with the HW controller connected to the serial interface.
In the config file i modified as following:

EBUSD_OPTS="--scanconfig --latency=20000 -d /dev/ttyAMA0"
Is this correct?

it seems to work but not completely. in the log i find following:
2019-08-11 14:27:38.275 [main notice] ebusd 3.3.v3.3 started with auto scan
2019-08-11 14:27:38.604 [bus notice] bus started with own address 31/36
2019-08-11 14:27:38.627 [bus notice] signal acquired
2019-08-11 14:27:55.225 [main error] scan: empty
2019-08-11 14:28:07.064 [main error] scan: empty
2019-08-11 14:29:42.153 [main notice] update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available
2019-08-11 14:32:41.442 [bus notice] new master 03, master count 2
2019-08-11 14:32:42.387 [bus notice] scan 08: ;Vaillant;BAI00;0609;3302
2019-08-11 14:32:42.387 [update notice] store 08 ident: done
2019-08-11 14:32:42.388 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0609;3302
2019-08-11 14:32:42.388 [bus notice] scan 08: ;Vaillant;BAI00;0609;3302
2019-08-11 14:32:42.736 [main notice] read common config file vaillant/scan.csv
2019-08-11 14:32:42.819 [main notice] read common config file vaillant/general.csv
2019-08-11 14:32:42.906 [main notice] read common config file vaillant/broadcast.csv
2019-08-11 14:32:42.994 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0609, HW3302
2019-08-11 14:32:43.208 [update notice] sent scan-read scan.08 id QQ=31: 
2019-08-11 14:32:43.467 [update notice] sent scan-read scan.08 id QQ=31: 
2019-08-11 14:32:43.727 [update notice] sent scan-read scan.08 id QQ=31: 
2019-08-11 14:32:43.990 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-08-11 14:32:44.527 [main notice] found messages: 210 (3 conditional on 25 conditions, 0 poll, 9 update)
2019-08-11 14:32:44.769 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-08-11 14:32:45.028 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-08-11 14:32:45.288 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-08-11 14:32:45.547 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-08-11 14:32:45.547 [bus notice] scan 08: ;21;19;06;0010021855;1610;005060;N9
2019-08-11 14:34:47.632 [main notice] update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available

the heating connected to is a new Bulex thema condens 30/35-B(E-BE) R2

how to proceed?
the /etc/ebusd/ folder stays empty?
is the room below normal?
when i try to connect to http://ebusd.eu/config/ i get HTTP ERROR 404
when i try try following command i get:
pi@raspberrypi:~ $ ebusd --checkconfig
2019-08-11 15:17:43.283 [main error] invalid configpath without scanconfig
pi@raspberrypi:~ $ ebusd --scanconfig --checkconfig
2019-08-11 15:18:25.280 [main notice] ebusd 3.3.v3.3 performing configuration check...
2019-08-11 15:18:25.596 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2019-08-11 15:18:25.598 [main notice] ebusd stopped
pi@raspberrypi:~ $ 

many thanks in advance for your help
Dirk

john30, 2019-08-15:
looks fine. you can find the available messages by using the `find` command, see the [wiki](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#find).
if you're using the webservice at ebusd.eu, /etc/ebusd does not have to contain any files and ebusd will not write in there anyway.
what do you mean with "room below"?
only ebusd is allowed to connect the webservice at ebusd.eu.

dirkdecoppel, 2019-08-22:
Hi John,
Thanks for your reaction.
The word “room” was an typing error.
You must remplace it with “reaction”.
In the meantime after reading the doos i understand your program ebusd & ebusctl a bit more.
At this moment i have only my RPI connected on the ebus.
My thermostat is connected on the 24v switch connection.
I am wondering if i could made my own thermostat in Python & maybe node-Red that communicaties directly on the ebus.
I think this uit be possible but i am misging at this moment the protocol commands.
Do you have any logging & explications of the commands for the communication between the Bulex & thermostat?
I am most interested in those with the Bulex MiGo or Vaillant Vsmart thermostat.

Many thanks in advance for you advice & help.

Dirk
(Belgium)

john30, 2019-09-17:
you can use the raw logging for looking at the low level communication. for nde-red you could integrate that via MQTT feature of ebusd