john30/ebusd #294: hcpumpmode 3

bubu66, 2019-09-11:
Hi and first sorry about my bad english ;)

I have a Vaillant ecotec and ebus is working very well on my raspi.
Thank you for this Wonderful Software!!

 My ecotec can provide the hcpumpmode from 0 to 3 (post run, permanent, winter, intermittend)

I can read These numbers over ebus with the read commad, but i can only write from 0 to 2 with the write ebus command. If i try this, the Error: "ERR:element not found" appears.

The number 3 (intermittend) can´t be written with the write command, i have to do it manually at the ecotec, than ist working.

Are there any hints? I want to write the Number 3 over the ebus write command!

Thank you very much.

Dieter

dwarning, 2019-09-16:
Hello Dieter,
I have no solution - but behaviour of my 3 years old ecoCompact is same.

Can you change PartloadHCKW?
pi@raspberrypi:~ $ ebusctl r -c bai -f PartloadHCKW
8
pi@raspberrypi:~ $ ebusctl w -c bai PartloadHCKW 10
done
pi@raspberrypi:~ $ ebusctl r -c bai -f PartloadHCKW
8
Means - value is unchanged.

john30, 2019-09-17:
you'd have to extend the CSV config file accordingly. what does the value 3 stand for?

andig, 2019-09-17:
> The number 3 (intermittend) can´t be written with the write command,

I have an Ecotec, too. Could you elaborate what `intermittend` mode really means? You could also check if this is perhaps part of the `SetMode` command.

> Can you change PartloadHCKW?

Please don't hijack issues with off-topic questions. As for `PartloadHCKW`, you cannot write it. Partload is determined by the BAI and would mostly depend on current vs. desired flowtemp (the latter an assumption). To set the temperatures you could again use `SetMode`.

bubu66, 2019-09-17:
> you'd have to extend the CSV config file accordingly. what does the value 3 stand for?

I have attched the 3=pulse. Then i reload the configuration files. but it doesn´t work.

The Value 3=pulse means, that the pumpe is periodicly turns on/off after the burner locking time.

Thanks for the help
Dieter

bubu66, 2019-09-17:
> Could you elaborate what intermittend mode really means?

Intermittend mode (I call it pulse mode) means:

After the burner locking time the pump turns off for 10-20 Minutes. then it turns on for 1-3 Minutes. If the flowtemp isn't sinking enough, the pump turns off for again 10-20 Minutes and then turns on again for 1-3 Minutes. And so on...

This mode is the most efficient in my opinion.

Thanks
Dieter

john30, 2019-10-05:
do you read the configuration files from your local folder where you've changed the csv accordingly @bubu66  ?

bubu66, 2019-10-07:
After changing the local Files and reloading the csv it works perfect!
That was the solution, thank you very much :)

Thanks
Dieter

closed 2019-10-08 by john30