john30/ebusd #295: [main notice] update check: revision

dwarning, 2019-09-15:
Hi John,
made a git update, rebuild executable and install it. Anything is working:
2019-09-15 14:43:24.157 [main notice] ebusd 3.3.v3.3-32-g3926e9c started with auto scan
2019-09-15 14:43:24.423 [bus notice] bus started with own address 31/36

But this message in log file is confusing:
2019-09-15 14:45:47.150 [main notice] update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available, vaillant/15.700.csv: different version available, vaillant/bai.0010015600.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available

It appears independent from ebusd "--configpath" setting. Let it empty I think it points to your ebusd.eu server. Also by linking /etc/ebusd to an up-to-date git folder on my raspi this message appears.

What is wrong in my Installation?
Thanks,
Dietmar

john30, 2019-09-17:
there is nothing wrong with your installation. it is probably an issue with calculating the hash of the config files. what is your system exactly?

dwarning, 2019-09-17:
Hello John,
my system:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Rev 1.2

my heater:
pi@raspberrypi:~ $ ebusctl info
version: ebusd 3.3.v3.3-32-g3926e9c
update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available, vaillant/15.700.csv: different version available, vaillant/bai.0010015600.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available
access: *
signal: acquired
symbol rate: 23
max symbol rate: 146
min arbitration micros: 706
max arbitration micros: 1071
min symbol latency: 5
max symbol latency: 11
reconnects: 0
masters: 3
messages: 599
conditional: 2
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0116;HW=9602", loaded "vaillant/bai.0010015600.inc" ([PROD='0010015600']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0206;HW=4103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

Do you need other informations?

Thank you,
Dietmar

john30, 2019-10-05:
everything is fine with your system. actually this is a duplicate of #266 and was corrected already. But the webservice needs an update of the hashes, which was postponed until 3.4 is ready.

closed 2019-10-05 by john30

john30, 2019-10-06:
it should be better now due to 20a8f4f7f650910ef7c493d40235b7e884c202d0 being deployed on the update webservice

andrewt27, 2023-09-28:
Hi I have the same issue:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --httpport=8889 --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=yi3ooxahgaiheiPhee2daew9zieGhaetaeS9lol0aikaisiag8aicue5iwuy2she --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en --latency=10 --accesslevel=* --mqttint=/config/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.168.1.82:9999 --log=all:notice --log=main:notice --log=bus:notice --log=update:notice --log=network:notice --log=other:notice
2023-09-28 10:30:00.208 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.82:9999, enhanced
2023-09-28 10:30:00.242 [bus notice] device status: resetting
2023-09-28 10:30:00.243 [bus notice] bus started with own address 31/36
2023-09-28 10:30:00.808 [mqtt notice] connection established
2023-09-28 10:32:10.501 [main notice] **update check: revision 23.2 available, broadcast.csv: different version available, memory.csv: different version available**

Home assistant:
eBUSd Current version: 23.2.3