john30/ebusd #296: ebusd is not able to connect to Vaillant VRC 700

CrazYoshi, 2019-09-19:
Hi,
I have update ebusd to the latest version and currently I'm not able to make it works as before using 2.x version.
What I'm getting from ebusctl info command is the following:
```
pi@homeassistant:~$ ebusctl info
version: ebusd 3.3.v3.3-32-g3926e9c
signal: acquired
symbol rate: 41
max symbol rate: 90
reconnects: 0
masters: 3
messages: 13
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
My current configuration in /etc/default/ebusd:
`EBUSD_OPTS="-d 192.168.1.23:8889 -l /var/log/ebusd.log --loglevel=error --latency=50000 --scanconfig --mqtthost=localhost --mqttport=1883 --mqttjson --mqttretain --mqttuser=serviceuser --mqttpass=Gr1sul0n386 --mqttversion=3.1.1"`

I'm using eBUS Adapter 2.0 Build: 20190407
From the adapter web page it is marked as correctly connected to eBus and ebusd:
```
ebusd connected: yes (inactive)
eBUS signal: acquired
```

Anyone can give me an hint on this issue?
Thanks

john30, 2019-10-05:
is youe WiFi signal ok? This was seen before when the signal strength is not good enough and Wemos disconnects from the router

CrazYoshi, 2019-10-05:
Hi,
yes, I can see this on eBUS adapter web page:
WiFi signal: 100% (-30dBm)

The access point is really close, Wemos is about 1 meter from that.

john30, 2019-10-07:
it seems the communication is too slow. do you have any media streaming active in your wlan or a huge download?

CrazYoshi, 2019-10-14:
I don't usually have huge network traffic but I'm currently use a powerline ethernet connector to connect WeMos to my home net.
I can do some test and see if something change. I will give you some feedback on this topic

john30, 2019-11-01:
hm, then probably the powerline adds some more latency to it. can you try without?

CrazYoshi, 2019-11-02:
Hi,
I have connected eBus using USB to Serial adapter and an RaspberryPi B+ but I'm still getting the same error:
This is my config:
`EBUSD_OPTS="-d /dev/ttyUSB0 -l /var/log/ebusd.log --loglevel=error --latency=50000 --scanconfig --mqtthost=192.168.1.6 --mqttport=1883 --mqttjson --mqttretain --mqttuser=USER --mqttpass=XXXXXX --mqttversion=3.1.1"
`
This is my ebusctl output:
`pi@ebuspi:~ $ ebusctl info
version: ebusd 3.4.v3.4
signal: acquired
symbol rate: 23
max symbol rate: 95
reconnects: 0
masters: 3
messages: 418
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0110;HW=2103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd`
And this are the logs:
`2019-11-02 13:15:32.296 [mqtt error] communication error: errno 32=Broken pipe
2019-11-02 13:15:32.297 [mqtt error] publish: The client is not currently connected.
2019-11-02 13:15:33.297 [mqtt error] communication error: not connected
2019-11-02 13:15:42.581 [main error] scan config 08: ERR: arbitration lost
2019-11-02 13:15:49.298 [mqtt error] publish: The client is not currently connected.
2019-11-02 13:16:02.325 [main error] scan config 08: ERR: arbitration lost
2019-11-02 13:16:23.150 [main error] scan config 08: ERR: arbitration lost
2019-11-02 13:16:52.381 [bus error] send to 15: ERR: arbitration lost, retry
2019-11-02 13:17:12.403 [main error] scan config 08: ERR: arbitration lost
2019-11-02 13:17:22.416 [bus error] send to 15: ERR: arbitration lost, retry
2019-11-02 13:17:23.475 [bus error] send to 15: ERR: arbitration lost, retry
2019-11-02 13:17:24.488 [bus error] send to 15: ERR: arbitration lost
2019-11-02 13:17:24.488 [bus error] send message part 0: ERR: arbitration lost
2019-11-02 13:17:24.489 [mqtt error] read 700 z1OpMode: ERR: arbitration lost
2019-11-02 13:17:52.730 [main error] scan config 08: ERR: arbitration lost
2019-11-02 13:17:53.003 [bus error] send to 15: ERR: arbitration lost, retry
2019-11-02 13:18:22.527 [bus error] send to 15: ERR: arbitration lost, retry
2019-11-02 13:18:23.846 [main error] scan config 08: ERR: arbitration lost
2019-11-02 13:18:24.365 [bus error] send to 15: ERR: arbitration lost, retry
2019-11-02 13:18:32.797 [bus error] send to 15: ERR: arbitration lost
2019-11-02 13:18:32.797 [bus error] send message part 0: ERR: arbitration lost
2019-11-02 13:18:32.798 [mqtt error] read 700 WaterPressure: ERR: arbitration lost
`
Due to the fact that now is connected directly I don't expect to have any issue on bus but only MQTT if the connection through powerline is slow/broken.
Do you have any suggestion?

john30, 2019-11-05:
the arbitration loss is still strange. are you sure that your adapter is able to send anything at all?
yes, the MQTT connection seems to be broken quite often. can you ping the MQTT broker reliably?

CrazYoshi, 2019-11-06:
How can I check if the adapter works properly? In the past it works correctly with previous versions of ebusd.
I have a sonoff device connected to the same access point of eBus adapter working on MQTT and it works fine.

john30, 2019-11-09:
the latency is an issue for correct ebus communication. I'm pretty sure you had the same problem with the previous ebusd version. would be good to know if the problem is repoducable with that version. can you do that?

CrazYoshi, 2019-11-09:
I will try to find the version that I was using last year.
What it is still not clear to me is why even I connect ebusd via UART I receive arbitration lost errors.
How can I check if something is broken on the adapter itself?

CrazYoshi, 2019-11-09:
I have removed everything and rebuild all from scratch getting the code from Github and now is seems to work fine
Thanks again for your support

closed 2019-11-09 by CrazYoshi

gtclub123, 2023-01-14:
How did you fixed it?

CrazYoshi, 2023-01-15:
It was long time ago but, as I mentioned in my latest comment, I have removed ebusd cleaning every configuration file and build it again from scratch from github.

gtclub123, 2023-01-16:
You just installed new?

nobbilie, 2024-02-07:
Hallo, ich habe das Problem, dass ich den ebusd an der VR 70 Einheit (Ecotec Plus mit Multimatic 700 BJ 2016) angeschlossen habe via USB aber kein Signal vom eBus erhalte. @john30 gibt es da irgend eine Konfiguration oder habe ich schlichtweg einen Denkfehler beim Anschluss und der V5 muss woanders angeschlossen werden als an Position 3?
[image]