john30/ebusd #298: Python -> ebusctl write -c [circuit] [command] [value] -> ERR: invalid numeric argument

danielkarg, 2019-10-09:
I have a problem with "ebusctl write" when calling the command from a python script:

**Python Script:**
eCircuit = "700"
eCommand = "OpMode"
eValue = "day"
os.popen("ebusctl write -c " + eCircuit + " " +  eCommand + " " +  eValue)

**It works if I run the script directly in the console:**

2019-10-09 20:57:57.284 [main debug] >>> write -c 700 OpMode off
2019-10-09 20:57:57.284 [bus info] send message: 3115b52408020100007b000000
2019-10-09 20:57:57.284 [network debug] [00008] wait for result
2019-10-09 20:57:57.339 [bus debug] arbitration delay 8046 micros
2019-10-09 20:57:57.339 [bus debug] switching from ready to send command
2019-10-09 20:57:57.359 [bus debug] send/receive symbol latency 10 ms
2019-10-09 20:57:57.379 [bus debug] send/receive symbol latency 9 ms
2019-10-09 20:57:57.398 [bus debug] send/receive symbol latency 12 ms
2019-10-09 20:57:57.418 [bus debug] send/receive symbol latency 12 ms
2019-10-09 20:57:57.434 [bus debug] send/receive symbol latency 8 ms
2019-10-09 20:57:57.450 [bus debug] send/receive symbol latency 9 ms
2019-10-09 20:57:57.473 [bus debug] send/receive symbol latency 15 ms
2019-10-09 20:57:57.473 [bus info] send/receive symbol latency 5 - 15 ms
2019-10-09 20:57:57.492 [bus debug] send/receive symbol latency 10 ms
2019-10-09 20:57:57.509 [bus debug] send/receive symbol latency 9 ms
2019-10-09 20:57:57.525 [bus debug] send/receive symbol latency 8 ms
2019-10-09 20:57:57.538 [bus debug] send/receive symbol latency 8 ms
2019-10-09 20:57:57.554 [bus debug] send/receive symbol latency 8 ms
2019-10-09 20:57:57.554 [bus debug] switching from send command to send command CRC
2019-10-09 20:57:57.570 [bus debug] send/receive symbol latency 8 ms
2019-10-09 20:57:57.570 [bus debug] switching from send command CRC to receive command ACK
2019-10-09 20:57:57.578 [bus debug] switching from receive command ACK to receive response
2019-10-09 20:57:57.639 [bus debug] switching from receive response to receive response CRC
2019-10-09 20:57:57.650 [bus debug] switching from receive response CRC to send response ACK
2019-10-09 20:57:57.664 [bus debug] send/receive symbol latency 9 ms
2019-10-09 20:57:57.664 [update info] sent MS cmd: 3115b52408020100007b000000 / 020000
2019-10-09 20:57:57.665 [update notice] sent write 700 OpMode QQ=31: off
2019-10-09 20:57:57.665 [bus debug] notify request: done

**But it doesn't work when I start the script in a crontab:**

2019-10-09 20:53:45.266 [main debug] >>> write -c 700 OpMode off

2019-10-09 20:53:45.266 [bus error] prepare message part 0: ERR: invalid numeric argument
2019-10-09 20:53:45.266 [main error] write 700 OpMode: ERR: invalid numeric argument
2019-10-09 20:53:45.266 [main debug] <<< ERR: invalid numeric argument
2019-10-09 20:53:45.266 [network debug] [00009] wait for result
2019-10-09 20:53:45.266 [main debug] >>> 

2019-10-09 20:53:45.266 [main debug] <<< ERR: command not found
2019-10-09 20:53:45.266 [network debug] [00009] wait for result
2019-10-09 20:53:45.266 [main debug] >>> 

Do you have a idea why it doesn't work?

Environment:
Linux version 4.9.0-9-amd64 (debian-kernel@lists.debian.org) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) ) #1 SMP Debian 4.9.168-1+deb9u3 (2019-06-16)

ebusd: 
ebusd-3.3_amd64-jessie.deb

andig, 2019-10-09:
Your cron log has an appended empty line- where would that come from?

danielkarg, 2019-10-10:
**There are many empty lines (each new line has an linefeed:**

2019-10-09 20:53:45.266 [main debug] >>> write -c 700 OpMode off

2019-10-09 20:53:45.266 [bus error] prepare message part 0: ERR: invalid numeric argument
2019-10-09 20:53:45.266 [main error] write 700 OpMode: ERR: invalid numeric argument
2019-10-09 20:53:45.266 [main debug] <<< ERR: invalid numeric argument
2019-10-09 20:53:45.266 [network debug] [00009] wait for result
2019-10-09 20:53:45.266 [main debug] >>> 

(many empty lines)

2019-10-09 20:53:45.266 [main debug] <<< ERR: command not found
2019-10-09 20:53:45.266 [network debug] [00009] wait for result
2019-10-09 20:53:45.266 [main debug] >>> 

(many empty lines)

2019-10-09 20:53:45.266 [main debug] <<< ERR: command not found
2019-10-09 20:53:45.266 [network debug] [00009] wait for result
2019-10-09 20:53:45.266 [main debug] >>> 

(many empty lines)

2019-10-09 20:53:45.266 [main debug] <<< ERR: command not found
2019-10-09 20:53:45.266 [network debug] [00009] wait for result
2019-10-09 20:53:45.266 [main debug] >>> 

(many empty lines)

2019-10-09 20:53:45.266 [main debug] <<< ERR: command not found
2019-10-09 20:53:45.266 [network debug] [00009] wait for result
2019-10-09 20:53:45.266 [main debug] >>> 

...

**That's the output from ebusctl write:**

#######################################################################################
09.10.2019 20:53:45
#######################################################################################
Circuit: 700
Command: OpMode
Value: off
ebusctl write -c 700 OpMode off
ERR: invalid numeric argument

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

ERR: command not found

I have no trouble when the commend is something with "ebsuctl read". Only in the case when I write something on the bus.

john30, 2019-10-10:
please provide the complete python script as well as the cron entry. I suspect that your cron entry does not consume the ebusctl output somehow

bizonxx, 2019-10-10:
I've got the same isse with crone I also did test in a backgroud process and it also ends with the same empty lines in ebusd log like @danielkargs log 
The script
```
#!/bin/sh 
while true
do
    echo $(date "+%Y-%m-%d %H:%M:%S")  $(/usr/bin/ebusctl read -f "FlowTemp") >> /tmp/ebusd.txt
    sleep 10
done
```

**Console No1**

```
root@OpenWrt:~# /etc/ebusd/test.sh &
root@OpenWrt:~#
```
**Console No2**
```
root@OpenWrt:~# tail -f /tmp/ebusd.txt

2019-10-09 22:46:40 20.38;ok
2019-10-09 22:46:50  20.38;ok
2019-10-09 22:47:00  20.38;ok
2019-10-09 22:47:11  20.38;ok
2019-10-09 22:47:21  ERR: element not found    <-------------- exit on Console No1
2019-10-09 22:47:31  ERR: element not found
2019-10-09 22:47:41  ERR: element not found
2019-10-09 22:47:51  ERR: element not found
2019-10-09 22:48:01  ERR: element not found

```
After close connection on **Console No1**  errors apears.

danielkarg, 2019-10-10:
Thanks bizonxx for the hint! I think that I have found the cause of the problem! 
If I attach the "&" to the cron entry (-> background), then it comes to the error. 

**Works:**
* * * * * python script.py

**Doesn't work:**  
* * * * * python script.py **&**

I will now work out a solution that my "very big" script does not have to run in the background.

Thanks!!!

closed 2019-10-10 by danielkarg

bizonxx, 2019-10-10:
Right now my solution (thanx @kofec) is to run it with screen 

`*/1 * * * * /usr/sbin/screen -d -m /etc/ebusd/ebus-get-data.sh`

it works with any problems

danielkarg, 2019-10-10:
Thanks Thanks Thanks! You have made a very happy person out of me!