john30/ebusd #300: Please consider adding support for configuration file

andr2000, 2019-10-21:
Currently the daemon has only possibility to be configured via its command line which can be long enough: please see [command line arguments](https://github.com/john30/ebusd/wiki/2.-Run)
While it can be ok for the most of the use-cases there is one that might be disturbing: mqtt credentials and password are easily seen with ps for example. There are many ways passwords can [leak](https://www.netmeister.org/blog/passing-passwords.html) and storing credentials in the configuration file may be an option.

closed 2019-10-27 by john30

andr2000, 2019-10-28:
Hm, could you please give me a hint how [this commit](https://github.com/john30/ebusd/commit/ae91be0874852259f47904da309d3a95359334a9) is going to solve the problem that process' command line is visible through `cat /proc/PID` for example?

And the more generic request also was not to hide MQTT password, but to consider having a configuration file....

john30, 2019-10-28:
the commit is hiding the secrets revealed by e.g. ps, at least during my tests.
yes, and this is a community project, so feel free to add the support for reading a configuration file yourself...

andr2000, 2019-10-28:
sure, `patches are welcome` ;)

commit ae91be08 (2019-10-27): simplified hiding secrets from cmdline (fixes #300)