john30/ebusd #302: Problem with Copy the right source list

Nijlpaard, 2019-11-03:
After succusfully adding the Add the GPG key to your trusted apt sources
as described here https://github.com/john30/ebusd-debian/blob/master/README.md
The next command : wget -O /etc/apt/sources.list.d/ebusd.list https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd-ARCH-nomqtt.list
i get error 404, also whem i did
wget -O /etc/apt/sources.list.d/ebusd.list https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd-3.4_armhf-stretch_mqtt1.deb
What is wrong ? (i have an Raspberry Pi with Stretch and logged in as root)

john30, 2019-11-04:
as written in the readme, you need to replace ARCH with your architecture.
furthermore, version 3.4 is not yet available as written in the release description

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30