john30/ebusd #303: no signal

cbeham, 2019-11-05:
Hello John!

Can you help me? 
Just installed ebusd on Raspberry Pi3 with esera ethernet adapter via LAN. 
But cannot get a signal!

sudo nano /etc/default/ebusd:
# /etc/default/ebusd:
# config file for ebusd service.

# Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="--scanconfig --enablehex --port=8888"

ebusctl info:
version: ebusd 3.4.v3.4
update check: OK
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

Logfile:
[main notice] ebusd 3.4.v3.4 started with auto scan
2019-11-05 20:08:44.105 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2019-11-05 20:08:44.105 [bus notice] bus started with own address 31/36
2019-11-05 20:08:44.105 [bus notice] device invalid
2019-11-05 20:08:49.105 [bus error] unable to open /dev/ttyUSB0: ERR: element not found

THank you very much!
Chris

Dinth, 2019-11-05:
https://github.com/john30/ebusd/wiki/2.-Run

You need at least --device option, otherwise ebusd will not know what type of ebus device you've got and where to look for it

cbeham, 2019-11-05:
Hi Dinth!

I have the Ethernet Koppler Service running under 10.0.0.25:5000 TCP 

and tried that in config:
--device=10.0.0.25:5000 --scanconfig --port=8888

but always the same now:
pi@raspberrypi001:~ $ ebusctl i
error connecting to localhost:8888

andig, 2019-11-05:
> error connecting to localhost:8888

Are you sure you're running ebusd and ebusctl on the same machine? Are you sure that ebusd is actually listening on the port or has something else occupied it before?

Dinth, 2019-11-05:
> --device=10.0.0.25:5000 --scanconfig --port=8888 
> but always the same now:
> pi@raspberrypi001:~ $ ebusctl i
> error connecting to localhost:8888

But has your ebusd log changed?

Because we are looking at two different issues here: your ebusd not connecting to your serial<>ethernet adapter and ebusctl not connecting to your ebusd daemon.

What happens when you run:
> telnet 10.0.0.25 5000
from your ebusd machine
and what happens when you run:
> telnet your_ebusd_ip 8888 from your raspberrypi001?

cbeham, 2019-11-06:
> > error connecting to localhost:8888
> 
> Are you sure you're running ebusd and ebusctl on the same machine? Are you sure that ebusd is actually listening on the port or has something else occupied it before?

i installed ebusd on the raspberry pi. 
netstat deliveres no such port 8888.

cbeham, 2019-11-06:
> > --device=10.0.0.25:5000 --scanconfig --port=8888
> > but always the same now:
> > pi@raspberrypi001:~ $ ebusctl i
> > error connecting to localhost:8888
> 
> But has your ebusd log changed?

What do you mean?

> Because we are looking at two different issues here: your ebusd not connecting to your serial<>ethernet adapter and ebusctl not connecting to your ebusd daemon.
> 

is ebusctl different programm?

> What happens when you run:
> 
> > telnet 10.0.0.25 5000
> > from your ebusd machine

ebusd is installed on the pi:

pi@raspberrypi001:~ $ telnet 10.0.0.25 5000
Trying 10.0.0.25...
telnet: Unable to connect to remote host: Connection refused

> > and what happens when you run:
> > telnet your_ebusd_ip 8888 from your raspberrypi001?

ebusd is on the same machine (pi) ?

andig, 2019-11-06:
Can you please show any indication from your pi that

a) ebusd is running and
b) has port 8888 open?

this is for the ebusctl can’t connect issue. Please have a single issue per problem to not confuse things.

cbeham, 2019-11-06:
yes, thank you. i think i understand now a little bit...

i think ebusd starts NOT automatically because when i write:

pi@raspberrypi001:~ $ service ebusd start
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Legitimierung ist zum Starten von »ebusd.service« notwendig.
Multiple identities can be used for authentication:
 1.  ,,, (pi)
 2.  root
Choose identity to authenticate as (1-2):

then i choose 1 and have to write the password. 
after that i get with ebusctl -i:

pi@raspberrypi001:~ $ ebusctl i
version: ebusd 3.4.v3.4
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

so how i can start ebusd automatically?

i tried that:
pi@raspberrypi001:~ $ sudo nano /etc/rc.local
and got this:
!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.
service ebusd start

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

exit 0

something wrong?

cbeham, 2019-11-06:
yes i found the problem: first line missing #
now the ebuds starts auto
but still not get a signal
THANK for HINT!!!!

cbeham, 2019-11-06:
GOT IT!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

changed port on the Ethernet Koppler from 5000 to 5007
[image]

restart Koppler
changes ebusd config file to:
[image]

ebusctl -i produces now:

```
version: ebusd 3.4.v3.4
update check: OK
signal: acquired
symbol rate: 28
max symbol rate: 50
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
THANK YOU VERY MUCH. SO HAPPY!!!!!!!!!!!!!!!!!!!!!!!!!! THANKS

whats next?

skyynet, 2020-05-17:
I also have the esera ebus connector ethernet and installed ebusd on my Synology as Docker image. The ebus connector is configured to TCP 192.168.1.70:5000

[image]

I can successfuly ping 192.168.1.70

I configured ebusd to start with parameters -f --scanconfig -d tcp:192.168.1.70:5000

Unfortunately I get the following error in terminal

[bus error] unable to open tcp:192.168.1.70:5000: ERR: generic I/O error
[bus notice] device invalid

I tried it with port 5007 though I didn't understand why that should work better. Unfortunately it doesn't solve my problem.

Any idea what to change to get it working?

Maddin2020, 2020-05-17:
Yes you want use TCP but your Esera Adapter is Udp you Must config tcp Server or make it How myself use With the selected Port.

skyynet, 2020-05-17:
The esera supports tcp & udp. I configured it to use TCP on IP 192.168.1.70 and Port 5000 (see screenshot) so this should work.

skyynet, 2020-05-24:
I found the error. I configured the device as TCP client instead of TCP server. Stupid me.

closed 2020-09-20 by john30

schmidhorst, 2021-07-03:
If the esera is set to UDP, then in the ebusd configuration file you  need to add "udp:" in the EBUSD_OPTS, e.g. 
--device=udp:192.168.xxx.xxx:5000

hbarnard, 2021-11-23:
Just started with an esera ethernet interface (beginning a bit of buyer's remorse but), so this is a _stupid_ question:

I've set TCP server, the remote host ip is the ip of eth0 in the Raspberry Pi that's  (currently) directly connected by ethernet cable to the interface? I seem to be getting handshaking anyway, but nothing much is coming into the ebusd.log except things like:

```
2021-11-22 17:41:41.194 [main error] scan config 08: ERR: element not found
2021-11-22 17:43:48.713 [main notice] update check: version 21.2 available
2021-11-22 21:37:03.278 [bus notice] new master 10, master count 2
2021-11-22 21:37:03.278 [update notice] received unknown BC cmd: 10feed8b08c026ec2cf8051003
```

I'm guessing that it's working and that I have templates and configuration not set up correctly? Sorry, that's two stupid questions, in fact.

schmidhorst, 2021-11-23:
If the esera is set to "TCP server", then the field "remote IP/host name" in the Configuration Tool is ignored. 
The "[bus notice] new master 10, master count 2" sounds good in my opinion: The 1st master is the ebusd on the Raspberry, the 2nd master is another device in your ebus system. 
Some time I got bad data because of wrong settings in the tab "Serial" of the Configuration Tool. "Enable debug message output" needs to be disabled. And the (esera internal) COM port need to be set to 2400 Baud, 8N1, NONE. 
Have you been able to adjust the potentiometer of the esera according to https://github.com/john30/ebusd/wiki/6.-Hardware#adjusting-the-potentiometer ?

hbarnard, 2021-11-24:
> Some time I got bad data because of wrong settings in the tab "Serial" of the Configuration Tool. "Enable debug message output" needs to be disabled. And the (esera internal) COM port need to be set [...]

Thanks for all this. I'll re-configure and try again also adjusting the potentiometer, when I saw the config tool, my heart sank somewhat. I may also try UDP as there's a suggestion somewhere that it deals better with esera latency problems.