john30/ebusd #304: MQTT request for all scan results

andr2000, 2019-11-05:
I cannot find any way scanning can be triggered via MQTT (minor) and what is more important there is no generic way to get scan results without knowing bus masters' addresses, e.g. if 2 devices on the bus exist, say 08 and 15, then one can directly query scan results for those with:
```
mosquitto_pub -h 192.168.10.200 -t 'ebusd/scan.08/id/get' -m '?'
mosquitto_pub -h 192.168.10.200 -t 'ebusd/scan.15/id/get' -m '?'
```
But we cannot use MQTT to discover all the devices known so far in a generic way. It would be great to have for MQTT the functionality already available for ebusctl:
```
~# ebusctl scan result 
08;Vaillant;BAI00;0104;7803;21;18;23;0010021961;0001;005167;N4
15;Vaillant;B7V00;0422;5503;21;18;36;0020197207;0082;013759;N6
```
How about something like
```
mosquitto_pub -h 192.168.10.200 -t 'ebusd/scan' -m '?'
```
just like `/list` does?

john30, 2019-11-06:
you can basically achieve this by subscribing to everything and publishing to the /list with only the ebusd prefix

andr2000, 2019-11-06:
As a workaround yes, but this implies unneeded load to the receiving side (it only needs scan results once during initialization, but will receive ALL messages during its lifetime which seems to be a waste of resources). I have the following scenario:
1. I use Home Assistant (HA) MQTT binding to connect to ebusd.
2. During HA initialization I need to understand current ebusd state, so I can align HA automations' states:
2.1. If ebusd is not running, then wait for ebusd/global/running
2.2. If ebusd runs, but hasn't yet acquired the signal, then wait for ebusd/global/signal
2.3. After the signal has been acquired we want to wait for the equipment to be scanned (in my case I want to set up polling priorities and this cannot be done unless scan is complete, e.g. supported messages become known to ebusd):
2.3.1. I subscribe to ebusd/scan.08/id and ebusd/scan.15/id
2.3.2. If ebusd has already performed scanning then ebusd/scan.XX/id are not updated as there is no change in their state and also the relevant nodes are not published with retain flag set.
So, here I have to trigger the scanning manually. As there is no generic scan request, e.g. ebusd/scan/get or something, I publish the following:
```
mosquitto_pub -h 192.168.10.200 -t 'ebusd/scan.08/id/get' -m '?'
mosquitto_pub -h 192.168.10.200 -t 'ebusd/scan.15/id/get' -m '?'
```
and finally get updates via ebusd/scan.XX/id

So, having a dedicated /scan node could definitely help in such automations that HA uses.

john30, 2019-11-09:
this doesn't sound like the right approach to me. instead I'd rather add a scan status to the global topic that allows to distinguish between a scan currently in progress, not yet started or completely finished

andr2000, 2019-11-09:
Sounds good, this will allow me to wait for "scan finished" event. But what would be the way to read scan results in a generic way? Today I have to explicitly specify "ebusd/scan.08/id" meaning that I already know about 08 etc. Would it be possible to also add "ebusd/global/scan_results"?
Can we please have both scan and scan_results published with retain flag set?

andr2000, 2019-12-07:
@john30 any progress on this?
Thank you

john30, 2019-12-08:
apparently not yet

john30, 2020-04-11:
fixed with commits f313658f90e11f1999343d8232fa28a8c3a81a2b and 6cea44947833385fdbb0539e886e40dc78c5f599. you can now make ebusd list all scans by using publishing "ebusd/scan.*/list" and listen for "global/scan" topic to check if a scan is active

closed 2020-04-11 by john30

andr2000, 2020-05-20:
Any plan to get the relevant commits into master?

john30, 2020-06-01:
just cherry picked them into master

commit f313658f (2020-04-11): add global/scan topic revealing status of --scanconfig option initiated scan (fixes #304)
commit edfe09a3 (2020-06-01): add global/scan topic revealing status of --scanconfig option initiated scan (fixes #304)