john30/ebusd #305: Vaillant VR920

christesting, 2019-11-06:
I am a newbie in the area of EEBus - so maybe my question is stupid...but I will try it.

I got a new heating system from Vaillaint. 
Vaillant installed the VR920 to connect the VRC 700 to the LAN / Internet. The connection between the VR920 and VRC 700 is the EEBus.

Vaillant wrote me, that it is possible to use the VR920 as "interface" to the EEBus and I don´t need extra equipment to connect to the EEBus. 

The VR920 has an open port: 4500

Is it possible to use ebusd to interact with the VR920 (and the VRC 700) via LAN?

thx

andig, 2019-11-06:
Ebusd is about ebus. Afaik Eebus is something different.

christesting, 2019-11-06:
they told me, that I can find everything at https://www.eebus.org and that I don´t have to buy extra equipment to "talk" to the VRC 700 ...

The principle is shown here (I have different euqipment):
[https://www.vaillant.de/heizung/produkte/vernetzte-systeme/eebus/](url)

ok - now I unterstand why I am confused: ebus vs eebus......

john30, 2019-11-09:
eebus seems to be a complete new protocol defined from scratch, so this is not related to ebusd at all

closed 2019-11-09 by john30