john30/ebusd #306: ebusctl running as a service

Christoph1111, 2019-11-07:
I just have a script written in python3 to get some parameters from the heating. Everything is working fine. The script sends some commands from the bai.xxx.inc via

```
#!/usr/bin/python3
 def get_EBus_Values(command):
    try:
        p=subprocess.Popen(["ebusctl", "read", "-f", command], stdout=subprocess.PIPE)
        p.wait()
 
        meas=p.stdout.read().decode().strip()
        m=meas.decode().strip().split(";")[0]
        

etc..
```
and dumps to disk.
But running this script via systemctl as a service I get "ERR: element not found" on many requests, some work. Is that a timing problem?
Any help  is welcome.
System: Raspberry 3B, Buster, Esera usb2ebus, Vaillant eco

Sorry for posting on ebusd-configuration, it was late...

Regards
Chris

john30, 2019-11-09:
there should be some reasonable logging from ebusd in the case of your script getting an element not found. try to find these first and look for the scan status at the time of that log entry. it might just be a startup issue, where ebusd still needs some time to finish the scanning

Christoph1111, 2019-11-09:
Hi,
Meanwhile I checked the log and saw that the default reading is without errors but my ebusctl requests (26) are not complete. You can see that the first run has 18 answers (starting with storageTemp, ending with PrEnergyCountHc1), the second has 21(storageTemp to WaterpressureMeasureCounter):
```

2019-11-08 00:02:46.940 [update notice] sent read bai StorageTemp QQ=31: 52.25;ok
2019-11-08 00:02:47.087 [update notice] sent read bai WP QQ=31: off
2019-11-08 00:02:47.231 [update notice] sent read bai extWP QQ=31: off
2019-11-08 00:02:47.381 [update notice] sent read bai PrimaryCircuitFlowrate QQ=31: 0.00
2019-11-08 00:02:47.533 [update notice] sent read bai FlowTemp QQ=31: 28.25;ok
2019-11-08 00:02:47.720 [update notice] sent read bai HwcHours QQ=31: 182
2019-11-08 00:02:47.873 [update notice] sent read bai HwcStarts QQ=31: 700
2019-11-08 00:02:48.021 [update notice] sent read bai ValveStarts QQ=31: 1121
2019-11-08 00:02:48.174 [update notice] sent read bai FanHours QQ=31: 510
2019-11-08 00:02:48.327 [update notice] sent read bai FanStarts QQ=31: 2373
2019-11-08 00:02:48.482 [update notice] sent read bai PrEnergySumHwc1 QQ=31: 232139340
2019-11-08 00:02:48.639 [update notice] sent read bai PrEnergyCountHwc1 QQ=31: 662382
2019-11-08 00:02:48.791 [update notice] sent read bai WaterHcFlowMax QQ=31: 5413
2019-11-08 00:02:48.940 [update notice] sent read bai WaterpressureVariantSum QQ=31: 65078
2019-11-08 00:02:49.080 [update notice] sent read bai WaterpressureMeasureCounter QQ=31: 131
2019-11-08 00:02:49.236 [update notice] sent read bai PrEnergySumHc1 QQ=31: 195366938
2019-11-08 00:02:49.393 [update notice] sent read bai PrEnergyCountHc1 QQ=31: 1030107
2019-11-08 00:02:56.150 [update notice] received read bai Status01 QQ=10: 28.0;28.5;7.312;-;52.0;off
2019-11-08 00:02:56.418 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2019-11-08 00:03:06.240 [update notice] received read bai Status01 QQ=10: 28.0;28.5;7.312;-;52.0;off
2019-11-08 00:03:06.508 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2019-11-08 00:03:06.765 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;65.0
2019-11-08 00:03:16.302 [update notice] received read bai Status01 QQ=10: 28.0;28.5;7.312;-;52.0;off
2019-11-08 00:03:16.570 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2019-11-08 00:03:26.346 [update notice] received read bai Status01 QQ=10: 28.0;28.5;7.312;-;52.0;off
2019-11-08 00:03:26.615 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2019-11-08 00:03:36.408 [update notice] received read bai Status01 QQ=10: 28.0;28.5;7.312;-;52.0;off
2019-11-08 00:03:36.675 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2019-11-08 00:03:36.949 [update notice] received read bai DateTime QQ=10: sync;-:-:-;-.-.-;7.312
2019-11-08 00:03:37.699 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;65.0
2019-11-08 00:03:37.827 [update notice] received scan-read scan.15  QQ=03: Vaillant;70000;0613;6903
2019-11-08 00:03:37.980 [update notice] received update-read broadcast vdatetime QQ=10: 00:03:36;08.11.2019
2019-11-08 00:03:41.500 [update notice] received unknown MS cmd: 1008b5110100 / 08c40116001f000000
2019-11-08 00:03:41.717 [update notice] received update-read broadcast outsidetemp QQ=10: 7.312
2019-11-08 00:03:46.481 [update notice] received read bai Status01 QQ=10: 28.0;28.5;7.312;-;52.0;off
2019-11-08 00:03:46.749 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2019-11-08 00:03:52.302 [update notice] sent read bai StorageTemp QQ=31: 52.25;ok
2019-11-08 00:03:52.446 [update notice] sent read bai extWP QQ=31: off
2019-11-08 00:03:52.595 [update notice] sent read bai CirPump QQ=31: off
2019-11-08 00:03:52.739 [update notice] sent read bai PumpPower QQ=31: 0
2019-11-08 00:03:52.896 [update notice] sent read bai FlowTemp QQ=31: 28.25;ok
2019-11-08 00:03:53.057 [update notice] sent read bai ReturnTemp QQ=31: 28.50;65079;ok
2019-11-08 00:03:53.212 [update notice] sent read bai OutdoorstempSensor QQ=31: 7.31;ok
2019-11-08 00:03:53.361 [update notice] sent read bai Flame QQ=31: off
2019-11-08 00:03:53.509 [update notice] sent read bai HcHours QQ=31: 284
2019-11-08 00:03:53.662 [update notice] sent read bai HwcHours QQ=31: 182
2019-11-08 00:03:53.811 [update notice] sent read bai HcStarts QQ=31: 1200
2019-11-08 00:03:53.962 [update notice] sent read bai HwcStarts QQ=31: 700
2019-11-08 00:03:54.110 [update notice] sent read bai PumpHours QQ=31: 798
2019-11-08 00:03:54.258 [update notice] sent read bai HcPumpStarts QQ=31: 2915
2019-11-08 00:03:54.407 [update notice] sent read bai ValveStarts QQ=31: 1121
2019-11-08 00:03:54.556 [update notice] sent read bai FanHours QQ=31: 510
2019-11-08 00:03:54.704 [update notice] sent read bai FanStarts QQ=31: 2373
2019-11-08 00:03:54.864 [update notice] sent read bai PrEnergySumHwc1 QQ=31: 232139340
2019-11-08 00:03:55.025 [update notice] sent read bai PrEnergyCountHwc1 QQ=31: 662382
2019-11-08 00:03:55.171 [update notice] sent read bai WaterHcFlowMax QQ=31: 5413
2019-11-08 00:03:55.317 [update notice] sent read bai WaterpressureMeasureCounter QQ=31: 131
```
No message about failed readings.

My MQTT client finds 3 topics from ebusd: global, broadcast, bai which are working well. My question is: where can I modify ebusd to send more requests via the bai topic? Maybe this could be a solution.

And thanks for this great project...

Regards
Chris

Christoph1111, 2019-11-09:
OK, seemed to be a timing problem.
I changed the "sporty" default values in /etc/default/ebusd to
 --acquiretimeout=100000
 --receivetimeout=250000
 --pollinterval=20

and since then messaging is running without breaks.

Have I understand it correctly to just add a r1-9 into the baixxx.inc file's first column to let ebusd automatically request these parameters?

Regards
Chris

 1Hilfe  2Spe~rn 3Mar~en 4Ers~en 5Kop~en 6Ver~en 7Suchen 8Lös~en 9Menüs 10Bee~en

Christoph1111, 2019-11-10:
Problem identified but not solved:
The python script with its request of 26 parameters with the "ebusctl read -f ***" command will get answers until the ebusd sends his requests periodically (in german "reingrätschen"). Then the error "Element not found" will occur. So it's neither a service problem nor a timing problem...
I close this topic.

Regards
Chris

closed 2019-11-10 by Christoph1111

andig, 2019-11-10:
Very strange. I don‘t have this problem regularly polling from node-red.