john30/ebusd #307: ebusd not auto start

cbeham, 2019-11-08:
after changing ebusd config file i do: 
`sudo reboot`

but then ebusd is not starting automatically:
if i do on a raspberry pi3:
```
sudo cp contrib/debian/init.d/ebusd /etc/init.d/
sudo update-rc.d ebusd defaults
sudo update-rc.d ebusd enable
sudo /etc/init.d/ebusd start
```
i got this:
`pi@raspberrypi001:/ebusd $ sudo /etc/init.d/ebusd start
[....] Starting ebusd (via systemctl): ebusd.serviceJob for ebusd.service failed because of unavailable resources or another system error.
See "systemctl status ebusd.service" and "journalctl -xe" for details.
 failed!
`
what is the difference between systemD and systemV?

if i do:
ebusctl i
`error connecting to localhost:8888
`
how i can start ebsud?

cbeham, 2019-11-09:
solved by deinstallation and new installation!

**New Problem:
 **[main error] unable to load scan config 38: no file from vaillant with prefix 38 found****

```
2019-11-08 20:03:27.197 [main notice] ebusd 3.4.v3.4 started with auto scan
2019-11-08 20:03:27.453 [bus notice] bus started with own address 31/36
2019-11-08 20:03:27.455 [bus notice] signal acquired
2019-11-08 20:05:31.922 [main notice] update check: OK
2019-11-08 21:17:42.304 [bus notice] new master 00, master count 2
2019-11-08 21:17:44.669 [main error] scan config 05: ERR: read timeout
2019-11-08 21:19:46.725 [main notice] update check: OK
2019-11-09 00:57:26.179 [bus notice] new master 33, master count 3
2019-11-09 00:57:35.651 [bus notice] scan 38: ;Vaillant;V32;0117;9802
2019-11-09 00:57:35.651 [update notice] store 38 ident: done
2019-11-09 00:57:35.651 [update notice] sent scan-read scan.38  QQ=31: Vaillant;V32;0117;9802
2019-11-09 00:57:35.651 [bus notice] scan 38: ;Vaillant;V32;0117;9802
2019-11-09 00:57:35.711 [main error] unable to load scan config 38: no file from vaillant with prefix 38 found
2019-11-09 00:57:35.711 [main error] scan config 38: ERR: element not found
2019-11-09 00:59:37.766 [main notice] update check: OK
```

john30, 2019-11-09:
covered here: https://github.com/john30/ebusd-configuration/issues/150
why do you create duplicate issues???

closed 2019-11-09 by john30

cbeham, 2019-11-09:
Ahhh, sorry, i changed the wrong topic name!! changed the tpoic here....