john30/ebusd #308: No connection with ebus adapter

mbbelgium, 2019-11-09:
I have a ebusadapter via wifi based on a wemos D1 mini. it is working and makes connection with the ebus: 
[image]

The ebusd is installed on a raspberry-pi 4 with raspbian on it. this is the output of the ebusctlinfo:
`
pi@raspberrypi:/var/log $ ebusctl info
version: ebusd 3.4.v3.4
update check: OK
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
`
Unfortunately something is not working, and I can't find why. It's stops at "update check: ok" and then keeps "performing regular tasks"...

`pi@raspberrypi:~ $ ebusd --scanconfig -f -d 192.168.0.220:9999 --loglevel=debug --logfile=/var/log/ebusd.log
2019-11-09 20:14:24.407 [main notice] ebusd 3.4.v3.4 started with auto scan
2019-11-09 20:14:24.408 [main info] loading configuration files from http://ebusd.eu/config/
2019-11-09 20:14:24.508 [main info] reading templates /
2019-11-09 20:14:24.607 [main info] read templates in /
2019-11-09 20:14:24.607 [main info] reading file memory.csv
2019-11-09 20:14:24.720 [main info] successfully read file memory.csv
2019-11-09 20:14:24.720 [main info] reading file broadcast.csv
2019-11-09 20:14:24.818 [main info] successfully read file broadcast.csv
2019-11-09 20:14:24.819 [main info] read config files
2019-11-09 20:14:24.857 [bus notice] bus started with own address 31/36
2019-11-09 20:14:24.857 [main info] registering data handlers
2019-11-09 20:14:24.857 [main info] registered data handlers
2019-11-09 20:14:34.858 [main debug] performing regular tasks
2019-11-09 20:14:44.858 [main debug] performing regular tasks
2019-11-09 20:14:54.858 [main debug] performing regular tasks
2019-11-09 20:15:04.858 [main debug] performing regular tasks
2019-11-09 20:15:14.859 [main debug] performing regular tasks
2019-11-09 20:15:24.859 [main debug] performing regular tasks
2019-11-09 20:15:34.859 [main debug] performing regular tasks
2019-11-09 20:15:44.860 [main debug] performing regular tasks
2019-11-09 20:15:54.860 [main debug] performing regular tasks
2019-11-09 20:16:04.860 [main debug] performing regular tasks
2019-11-09 20:16:14.861 [main debug] performing regular tasks
2019-11-09 20:16:24.861 [main debug] performing regular tasks
2019-11-09 20:16:34.861 [main debug] performing regular tasks
2019-11-09 20:16:34.949 [main notice] update check: OK
2019-11-09 20:16:44.949 [main debug] performing regular tasks
`
Can someone point me into the right direction?

john30, 2019-11-10:
you don't have a signal, consequently there is not much that ebusd can do here

mbbelgium, 2019-11-10:
Hi John, thanx for your answer, but wat you mean no signal? No signal on the ebus? If yes, why does the ebus-adapter say  "ebus signal acquired" on the webinterface?
Ebusd sees the adapter, because if I put a wrong IP adress, I get "device invalid". 

Do you have an idea on how  to continue debugging to find where the problem comes from? In the mean time I will check the ebus adapter circuit again.

john30, 2019-11-10:
info shows "signal: no signal" and "signal acquired" is missing in ebusd log, so ebusd is not connected to the right device I guess. does the device string of ebusd-esp match that of ebusd commandline argument? and can you ping the address?
at the end, ebusd-esp needs to show "ebusd connected: yes"

mbbelgium, 2019-11-10:
Hi John,

I can ping the device from the PI:
> pi@raspberrypi:~ $ ping 192.168.0.220
PING 192.168.0.220 (192.168.0.220) 56(84) bytes of data.
64 bytes from 192.168.0.220: icmp_seq=1 ttl=254 time=20.7 ms
64 bytes from 192.168.0.220: icmp_seq=2 ttl=254 time=21.7 ms

this is what I use in ebusd:

> ebusd --scanconfig -f -d 192.168.0.220:9999 --loglevel=debug                                                                                                              --logfile=/var/log/ebusd.log

this is the setting in the adapter:
[image]

I have been testing a little, and maybe I have a dead optocoupler in the adapter. I think that the adapter sees messages and gives them to ebusd, and Ebusd can't send messages. Maybe that's the reason why the adapter says that conection is ok, but ebusd has no connection. Unfortunately I don't have the right optocoupler in spare (cny17f-2 vs cny17f-3). I will order new once to test again...

mbbelgium, 2019-11-11:
Hi John, with the new binaries I got a little further! I have a signal and communication with ebusd:
[image]

and the output looks like this now: 

> pi@raspberrypi:~ $ ebusd --scanconfig -f -d 192.168.0.212:9999 --loglevel=debug --logfile=/var/log/ebusd.log
2019-11-11 12:38:38.862 [main notice] ebusd 3.4.v3.4 started with auto scan
2019-11-11 12:38:38.863 [main info] loading configuration files from http://ebusd.eu/config/
2019-11-11 12:38:38.967 [main info] reading templates /
2019-11-11 12:38:39.072 [main info] read templates in /
2019-11-11 12:38:39.072 [main info] reading file memory.csv
2019-11-11 12:38:39.167 [main info] successfully read file memory.csv
2019-11-11 12:38:39.168 [main info] reading file broadcast.csv
2019-11-11 12:38:39.280 [main info] successfully read file broadcast.csv
2019-11-11 12:38:39.280 [main info] read config files
2019-11-11 12:38:39.311 [bus notice] bus started with own address 31/36
2019-11-11 12:38:39.312 [bus debug] ERR: SYN received during no signal, switching to ready
2019-11-11 12:38:39.312 [bus notice] signal acquired
2019-11-11 12:38:39.312 [main info] registering data handlers
2019-11-11 12:38:39.312 [main info] registered data handlers
2019-11-11 12:38:43.042 [bus notice] new master 10, master count 2
2019-11-11 12:38:43.109 [bus notice] new master 03, master count 3
2019-11-11 12:38:43.110 [update info] received MS cmd: 1008b503020001 / 0affffffffffffffffffff
2019-11-11 12:38:43.110 [update notice] received unknown MS cmd: 1008b503020001 / 0affffffffffffffffffff
2019-11-11 12:38:43.386 [update info] received MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-11-11 12:38:43.386 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-11-11 12:38:43.664 [update info] received MS cmd: 1008b503020003 / 0a1f00ffffffffffffffff
2019-11-11 12:38:43.664 [update notice] received unknown MS cmd: 1008b503020003 / 0a1f00ffffffffffffffff
2019-11-11 12:38:43.946 [update info] received MS cmd: 1008b503020004 / 0affffffffffffffffffff
2019-11-11 12:38:43.946 [update notice] received unknown MS cmd: 1008b503020004 / 0affffffffffffffffffff
2019-11-11 12:38:44.038 [bus notice] max. symbols per second: 106
2019-11-11 12:38:44.224 [update info] received MS cmd: 1008b51009000000ffffff01ff00 / 0101
2019-11-11 12:38:44.224 [update notice] received unknown MS cmd: 1008b51009000000ffffff01ff00 / 0101
2019-11-11 12:38:44.486 [update info] received MS cmd: 1008b5110101 / 09272a0080ffff0000ff
2019-11-11 12:38:44.486 [update notice] received unknown MS cmd: 1008b5110101 / 09272a0080ffff0000ff
^X2019-11-11 12:38:47.896 [update info] received MS cmd: 1008b5110102 / 06033c96468c5a
2019-11-11 12:38:47.896 [update notice] received unknown MS cmd: 1008b5110102 / 06033c96468c5a
 
But still a no signal error: 

> pi@raspberrypi:~ $ ebusctl state
ERR: no signal

> pi@raspberrypi:~ $ ebusctl info
version: ebusd 3.4.v3.4
update check: OK
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

I will continue searching, but if you have any idea's, let me know...

mbbelgium, 2019-11-13:
Optocouples are in and well well well, we got something: 

> 2019-11-13 18:57:17.895 [update notice] sent scan-read scan.15 id QQ=31: 21;15;44;0020197223;0932;012720;N8
2019-11-13 18:57:17.895 [bus debug] notify request: done
2019-11-13 18:57:17.895 [bus notice] scan 15: ;21;15;44;0020197223;0932;012720;N8
2019-11-13 18:57:17.896 [bus debug] switching from send response ACK to send SYN
2019-11-13 18:57:17.908 [bus debug] send/receive symbol latency 12 ms
2019-11-13 18:57:18.039 [bus notice] max. symbols per second: 112
2019-11-13 18:57:18.071 [main debug] found 21 matching scan config files from vaillant with prefix 15: done
2019-11-13 18:57:18.071 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "scc00", SW1800, HW3903
2019-11-13 18:57:18.071 [main error] scan config 15: ERR: element not found
2019-11-13 18:57:20.071 [main debug] performing regular tasks
2019-11-13 18:57:24.262 [update info] received MS cmd: 1008b503020001 / 0affffffffffffffffffff
2019-11-13 18:57:24.262 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-11-13 18:57:24.541 [update info] received MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-11-13 18:57:24.541 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-11-13 18:57:24.819 [update info] received MS cmd: 1008b503020003 / 0a04002900ffffffffffff
2019-11-13 18:57:24.820 [update notice] received unknown MS cmd: 1008b503020003 / 0a04002900ffffffffffff
2019-11-13 18:57:25.098 [update info] received MS cmd: 1008b503020004 / 0affffffffffffffffffff
2019-11-13 18:57:25.098 [update notice] received unknown MS cmd: 1008b503020004 / 0affffffffffffffffffff
2019-11-13 18:57:25.376 [update info] received MS cmd: 1008b5100900003affffff00ff00 / 0101
2019-11-13 18:57:25.377 [update notice] received update-write bai SetMode QQ=10: auto;29.0;-;-;0;0;0;0;0;0
2019-11-13 18:57:25.653 [update info] received MS cmd: 1008b5110101 / 093a300080ffff0100ff
2019-11-13 18:57:25.654 [update notice] received read bai Status01 QQ=10: 29.0;24.0;-;-;-;on

what bothers me now is this:

> 2019-11-13 18:57:18.071 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "scc00", SW1800, HW3903

Let's check how to fix this so that I can read the values in a more understandable way. If someone already got there and can help me, don't hesitate!

john30, 2019-11-16:
exactly, there is just no CSV definition file for scc yet

mbbelgium, 2019-11-16:
I suppose we can close this one then! :-)
Shall we open a new "issue" for the scc definition file? Are you planning to make one? If you can point me out how to do it and with what file to start with, I will be glad to help you make one.

john30, 2019-11-17:
I guess so.
you'll have to do the main part of it, start looking at the [wiki howto](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files) and/or check if you can borrow from another one

closed 2019-11-17 by john30