john30/ebusd #309: Wrong time schedule over MQTT topic

andr2000, 2019-11-10:
I am trying to request a single time schedule topic over MQTT:
`mosquitto_pub -h 192.168.10.200 -t 'ebusd/b7v/z1Timer.Sunday/get' -m '?'`
but instead I first get the correct result and then the same topic updated with the wrong one.
Even more after that I have updates for topics I didn't request:
```
ebusd/b7v/z1Timer.Sunday/get ?
ebusd/b7v/z1Timer.Sunday 07:30;22:00;-:-;-:-;-:-;-:-
ebusd/b7v/noiseReductionTimer.Saturday 07:30;23:30;-:-;-:-;-:-;-:-
ebusd/b7v/ccTimer.Friday 07:30;23:30;-:-;-:-;-:-;-:-06:00;22:00;-:-;-:-;-:-;-:-
ebusd/b7v/z1Timer.Sunday 07:30;23:30;-:-;-:-;-:-;-:-06:00;22:00;-:-;-:-;-:-;-:-07:30;22:00;-:-;-:-;-:-;-:-
```
The problem is the last update:
```
ebusd/b7v/z1Timer.Sunday 07:30;23:30;-:-;-:-;-:-;-:-06:00;22:00;-:-;-:-;-:-;-:-07:30;22:00;-:-;-:-;-:-;-:-
```
which makes it impossible to use the result as Home Assistant automation I have can't tell which result is good and which is not.
Please also note some time ranges in the wrong result are negative, e.g. `-06:00;22:00` and `-07:30;22:00`. While querying with ebusctl everything works as expected:
```
~# ebusctl r z1Timer.Sunday
07:30;22:00;-:-;-:-;-:-;-:-
```

john30, 2019-11-10:
please share the b7v configuration by the output of "ebusctl f -f -a z1Timer"

andr2000, 2019-11-10:
```
~# ebusctl f -f -a z1Timer
r,b7v,z1Timer.Friday,timer heating friday,,15,b524,0303000204,,s,IGN:1,,,whether timeslot is valid,from,s,TTM,,,slot 1 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 2 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 3 from/to,to,s,TTM,,,to
w,b7v,z1Timer.Friday,timer heating friday,,15,b524,0403000204,from,m,TTM,,,slot 1 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 2 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 3 from/to,to,m,TTM,,,to
r,b7v,z1Timer.Monday,timer heating monday,,15,b524,0303000200,,s,IGN:1,,,whether timeslot is valid,from,s,TTM,,,slot 1 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 2 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 3 from/to,to,s,TTM,,,to
w,b7v,z1Timer.Monday,timer heating monday,,15,b524,0403000200,from,m,TTM,,,slot 1 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 2 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 3 from/to,to,m,TTM,,,to
r,b7v,z1Timer.Saturday,timer heating saturday,,15,b524,0303000205,,s,IGN:1,,,whether timeslot is valid,from,s,TTM,,,slot 1 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 2 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 3 from/to,to,s,TTM,,,to
w,b7v,z1Timer.Saturday,timer heating saturday,,15,b524,0403000205,from,m,TTM,,,slot 1 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 2 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 3 from/to,to,m,TTM,,,to
r,b7v,z1Timer.Sunday,timer heating sunday,,15,b524,0303000206,,s,IGN:1,,,whether timeslot is valid,from,s,TTM,,,slot 1 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 2 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 3 from/to,to,s,TTM,,,to
w,b7v,z1Timer.Sunday,timer heating sunday,,15,b524,0403000206,from,m,TTM,,,slot 1 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 2 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 3 from/to,to,m,TTM,,,to
r,b7v,z1Timer.Thursday,timer heating thursday,,15,b524,0303000203,,s,IGN:1,,,whether timeslot is valid,from,s,TTM,,,slot 1 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 2 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 3 from/to,to,s,TTM,,,to
w,b7v,z1Timer.Thursday,timer heating thursday,,15,b524,0403000203,from,m,TTM,,,slot 1 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 2 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 3 from/to,to,m,TTM,,,to
r,b7v,z1Timer.Tuesday,timer heating tuesday,,15,b524,0303000201,,s,IGN:1,,,whether timeslot is valid,from,s,TTM,,,slot 1 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 2 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 3 from/to,to,s,TTM,,,to
w,b7v,z1Timer.Tuesday,timer heating tuesday,,15,b524,0403000201,from,m,TTM,,,slot 1 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 2 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 3 from/to,to,m,TTM,,,to
r,b7v,z1Timer.Wednesday,timer heating wednsday,,15,b524,0303000202,,s,IGN:1,,,whether timeslot is valid,from,s,TTM,,,slot 1 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 2 from/to,to,s,TTM,,,to,from,s,TTM,,,slot 3 from/to,to,s,TTM,,,to
w,b7v,z1Timer.Wednesday,timer heating wednsday,,15,b524,0403000202,from,m,TTM,,,slot 1 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 2 from/to,to,m,TTM,,,to,from,m,TTM,,,slot 3 from/to,to,m,TTM,,,to

```

john30, 2019-11-17:
I cannot reproduce this. which tool do you use for listening to MQTT topics? With mosquitto_sub I don't see such concatenation at all

closed 2019-11-17 by john30

andr2000, 2019-11-17:
mosquitto_sub version 1.4.15 running on libmosquitto 1.4.15.

andr2000, 2019-11-17:
seems it has fixed the problem, thank you
```
ebusd/b7v/z1Timer.Monday/get ?
ebusd/b7v/z1Timer.Monday 06:00;22:00;-:-;-:-;-:-;-:-
ebusd/bai/Flame off
ebusd/bai/PumpPower 0
ebusd/b7v/z1Timer.Monday 06:00;22:00;-:-;-:-;-:-;-:-

```
interestingly, the update sent twice for some reason?

john30, 2019-11-18:
the first is the direct answer to the query, the second is the change notification

andr2000, 2019-11-18:
ok, so this is by design?

john30, 2019-11-24:
yes.

commit 87b3705e (2019-11-17): fix for concatenation of several update messages (fixes #309 and fixes #312)