john30/ebusd #310: http requests with option def returns always write=false

bumaas, 2019-11-10:
Hi,

with ebusctl I can read and change values of my bai successfully. Example:

`read -c bai BlocktimeHcMax` 
returns 20

`write -c bai BlocktimeHcMax 19`
returns done

`read -c bai BlocktimeHcMax `
returns 19

But when I use the http interface to get the information whether a value is writable or not, it always reports 
"write": false

`http://192.168.178.82:8080/data/bai/BlocktimeHcMax?def`

returns

```
{
 "bai": {
  "messages": {
   "BlockTimeHcMax": {
    "name": "BlockTimeHcMax",
    "passive": false,
    "write": false,
    "lastup": 0,
    "zz": 8,
    "id": [181, 9, 13, 33, 0],
    "fielddefs": [
     { "name": "minutes0", "slave": true, "type": "UCH", "isbits": false, "length": 1, "unit": "min", "comment": "Max. Brennersperrzeit bei einem Vorlaufsollwert von 20°C"}
    ]
   }
  }
 }
```

What could be wrong in my configuration, that I always get false for writable values? Or is there any other way to find out whether a value is writable?

Burkhard

john30, 2019-11-11:
you need to query for the write messages in the HTTP request as described [in the wiki](https://github.com/john30/ebusd/wiki/3.2.-HTTP-client). Otherwise you will get read messages only.

closed 2019-11-11 by john30

bumaas, 2019-11-11:
Thank you! Now I've got it.

bumaas, 2019-11-14:
BTW: can you explain me, what 

`"passive": false/true`

means? I can't find an explanation for that point. I assume, that 'passive' messages cant be read. Is that correct?

john30, 2019-11-16:
a passive message is one that ebusd is not able to emit by itself, i.e. ebusd will only listen to those but cannot actively write these to the bus