john30/ebusd #312: MQTT 2 messages connected together

max1978, 2019-11-15:
I have a problem with MQTT. Sometimes two messages (similar address) are connected together.

I get the correct values with ebusctl find.
But under MQTT the message is sent with the message before. The Problem is only with some messages. 
If I restart and the first message is not read all is ok.

I have tried to change the names (without “-“), but it didn’t helped. 
I have the newest Version, but I have the same problem with V3.3. Also with the package and with the source compiled.
I have an OTE3 and I’m working on the config file.  

```
15,0621,7d,840002 00-071-IST-TEMP-TQE-Eintritt
15,0621,77,840008 00-002-Heizkreis-Vorlauf

ebusctl find
HZ-SOLL-IST 00-002-Heizkreis-Vorlauf         =   2;0;0d;02;1000;0;35.5
WP-SOLL-IST 00-071-IST-TEMP-TQE-Eintritt = 71;0;0d;02;1000;0;10.8
```

```
2019-11-13 23:41:24.509 [mqtt debug] publish ebusd/HZ-SOLL-IST/00-002-Heizkreis-Vorlauf {
     "group": {"value": 2},
     "value": {"value": 0},
     "type": {"value": "0d"},
     "unit": {"value": "02"},
     "max": {"value": 1000},
     "min": {"value": 0},
     "temperature": {"value": 35.5}}
2019-11-13 23:41:24.509 [mqtt debug] log Client ebusd_3.4_2575 sending PUBLISH (d0, q0, r0, m534, 'ebusd/HZ-SOLL-IST/00-002-Heizkreis-Vorlauf', ... (209 bytes))
2019-11-13 23:41:24.510 [mqtt debug] publish ebusd/WP-SOLL-IST/00-071-IST-TEMP-TQE-Eintritt {
     "group": {"value": 2},
     "value": {"value": 0},
     "type": {"value": "0d"},
     "unit": {"value": "02"},
     "max": {"value": 1000},
     "min": {"value": 0},
     "temperature": {"value": 35.5}}{
     "group": {"value": 71},
     "value": {"value": 0},
     "type": {"value": "0d"},
     "unit": {"value": "02"},
     "max": {"value": 1000},
     "min": {"value": 0},
     "temperature": {"value": 4.4}}
2019-11-13 23:41:24.510 [mqtt debug] log Client ebusd_3.4_2575 sending PUBLISH (d0, q0, r0, m535, 'ebusd/WP-SOLL-IST/00-071-IST-TEMP-TQE-Eintritt', ... (418 bytes))
```

andr2000, 2019-11-15:
@john30 is it the same I'm seeing in #309 ?

john30, 2019-11-16:
> @john30 is it the same I'm seeing in #309 ?

not sure yet, but might be the case. needs more analysis

john30, 2019-11-17:
please post your message definition, maybe it is related to that

commit 87b3705e (2019-11-17): fix for concatenation of several update messages (fixes #309 and fixes #312)
closed 2019-11-17 by john30