john30/ebusd #314: maxage is not working for http request

joergensen70, 2019-12-04:
First of all - amazing project the ebusd. It's somehow difficult to get started as beginner but finally I got everything to work :-) Great Wiki helps here.

One of the last things I'm struggling with is the 'maxage' attribute for the http request. I'm using this access on a small WebPage but unfortunately it doesn't force an bus-update.
For some reason it was working sometimes (without any changes to the web page or csv files) actually just after a Raspberry reboot.
After the reboot it worked sometime and updating my little Web Page took about 2 Minutes but after that all data was updated.
Now it stopped working again and I don't know why. I need some help here to troubleshoot.

It might be something I'm doing wrong but I don't have any idea. One interesting entry in the log file is:

2019-12-04 15:37:44.967 [bus error] prepare message part 0: ERR: end of input reached

which is coming frequently (but see some of those entries also if it was working).

Could somebody help here? Or is there any way to just empty the cache so that the daemon is forced to update from bus?

My script looks like this and is executed every 5 seconds. Of course it's too fast but I don't think this should hurt

	function getData()
	{
		var xhr = new XMLHttpRequest();
		xhr.open("GET", "/data/?maxage=0&required");
		xhr.responseType = 'json';
		xhr.send();
		xhr.onloadend = function()
		{
			var responseObj = xhr.response;
			displayData(responseObj);
		}
	}

I tried also to avoid a too fast data access by avoiding a new http request until the old sent data and the data has been processed. But no change.

I would need some help on that issue.

john30, 2019-12-09:
why are you multiplying your questions by posting on several issues? please stop that.
as commented here https://github.com/john30/ebusd/issues/233#issuecomment-563106911

closed 2019-12-09 by john30

joergensen70, 2019-12-09:
Sorry about multiplying my messages. There was not answer. After submitting my question I searched even more and found an older topic which was fitting and I added my question there again.

I’m really looking for these functions to troubleshoot my heating system. It’s not running as expected and I need most of the information in one shot.

 

In the other message you wrote that the maxage is only working for single messages. But requesting one single datapoint takes almost the same time than requesting everything. Is there any way to get all the requested data in one http request considering the maxage?

 

as commented here #233 (comment) <https://github.com/john30/ebusd/issues/233#issuecomment-563106911>

andig, 2019-12-09:
> I’m really looking for these functions to troubleshoot my heating system. It’s not running as expected and I need most of the information in one shot.

> Is there any way to get all the requested data in one http request considering the maxage?

It seems you're trying to force something that is not working. Why not keep it simple as use MQTT instead? On MQTT to get any message at the moment it appears on the bus (..and is how I am monitoring my heating).

joergensen70, 2019-12-09:
Hi Andig, I struggled so far with MQTT and therefore focused on http. Do you use the MQTT values on a WebPage and would you mind to share an example with me?

andig, 2019-12-09:
I‘m using ebusd with mqtt and node-red for logging, data does into volkszaehler (https://volkszaehler.org). My flows are at https://github.com/andig/ebusd-logging.

joergensen70, 2019-12-11:
Hi andig,
I looked into your flows but still struggling. I'm not very experienced in this topic yet. And for a newcomer it's very difficult to answer following questions:

- Is my mqtt Server on my Raspi working as expected? Didn't find any way to test it so far but I think it's running since the ebusd log says - [mqtt error] publish: The client is not currently connected.
- If I decide to use node-red - how to get started? The documentation keeps many questions open (at least for me). What are the basic steps, where to install (Raspi?). Do I need a broker?

By any chance - could you provide some basic directions at least or documented you something (like I usually do) how you achieved your results. Getting the values to Volkszähler would be very nice since I'm running it on my Raspi.
My biggest problem is that I have to find solutions fast. I have a heating system which is not working as expected and even Vaillant cannot find the issue. Having data available will help to get to the root cause.
If there is any way to help me to start I would really appreciate. A PN would be nice.

andig, 2019-12-11:
This is totally off-topic, but:

> Is my mqtt Server on my Raspi working as expected?

Try `mosquitto_pub/sub`.

> If I decide to use node-red - how to get started

Sorry, that does not belong here by far.

Your point was "problems with mqtt", now I understand you'll need to get a basic understanding of mqtt even. I'd suggest to do that and can assure you that ebusd with mqtt is working as expected for me.

joergensen70, 2019-12-12:
Ok, hopefully I'm on-topic again.... I followed andig's recommendation and tried MQTT using Node-Red.
I installed Node-Red on the Raspberry, installed the Mosquitto MQTT Broker and what should I say - it's just working fine (Andig - thanks for publishing your flows - very helpful).
Currently I'm trying to figure out how to get the additional data points which I got already through http request. I'm looking for:

- VR700\SystemFlowTemp
- vR700\Hc2HeatCurve
- VR700\Hc2MinFlowTempDesired
- VR700\Hc2MaxFlowTempDesired
- VR700\WaterPressure
- VR700\Hc2SummerTempLimit
- VR700\PrEnergySum
- VR700\YieldTotal
- VR700\PrEnergySumHc
- HMU\YieldHc
- HMU\CopHc
- HMU\YieldHwc
- HMU\CopHwc
- HMU\HoursHc
- HMU\HoursHwc
- HMU\Hours

All of these values I was able to get through http request (but unfortunately outdated)

In addition I'm actually looking for the Sole Temperature (not sure how this is called in english) but I didn't find these datapoints so far.
Is there any way to receive the data above? The csv files do support all of them except the 'Sole Temperature'

andig, 2019-12-12:
> ...and what should I say - it's just working fine

Glad you're making progress! Be sure to check the wiki (https://github.com/john30/ebusd/wiki/3.3.-MQTT-client). Note that you can append `/get` to a topic for force a bus read (I'm using that in my flows to get the values that aren't published otherwise).