john30/ebusd #315: Missing Temps after upgrade

MBj1703, 2019-12-06:
Hi,

did an upgrade today from v1.X to the actual one.
As I understand, the config files now only online?
I had the "Rücklauftemperatur" in the past, called sdtrt.
This is now missing. How can I add this back again?

Thanks
Markus

andig, 2019-12-06:
You can still use local config files or open a PR to get missing registers merged.

MBj1703, 2019-12-06:
Sorry, this is new for me, what is a PR?

MBj1703, 2019-12-06:
This maybe the missing part (of my old csv): r,,SDTRT,d.41 Rücklauftemperatur,,,,"9800",,,temp,,,,tempmirror,,UIN,,,,,,sensor,,,Rücklauftemperatur

How can I setup local config files?

dwarning, 2019-12-08:
It is not really an issue but perhaps a potential overflow problem. This I got compiling on a RPi 3 on actual git master branch:
`mainloop.cpp: In member function ‘ebusd::result_t ebusd::MainLoop::executeFind(const std::vector<std::__cxx11::basic_string<char> >&, const string&, std::ostringstream*)’:
mainloop.cpp:1464:46: warning: ‘-’ directive output may be truncated writing 1 byte into a region of size between 0 and 16 [-Wformat-truncation=]
           snprintf(str+len, sizeof(str)-len, ", lastup=%04d-%02d-%02d %02d:%02d:%02d",
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
mainloop.cpp:1464:19: note: ‘snprintf’ output between 29 and 81 bytes into a destination of size 32
           snprintf(str+len, sizeof(str)-len, ", lastup=%04d-%02d-%02d %02d:%02d:%02d",
           ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             td.tm_year+1900, td.tm_mon+1, td.tm_mday,
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             td.tm_hour, td.tm_min, td.tm_sec);
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`

john30, 2019-12-08:
a PR is a pull request here on github.
for local configs see ["-d" option in the wiki](https://github.com/john30/ebusd/wiki/2.-Run#device-options)

john30, 2019-12-08:
as written on fhem, the name of the message definition has changed, so not an issue

closed 2019-12-08 by john30