john30/ebusd #317: Where to find and edit the configuration files when using /ebusd.eu/config/

twischi, 2019-12-09:
Dear All,
with ebusd I am using the automatic configuration with getting the newest files from "ebusd.eu" what is working fine.
`--configpath=http://ebusd.eu/config/`

I want to manipulate the configuration to write values with the SetMode command (I am using no "regulator" on ebus).
But I have the simple problem: I don't find the files

`vaillant/bai.0010006101.inc`
`vaillant/08.bai.csv` 

to modify the config.

`sudo find / -name 08.bai.csv`  leads to no result

Guess they are only existing in memory, when using this auto-config?
If not where to find? 

Regards
Thomas

Maddin2020, 2019-12-09:
I think you must only use Google then you can found the files on Guthaben and you can copy this Files to your system

Maddin2020, 2019-12-09:
https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/08.bai.csv

Maddin2020, 2019-12-09:
.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/bai.0010006101.inc

Maddin2020, 2019-12-09:
https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/bai.0010006101.inc

john30, 2019-12-09:
it is [described here](https://github.com/john30/ebusd-configuration/blob/master/README.md)

domi-a, 2020-12-05:
@twischi did you manage to manipulate a config file to write with SetMode.
it is exactly what i wanna do.
you downloaded the configuration files and start the deamon with --configpath=<path to your edited files> is that all?

twischi, 2020-12-07:
@domi-a: sorry, no for me it never worked and than i lost track.
Guess in my system the problem COULD be that I took the external regulator "calorMatic 470f" out of my system. Maybe only this device would have been able to take write commands. But I am not sure.

My Vaillant System: VSC 126/3-5 140-HL
not used external regulator @ebus  calorMatic 470f

john30, 2021-01-10:
if you cloned the files to /etc/config, then simply edit /etc/config/vaillant/hcmode.inc in line 10 and replace the "uw" at the beginning of the line with a "w". Then SetMode will be writable after you have restarted ebusd.

closed 2021-01-10 by john30