john30/ebusd #318: Is it possible to switch on heating using ebus

Dinth, 2019-12-11:
Is it possible to do that through ebus in similar fashion as with external thermostat wired up to the boiler?
Im controlling my boiler using VRC470 - and while the single thermostat works perfect for all my bedrooms, my living room and kitchen are cold so i would like to use Home Assistant and a wireless thermostat to keep them warm too.

andig, 2019-12-13:
Not sure what you need? You can control `SetMode` for example.

Dinth, 2019-12-13:
Sorry, i was not precise. So basically every boiler has an external thermostat physical contacts. Depending on the model, if those are shorted or theres a voltage on them, the boiler goes into "heat required" mode, basically switching on the flame and constantly heating the water in heating circuit, regardless of normal/builtin heating controls. 
Is there any way of emulating this behavior through ebus interface, for example by setting a value which makes boiler think that there's a need for heating?

andig, 2019-12-13:
Nit that I know, but whats wrong with SetMode.

Dinth, 2019-12-14:
My existing VRC would constantly overwrite it.

Erbit, 2019-12-14:
Hi
You want to turn on the boiler when VRC470 turns it off. This is a bad approach. You must change the logic. You must turn off the boiler when VRC470 wants to continue heating. This is correct logic.

How to do it?
1) find in the manual a place to connect a regular thermostat (on/off). Originally a jumper is installed.
2) remove the jumper and connect the relay in its place
3) increase the tempeaura a bit in VRC470 so that it does not turn off the boiler too soon
4) when in all rooms you get the desired temperature turn off the relay. The boiler will turn off central heating but will not turn off water heating.

andig, 2019-12-15:
Afaik the room controller should be placed in the room with the highest warmth demand- so it belongs into the living room or kitchen. If bedrooms become too warm this should be fixed by either room thermostats or hydraulic leveling. Anything else is imho inefficient treating of symptoms rather than fixing the physical problem.

Erbit, 2019-12-15:
If you refer to my answer then I agree that there should be room thermostats. In point 4 I assumed that room thermostats were installed.

Dinth, 2019-12-15:
@andig thanks for the help, maybe i will be forced to experiment with it.
Although Im not sure if that's going to work, because while my bedrooms have regular radiators, my ground floor has mix of underfloor heating as a main way of heating and a fan radiator if the temperature needs to be raised quickly.

Dinth, 2019-12-29:
Actually i found out that Vailant VR81/2 thermostat should be doing exactly what i want to do, its just a matter of emulating it. Unfortunately, VR81/2 is wired so it wont work in my setup.

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30