john30/ebusd #319: edusd service with MQQT settings is not starting...

darekhome, 2019-12-13:
My broker is installed on RPI 192.168.1.35
ebus board (via Wemos is on 192.168.135).
Ebusd config: 

`EBUSD_OPTS="--scanconfig -d 192.168.1.135:9999 --httpport=801 --mqtthost=192.168.1.35 --mqttport=1883 --mqttjson --accesslevel="*" --latency=20000" `

And this above settings is causing error:

`pi@raspberrypi:~ $ sudo service ebusd start
Job for ebusd.service failed because the control process exited with error code.
See "systemctl status ebusd.service" and "journalctl -xe" for details.
pi@raspberrypi:~ $ systemctl status ebusd.service
● ebusd.service - ebusd, the daemon for communication with eBUS heating systems.
   Loaded: loaded (/etc/systemd/system/ebusd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2019-12-13 16:45:27 CET; 24s ago
  Process: 10376 ExecStart=/usr/bin/ebusd $EBUSD_OPTS (code=exited, status=64)
`

But without MQQT setting ebusd service is starting:

`EBUSD_OPTS="--scanconfig -d 192.168.1.135:9999 --httpport=801 --accesslevel="*" --latency=20000"`

Not sure what is wrong. I tryied with 2 different MQQT brokers - result still the same...

andig, 2019-12-13:
What is the error message from the journal? You‘ve missed the interesting part ;)

darekhome, 2019-12-13:
Yes, sorry. See below:

`pi@raspberrypi:~ $ journalctl -xe
--
-- A stop job for unit ebusd.service has finished.
--
-- The job identifier is 153638 and the job result is done.
Dec 13 17:40:07 raspberrypi sudo[10675]: pam_unix(sudo:session): session closed for user root
Dec 13 17:40:09 raspberrypi kernel: Voltage normalised (0x00000000)
Dec 13 17:40:11 raspberrypi sudo[10707]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/sbin/service ebusd start
Dec 13 17:40:11 raspberrypi sudo[10707]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Dec 13 17:40:14 raspberrypi systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
-- Subject: A start job for unit ebusd.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit ebusd.service has begun execution.
--
-- The job identifier is 153639.
Dec 13 17:40:14 raspberrypi ebusd[10735]: /usr/bin/ebusd: unrecognized option '--mqtthost=192.168.1.35'
Dec 13 17:40:14 raspberrypi ebusd[10735]: Try `ebusd --help' or `ebusd --usage' for more information.
Dec 13 17:40:14 raspberrypi systemd[1]: ebusd.service: Control process exited, code=exited, status=64/USAGE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit ebusd.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 64.
Dec 13 17:40:14 raspberrypi systemd[1]: ebusd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit ebusd.service has entered the 'failed' state with result 'exit-code'.
Dec 13 17:40:14 raspberrypi systemd[1]: Failed to start ebusd, the daemon for communication with eBUS heating systems..
-- Subject: A start job for unit ebusd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit ebusd.service has finished with a failure.
--
-- The job identifier is 153639 and the job result is failed.
Dec 13 17:40:14 raspberrypi sudo[10707]: pam_unix(sudo:session): session closed for user root
Dec 13 17:40:17 raspberrypi kernel: Voltage normalised (0x00000000)
lines 4224-4265/4265 (END)
`

andig, 2019-12-13:
> unrecognized option '--mqtthost=192.168.1.35'

darekhome, 2019-12-14:
Ah, I was blind. Now I see my error.
Should install ‘mqqt’ version...
Thx for pointing...

closed 2019-12-14 by darekhome