john30/ebusd #320: ERR: SYN received

PureDJ, 2019-12-16:
I have a problem, ebusd is running on an raspberry pi, with a DIY converter.
When I try to read any parameter I get "ERR: element not found"
When I scan a slave on the bus I get "ERR: SYN received"
I checked the hardware with a PSU and 330 ohm in series, when I send data, I see this on the scope, so it is working (voltages are correct)
This is the ebusctl info
version: ebusd 3.4.v3.4-5-gf152798
update check: revision v3.4 available
signal: acquired
symbol rate: 26
max symbol rate: 92
min arbitration micros: 26
max arbitration micros: 46
min symbol latency: 26
max symbol latency: 28
reconnects: 0
masters: 4
messages: 14
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 7f: master #24

If I run it for a long time, it will find more in the bus and fills in the correct brand (Valliant) 

What could be the issue?

PureDJ, 2019-12-17:
a little update, now I have one address scanned (happened over night)
address 03: master #11
address 08: slave #11
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=36000;SW=0139;HW=7301"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 7f: master #24

If I let it run for a couple of days, #11 will be scanned also, but I can't get anything out of it, every time I get "ERR: element not found"

andig, 2019-12-17:
Apparently your connection may be faulty?

john30, 2019-12-17:
what serial converter are you using?

PureDJ, 2019-12-17:
I use the raspberry pi's uart
Pi is configured right (I think) when I send some characters over the serial port I see them on the oscilloscope
I have added some extra info to my previous post, the time between SYN packages seems to be very short (40 to 45ms)

andig, 2019-12-17:
Before investigating on the ebusd side I'd suggest to verify behaviour with a proven adapter.

PureDJ, 2019-12-17:
> 
> 
> Before investigating on the ebusd side I'd suggest to verify behaviour with a proven adapter.

What should I check more?
It is receiving the data, and with a power supply connected with 330ohm in series I see the TX'ed data on the scope, with the voltages within spec, high =20V (psu set to 20V), low = ~11V

andig, 2019-12-17:
> What should I check more?

It would help to check that your DIY converter fully works by a/b testing with a proven adapter.

PureDJ, 2019-12-17:
I don't have another adapter, it is build from a proven schematic (https://ebus.github.io/adapter/index.en.html) and in the test setup it works as it suppose to do.

Before I spend money on buying another adapter I would run some test first, I'm not so sure another adapter will solve this, as with this adapter I get "signal: acquired" and TX is also within specs.
There is so much more that could be a potential issue.

My question marks are now at the Raspberry pi (is the serial port suitable for this low latency bus?) and with the bus, it every 40 ms a SYN byte normal for a Ebus (if so, the Pi needs to send it commands within the 40 ms, actually I would say within 30ms to be safe, is the Pi indeed sending it's requests within this time frame)

PureDJ, 2019-12-17:
I managed to solve the problem, it was not the converter, but the pi itself.
Because the pi uses a FIFO that could not be disabled (normally), and cannot be set to 1 (lowest is 4) it's always to late when it actually processes a SYN package.
I found another project to disable the FIFO for the raspberry Pi (https://github.com/eBUS/ttyebus), this solved my problem.
So for anyone who wants to use ebusd on a Raspberry Pi and use the Pi's serial port (/dev/ttyAMA0). install the ttyebus device, otherwise you won't get you ebus converter to spec.
@john30 maybe you can make a note on github about this?

john30, 2019-12-25:
it is already [mentioned here](https://github.com/john30/ebusd/wiki/6.-Hardware#important-note-about-uarts)

closed 2019-12-25 by john30