john30/ebusd #321: MQTT get is not coming through / node red on Homematic CCU3, ebusd on Raspberry

joergensen70, 2019-12-18:
After making much much progress on my Raspberry (Node Red is just running fine and polling almost all data I need from my Vaillant flexotherm and send parts of this data to my Volkszähler) I need some help again.
I hope I'm not off-topic again because I don't know exactly where the problem is.

As I said. Everything on the Raspberry is running just fine but I have seen some performance issues with my Node Red and displaying the data.
Therefore I moved my Node Red installation to an Homematic CCU3 (original, not Raspberry). Almost everything is running just fine again BUT:
The 'get' requests don't come through anymore. I kept the Mosquitto Broker and ebusd on the Raspberry just moved the poll command to the CCU3.

Before I have seen many reads in the ebusd.log, after migrating these reads are gone.
I still can receive the broadcast data and the ebusd get function is showing connected.

In the ebusd get flow I just changed localhost to my Raspberry's IP, same Port 1883

Do I miss something? The architecture of the ebusd / broker and the data flow is still not 100% clear to me (still in learning process)

Any idea how to troubleshoot or guidance what to do next would be helpful and appreciated.

Current setup:
- RaspberryPi with ebusd and Mosquito Broker (port 1883)
- CCU3 with Redmatic and using the ebusd get node to poll data
- ebusd get node configured with IP of Raspi, Port and user credentials (not sure if required)
- CCU3 currently configured that all ports are open
- Receiving ebusd broadcast data but cannot poll required data points
- Communication to Volkszähler on Raspberry still active and works just fine

andig, 2019-12-18:
> I hope I'm not off-topic again because I don't know exactly where the problem is.

It's really not an ebus topic imho.

> The 'get' requests don't come through anymore. I kept the Mosquitto Broker and ebusd on the Raspberry just moved the poll command to the CCU3.

For sake of resolution: unclear what you're asking. You can always use e.g. `mosquitto_pub/sub` and/or `curl` to try without complex layers on top.

john30, 2019-12-25:
agreed, it is not really clear to me either what the question is. sounds like connectivity problems between the components to me. you just need to debug these i guess

closed 2019-12-25 by john30