john30/ebusd #323: unable to load scan config 15: no file from vaillant with prefix 15 matches ID "scc00", SW2100, HW3903

dirkdecoppel, 2019-12-27:
Hello,
I have connected a new thermostat type Bulex Migo to the ebus of my Bulec heating.
Below some logging that shows that the thermostat not correctly identified.
I observe also some other commands that are not recognized.
What can i do to resolve this?

Dirk
2019-12-26 21:10:40.054 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with auto scan
2019-12-26 21:10:40.387 [bus notice] bus started with own address 31/36
2019-12-26 21:10:40.411 [bus notice] signal acquired
2019-12-26 21:10:41.499 [bus notice] new master 10, master count 2
2019-12-26 21:10:41.543 [bus notice] new master 03, master count 3
2019-12-26 21:10:41.544 [update notice] received unknown MS cmd: 1008b5100900000078ffff01ff00 / 0101
2019-12-26 21:10:49.261 [update notice] received unknown MS cmd: 1008b503020001 / 0affffffffffffffffffff
2019-12-26 21:10:49.540 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:10:50.030 [update notice] received unknown MS cmd: 1008b503020004 / 00
2019-12-26 21:10:50.298 [update notice] received unknown MS cmd: 1008b5110101 / 098f8e0080ffff0000ff
2019-12-26 21:10:50.611 [bus notice] scan 08: ;Vaillant;BAI00;0609;3302
2019-12-26 21:10:50.611 [update notice] store 08 ident: done
2019-12-26 21:10:50.611 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0609;3302
2019-12-26 21:10:50.612 [bus notice] scan 08: ;Vaillant;BAI00;0609;3302
2019-12-26 21:10:50.959 [main notice] read common config file vaillant/scan.csv
2019-12-26 21:10:51.036 [main notice] read common config file vaillant/general.csv
2019-12-26 21:10:51.115 [main notice] read common config file vaillant/broadcast.csv
2019-12-26 21:10:51.198 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0609, HW3302
2019-12-26 21:10:51.432 [update notice] sent scan-read scan.08 id QQ=31: 
2019-12-26 21:10:51.593 [update notice] received unknown MS cmd: 1008b5100900000078ffff01ff00 / 0101
2019-12-26 21:10:51.875 [update notice] sent scan-read scan.08 id QQ=31: 
2019-12-26 21:10:52.134 [update notice] sent scan-read scan.08 id QQ=31: 
2019-12-26 21:10:52.393 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-12-26 21:10:52.777 [main notice] found messages: 210 (3 conditional on 25 conditions, 0 poll, 9 update)
2019-12-26 21:10:53.000 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-12-26 21:10:53.260 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-12-26 21:10:53.519 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-12-26 21:10:53.778 [update notice] sent scan-read scan.08 id QQ=31: 21;19;06;0010021855;1610;005060;N9
2019-12-26 21:10:53.779 [bus notice] scan 08: ;21;19;06;0010021855;1610;005060;N9
2019-12-26 21:10:56.004 [bus notice] scan 15: ;Vaillant;SCC00;2100;3903
2019-12-26 21:10:56.005 [update notice] store 15 ident: done
2019-12-26 21:10:56.005 [update notice] sent scan-read scan.15  QQ=31: Vaillant;SCC00;2100;3903
2019-12-26 21:10:56.005 [bus notice] scan 15: ;Vaillant;SCC00;2100;3903
2019-12-26 21:10:56.263 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231313931313030
2019-12-26 21:10:56.527 [update notice] sent scan-read scan.15 id QQ=31: 
2019-12-26 21:10:56.785 [update notice] sent scan-read scan.15 id QQ=31: 
2019-12-26 21:10:57.045 [update notice] sent scan-read scan.15 id QQ=31: 21;19;11;0020197230;0932;005518;N0
2019-12-26 21:10:57.046 [bus notice] scan 15: ;21;19;11;0020197230;0932;005518;N0
2019-12-26 21:10:57.137 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "scc00", SW2100, HW3903
2019-12-26 21:10:57.137 [main error] scan config 15: ERR: element not found
2019-12-26 21:10:58.870 [update notice] received read bai Status02 QQ=10: off;20;75.0;76;60.0
2019-12-26 21:10:59.264 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:10:59.542 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:11:00.303 [update notice] received read bai Status01 QQ=10: 56.0;70.5;-;-;-;off
2019-12-26 21:11:01.606 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2019-12-26 21:11:09.307 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:11:09.584 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:11:09.836 [update notice] received unknown MS cmd: 1008b503020003 / 00
2019-12-26 21:11:10.070 [update notice] received unknown MS cmd: 1008b503020004 / 00
2019-12-26 21:11:10.342 [update notice] received read bai Status01 QQ=10: 53.0;70.0;-;-;-;off
2019-12-26 21:11:19.369 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:11:19.646 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:11:20.409 [update notice] received read bai Status01 QQ=10: 53.0;70.0;-;-;-;off
2019-12-26 21:11:28.954 [update notice] received read bai Status02 QQ=10: off;20;75.0;76;60.0
2019-12-26 21:11:29.479 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:11:29.756 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:11:30.242 [update notice] received unknown MS cmd: 1008b503020004 / 00
2019-12-26 21:11:30.513 [update notice] received read bai Status01 QQ=10: 54.0;70.0;-;-;-;off
2019-12-26 21:11:39.553 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:11:39.827 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:11:40.313 [update notice] received unknown MS cmd: 1008b503020004 / 00
2019-12-26 21:11:40.584 [update notice] received read bai Status01 QQ=10: 56.0;69.5;-;-;-;off
2019-12-26 21:11:49.591 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
[... 18 lines cut ...]
2019-12-26 21:12:30.109 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:12:30.871 [update notice] received read bai Status01 QQ=10: 62.0;69.0;-;-;-;off
2019-12-26 21:12:39.891 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:12:40.167 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:12:40.928 [update notice] received read bai Status01 QQ=10: 62.5;68.5;-;-;-;off
2019-12-26 21:12:49.952 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:12:50.229 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:12:50.987 [update notice] received read bai Status01 QQ=10: 63.0;68.5;-;-;-;off
2019-12-26 21:12:59.213 [update notice] received read bai Status02 QQ=10: off;20;75.0;76;60.0
2019-12-26 21:12:59.943 [main notice] update check: revision v3.4 available
2019-12-26 21:13:00.038 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:13:00.316 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:13:00.566 [update notice] received unknown MS cmd: 1008b503020003 / 00
2019-12-26 21:13:01.073 [update notice] received read bai Status01 QQ=10: 63.0;68.5;-;-;-;off
2019-12-26 21:13:10.096 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:13:10.374 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:13:10.625 [update notice] received unknown MS cmd: 1008b503020003 / 00
2019-12-26 21:13:11.134 [update notice] received read bai Status01 QQ=10: 63.0;68.5;-;-;-;off
2019-12-26 21:13:12.354 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2019-12-26 21:13:20.159 [update notice] received read bai currenterror QQ=10: -;-;-;-;-
2019-12-26 21:13:20.435 [update notice] received unknown MS cmd: 1008b503020002 / 0affffffffffffffffffff
2019-12-26 21:13:20.923 [update notice] received unknown MS cmd: 1008b503020004 / 00
2019-12-26 21:13:21.192 [update notice] received read bai Status01 QQ=10: 63.0;68.0;-;-;-;off
2019-12-26 21:13:22.412 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
[ebusd_2019-12-27_10h50.log](https://github.com/john30/ebusd/files/4005621/ebusd_2019-12-27_10h50.log)

john30, 2019-12-30:
build you own config file as described in the wiki or maybe try to "borrow" from an existing one

dirkdecoppel, 2019-12-31:
Ok; i’ll try to build one;
With your experience: from who will you start to “borrow”?
How to get the list of all possible commands?
Is there somebody you know that has knowledge of the available commands for this Bulex MiGo wifi thermostat? Looks the same as the Vaillant Vsmart wifi thermostat.

john30, 2020-01-03:
I'd start with one of the 15.f ones, like [this](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.f47.csv)

john30, 2021-10-06:
closed due to inactivity