john30/ebusd #325: Cannot write message values by name if name is too long

guillaumezin, 2020-01-01:
Hello,

Happy new year :-)

I noticed that some commands can only be passed using numeric value, not text value, if text is too long, for instance:

`localhost: write -c f47 hc1SFmode none 
done

localhost: write -c f47 hc1SFmode party
done

localhost: write -c f47 hc1SFmode onedaybankholiday
ERR: invalid numeric argument

localhost: write -c f47 hc1SFmode onedayholiday
done

localhost: write -c f47 hc1SFmode holiday
done

localhost: write -c f47 hc1SFmode bankholiday
done

localhost: write -c f47 hc1SFmode quickveto
done

localhost: write -c f47 hc1SFmode onetimeventilation
ERR: invalid numeric argument

localhost: write -c f47 hc1SFmode onetimewater
done

localhost: write -c f47 hc1SFmode 2
done

localhost: write -c f47 hc1SFmode 7
done`

Is it a parsing problem?

Thank you.

guillaumezin, 2020-01-02:
Seems like the problem lies in data.cpp, in the first reinterpret_cast of ValueListDataField::writeSymbols, I'm trying to figure out why...

guillaumezin, 2020-01-02:
Based on this article: https://stackoverflow.com/questions/21034834/is-there-issue-will-stringstream-str-c-str it seems that it's not a good idea to make a c_str on a str() from a stringstream

If I do this
```
auto istr = input->str();
const char* str = istr.c_str();
```
it works.

I didn't analyse the whole code, maybe there is the same error elsewhere.

john30, 2020-01-11:
thanks for spotting this @guillaumezin !

commit 3b569a01 (2020-01-11): fix for bad string.c_str() usage on temporary strings (seems to be problematic starting with gcc>=6.3.0, fixes #325 and #326)
closed 2020-01-11 by john30