john30/ebusd #329: HolidayPeriods not working on Vaillant geoTHERM VWL 81_3 S

stephanritscher, 2020-01-12:
Hi,

I use ebusd 3.3 and HolidayPeriods is not working properly.

Setting on device:
```
1st holiday period: 01.02.17 - 03.04.18
2nd holiday period: 05.06.10 - 07.08.11
```
Result of: ebusctl read -f HolidayPeriods
```
01.02.2017;-.05.2006;03.04.2018;-.07.2008
```
Output of ebusd --lograwdata
```
[bus notice] >3115b509030d4300fd<000c010211ff0506030412ff070886>00
```

After changing settings on device to:
```
1st holiday period: 01.02.17 - 03.04.18
2nd holiday period: 05.06.13 - 07.08.15
```
the read command and raw log contain (exactly the same).
```
01.02.2017;-.05.2006;03.04.2018;-.07.2008
[bus notice] >3115b509030d4300fd<000c010211ff0506030412ff070886>00
```
The software versions reported by the device are:
```
i/o card 1: 4.19
User Interface 1: 3.74
OMU1 1: 1.18
```

To  me it seems that the Vaillant device has a bug regarding the holiday periods.
Does anyone have experience with this? Is there a way to update the device software?

Best regards
Stephan

john30, 2020-02-02:
you basically answered yourself: the message seems not to be related to the setting on the device. you could use the grab feature of ebusd to do a list before making a change in the device and after and then comparing the two lists. this way you should be able to identify the right message

closed 2020-04-11 by john30

stephanritscher, 2023-01-06:
Sorry for the late follow-up.
My theory is that the HolidayPeriods command is correctly implemented in ebusd but has a pointer one-byte off bug in the device firmware.
As seen above, hfrom1 and hto1 (first holiday period) are read correctly.
Looking more detailled at the data string (`0c010211ff0506030412ff0708`), one can see:
`0c`: length of response
`010211`: hfrom1 (01.02.17 written in hex)
`ff`: some byte unrelated to the holiday periods
`0506`: day and month of hfrom2 (05.06.; byte for year is missing in the response as already my first experiment showed)
`030412`: hto1 (03.04.18)
`ff`: some byte unrelated to the holiday periods
`0708`: day and month of hfrom2 (07.08.;  byte for year is missing)
I was able to confirm this with additional checks this week (the unrelated byte now was different, but fixed over my experiments). Also I tried the grab approach you mentioned but found no broadcast related to changing the holiday periods in the device and no additional messages sent as response to the HolidayPeriods read command.
Basically this means, one can use only the first holiday period via ebus and should at least preserve the unrelated bytes when writing (best not to change hfrom2 and hto2 at all).

stephanritscher, 2023-01-06:
Finally, it seems that - at least for my device, the order of the fields in the message definition is switched:
```
ebusctl read -m 0 -v -c uih HolidayPeriods
> uih HolidayPeriods hfrom=03.01.2023;hto=04.01.2001;hfrom=05.01.2023;hto=04.01.2001
```
Current settings:
* Holiday period 1: from 03.01.2023 to 05.01.2023
* Holiday period 2: from 01.01.2020 to 01.01.2020

So in `15.uih.csv` the line
```
r;w,,HolidayPeriods,HolidayPeriods,,,,"4300",,,hfrom;hto;hfrom;hto,,,Holiday periods
```
should be changed to
```
r;w,,HolidayPeriods,HolidayPeriods,,,,"4300",,,hfrom;hfrom;hto;hto,,,Holiday periods
```
Can anyone confirm this?