john30/ebusd #330: Vaillant VR70 S7 Solarpump

MatthiasKunze, 2020-01-16:
Hello, the pump for the solar system is connected to connection R1 / S7 on my Vaillant VR70.
With ebusd I get the following data for sensor 7:

19 50 00 | 15.01.2020 12:54
2a 50 00 | 15.01.2020 12:56
24 50 00 | 15.01.2020 12:58
19 50 00 | 15.01.2020 13:00
1b 50 00 | 15.01.2020 13:02
21 50 00 | 15.01.2020 13:04
1c 50 00 | 15.01.2020 13:06
17 50 00 | 15.01.2020 13:08
19 50 00 | 15.01.2020 13:10
19 50 00 | 15.01.2020 13:16
21 50 00 | 15.01.2020 13:18
25 50 00 | 15.01.2020 13:20
26 50 00 | 15.01.2020 13:22
1c 50 00 | 15.01.2020 13:24
1a 50 00 | 15.01.2020 13:26
18 50 00 | 15.01.2020 13:28
1b 50 00 | 15.01.2020 13:30
29 50 00 | 15.01.2020 13:32
25 50 00 | 15.01.2020 13:34
1d 50 00 | 15.01.2020 13:36
18 50 00 | 15.01.2020 13:38
20 50 00 | 15.01.2020 13:40
29 50 00 | 15.01.2020 13:42
25 50 00 | 15.01.2020 13:44
18 50 00 | 15.01.2020 13:46
18 50 00 | 15.01.2020 13:48
1f 50 00 | 15.01.2020 13:50
28 50 00 | 15.01.2020 13:52
21 50 00 | 15.01.2020 13:54
18 50 00 | 15.01.2020 13:56
17 50 00 | 15.01.2020 13:58
c8 50 00 | 15.01.2020 14:00
c8 50 00 | 15.01.2020 14:04
1b 50 00 | 15.01.2020 14:08
1b 50 00 | 15.01.2020 14:10
18 50 00 | 15.01.2020 14:12
1e 50 00 | 15.01.2020 14:14
19 50 00 | 15.01.2020 14:16
19 50 00 | 15.01.2020 14:18
18 50 00 | 15.01.2020 14:20
18 50 00 | 15.01.2020 14:22
c8 50 00 | 15.01.2020 14:24
c8 50 00 | 15.01.2020 15:20
2b 50 00 | 15.01.2020 15:22
19 50 00 | 15.01.2020 15:24
c8 50 00 | 15.01.2020 15:26
c8 50 00 | 15.01.2020 15:40
18 50 00 | 15.01.2020 15:42
c8 50 00 | 15.01.2020 15:44
c8 50 00 | 15.01.2020 20:36

I think the data shows the flow of the solar system in liters per minute.
c8 50 00 is the value when the solar pump is off.
Do you have an idea how the data can be converted into liters per minute?

john30, 2020-02-02:
what are reasonable values corresponding to those hex values?

MatthiasKunze, 2020-03-02:
Useful values would be 0 to 6 liters.
I currently solved it like this:
S7, UCH, 200 = 0 ,,
I then divide the values still displayed by 10.

john30, 2020-04-11:
can you check what ebusctl grab decode show for the message, especially in the lines with the EXP data type?

MatthiasKunze, 2020-05-28:
How can I check that exactly?

john30, 2020-06-01:
`ebusctl grab result decode`

MatthiasKunze, 2020-12-10:
Sorry for the late reply, I forgot.

`pi@ioBroker:~ $ ebusctl grab result decode
1008b5110100 / 0888010d0000000001 = 1040
 BCD   88=88, 01=1, 00=0, 00=0, 00=0, 00=0, 01=1
 BDY   01=Tue, 00=Mon, 00=Mon, 00=Mon, 00=Mon, 01=Tue
 BTI   000000="00:00:00", 000000="00:00:00", 000001="01:00:00"
 BTM   0000="00:00", 0000="00:00", 0000="00:00", 0001="01:00"
 D1B   88=-120, 01=1, 0d=13, 00=0, 00=0, 00=0, 00=0, 01=1
 D1C   88=68.0, 01=0.5, 0d=6.5, 00=0.0, 00=0.0, 00=0.0, 00=0.0, 01=0.5
 D2B   8801=1.531, 010d=13.004, 0d00=0.051, 0000=0.000, 0000=0.000, 0000=0.000, 0001=1.000
 D2C   8801=24.50, 010d=208.06, 0d00=0.81, 0000=0.00, 0000=0.00, 0000=0.00, 0001=16.00
 DAY   8801="28.01.1901", 010d="12.02.1909", 0d00="16.01.1900", 0000="03.01.1900", 0000="03.01.1900", 0000="03.01.1900", 0001="14.09.1900"
 EXP   88010d00=1.19441e-39, 010d0000=4.66492e-42, 0d000000=1.82169e-44, 00000000=0.0, 00000001=2.35099e-38
 EXR   88010d00=-3.88348e-34, 010d0000=2.58976e-38, 0d000000=3.9443e-31, 00000000=0.0, 00000001=1.4013e-45
 FLR   8801=-30.719, 010d=0.269, 0d00=3.328, 0000=0.000, 0000=0.000, 0000=0.000, 0001=0.001
 FLT   8801=0.392, 010d=3.329, 0d00=0.013, 0000=0.000, 0000=0.000, 0000=0.000, 0001=0.256
 HCD   010d0000=1301, 0d000000=13, 00000000=0, 00000001=1000000
 HDY   01=Wed, 00=Tue, 00=Tue, 00=Tue, 00=Tue, 01=Wed
 HEX:8 88010d0000000001="88 01 0d 00 00 00 00 01"
 HTI   010d00="01:13:00", 0d0000="13:00:00", 000000="00:00:00", 000000="00:00:00", 000001="00:00:01"
 HTM   010d="01:13", 0d00="13:00", 0000="00:00", 0000="00:00", 0000="00:00", 0001="00:01"
 MIN   8801="06:32", 0d00="00:13", 0000="00:00", 0000="00:00", 0000="00:00", 0001="04:16"
 NTS:8 88010d0000000001="?"
 PIN   8801=8801, 0000=0000, 0000=0000, 0000=0000, 0001=0001
 S3N   88010d=852360, 010d00=3329, 0d0000=13, 000000=0, 000000=0, 000001=65536
 S3R   88010d=-7864051, 010d00=68864, 0d0000=851968, 000000=0, 000000=0, 000001=1
 SCH   88=-120, 01=1, 0d=13, 00=0, 00=0, 00=0, 00=0, 01=1
 SIN   8801=392, 010d=3329, 0d00=13, 0000=0, 0000=0, 0000=0, 0001=256
 SIR   8801=-30719, 010d=269, 0d00=3328, 0000=0, 0000=0, 0000=0, 0001=1
 SLG   88010d00=852360, 010d0000=3329, 0d000000=13, 00000000=0, 00000001=16777216
 SLR   88010d00=-2013197056, 010d0000=17629184, 0d000000=218103808, 00000000=0, 00000001=1
 STR:8 88010d0000000001="?  "
 TEM_P 8801=03-008, 010d=26-001, 0d00=00-013, 0000=00-000, 0000=00-000, 0000=00-000, 0001=02-000
 TTM   88="22:40", 01="00:10", 0d="02:10", 00="00:00", 00="00:00", 00="00:00", 00="00:00", 01="00:10"
 U3N   88010d=852360, 010d00=3329, 0d0000=13, 000000=0, 000000=0, 000001=65536
 U3R   88010d=8913165, 010d00=68864, 0d0000=851968, 000000=0, 000000=0, 000001=1
 UCH   88=136, 01=1, 0d=13, 00=0, 00=0, 00=0, 00=0, 01=1
 UIN   8801=392, 010d=3329, 0d00=13, 0000=0, 0000=0, 0000=0, 0001=256
 UIR   8801=34817, 010d=269, 0d00=3328, 0000=0, 0000=0, 0000=0, 0001=1
 ULG   88010d00=852360, 010d0000=3329, 0d000000=13, 00000000=0, 00000001=16777216
 ULR   88010d00=2281770240, 010d0000=17629184, 0d000000=218103808, 00000000=0, 00000001=1
 VTI   010d00="00:13:01", 0d0000="00:00:13", 000000="00:00:00", 000000="00:00:00", 000001="01:00:00"
 VTM   010d="13:01", 0d00="00:13", 0000="00:00", 0000="00:00", 0000="00:00", 0001="01:00"
10feb510020600 = 204
 BCD   06=6, 00=0
 BDA   0600="09.01.1900"
 BDY   06=Sun, 00=Mon
 BTI   0600="00:06"
 BTM   0600="00:06"
 D1B   06=6, 00=0
 D1C   06=3.0, 00=0.0
 D2B   0600=0.023
 D2C   0600=0.38
 DAY   0600="09.01.1900"
 EXP   0600=8.40779e-45
 EXR   0600=2.15239e-42
 FLR   0600=1.536
 FLT   0600=0.006
 HCD   0600=6
 HDA   0600="09.01.1900"
 HDY   06=, 00=Tue
 HEX:2 0600="06 00"
 HTI   0600="06:00"
 HTM   0600="06:00"
 MIN   0600="00:06"
 NTS:2 0600=""
 PIN   0600=0600
 S3N   0600=6
 S3R   0600=1536
[... 478 lines cut ...]
 HTI   01="00:01"
 HTM   01="00:01"
 MIN   01=""
 NTS:1 01=""
 PIN   01=01
 S3N   01=1
 S3R   01=1
 SCH   01=1
 SIN   01=1
 SIR   01=1
 SLG   01=1
 SLR   01=1
 STR:1 01=" "
 TEM_P 01=00-001
 TTM   01="00:10"
 U3N   01=1
 U3R   01=1
 UCH   01=1
 UIN   01=1
 UIR   01=1
 ULG   01=1
 ULR   01=1
 VTI   01="00:01"
 VTM   01="00:01"
`

john30, 2021-01-10:
how did you get the "c8 50 00" values from your initial post? i.e. is this related to a specific message?

MatthiasKunze, 2021-01-12:
I received the message "c8 50 00" after activating the sensor value S7 in the file 52.vr_70.csv.

Original
`r,,SensorData,,,,,"03",S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,ignore,,HEX:3,,,`

Changed
`r,,SensorData,,,,,"03",S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,S7,,HEX:3,,,`

john30, 2021-02-20:
I doubt that the value will consist of 3 bytes. Even if its just 2 bytes, the value range would be 5017 to 50c8 in hex, so 20503-20680 in dec, which is way outside of something in the range 0-6 liters no matter which divider you use.
So I think it is just the first byte which is relevant and with a range of 0x17 to 0xc8 = 23 - 200 in dec you could divide that by 33 getting a final value of 0.7 to 6.0. But a divider of 33 is very unlikely, so I guess it could be 32, which would result in a range of 0.72 -  6.25. Does that sound right?

MatthiasKunze, 2021-02-20:
Yes that could be true. How can I implement the settings?

john30, 2021-02-21:
just use UCH as datatype and put 32 in the divisor column behind

brezelmann01, 2021-05-13:
@MatthiasKunze did you get it work? I tried it like John said. If the pump isn't running, 12.5l is displayed.

brezelmann01, 2021-07-29:
Please, can anybody help me with my problem?

>  I tried it like John said. If the pump isn't running, 12.5l is displayed.

brezelmann01, 2021-10-20:
Hi John,
I tried exactly the way you said. If I use 32 (16) as a divider, then I get the value 6.25 (12.5) when the pump is stopped. The more the pump is controlled, the smaller the valu
[image]
e.

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30