john30/ebusd #333: Different configpath

tttomson, 2020-02-02:
Hi all,

i am still searching the problem with my config files.
In the meanwhile something came up i do absolutely not understand:

When you look on the following screenshots:
It shows that ebusd looks for the config file under //ebusd.eu/config/ and sometimes under /etc/ebusd/config.

The second path is the one i specified in /etc/default/ebusd

For me is unclear which one is taken now??

Can anybody help?

root@eBusd:~# ebusd --loglevel debug --checkconfig --scanconfig
2020-02-02 12:51:44.685 [main notice] ebusd 3.4.v3.3-51-g57eae05 performing configuration check...
**2020-02-02 12:51:44.685 [main info] loading configuration files from http://ebusd.eu/config/**
2020-02-02 12:51:44.744 [main info] reading templates /
2020-02-02 12:51:44.804 [main info] read templates in /
2020-02-02 12:51:44.804 [main info] reading file memory.csv
2020-02-02 12:51:44.863 [main info] successfully read file memory.csv
2020-02-02 12:51:44.863 [main info] reading file broadcast.csv
2020-02-02 12:51:44.923 [main info] successfully read file broadcast.csv
2020-02-02 12:51:44.923 [main info] read config files
2020-02-02 12:51:44.923 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2020-02-02 12:51:44.923 [main notice] ebusd stopped
root@eBusd:~# nano /etc/default/ebusd
Benutzen Sie „fg“, um zu nano zurückzukehren.

[1]+  Angehalten              nano /etc/default/ebusd
root@eBusd:~# systemctl status ebusd.service
● ebusd.service - LSB: controls ebusd, the daemon for communication with eBUS heating systems.
   Loaded: loaded (/etc/init.d/ebusd)
   Active: active (running) since So 2020-02-02 12:47:35 CET; 8min ago
  Process: 417 ExecStart=/etc/init.d/ebusd start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/ebusd.service
           └─**431 /usr/bin/ebusd --pidfile /var/run/ebusd.pid -d 192.168.11.45:5000 -n --latency=50000 --configpath=/etc/ebusd/config -p 8888 --s.**

john30, 2020-02-02:
if you run ebusd in foreground to check config etc, none of the settings from /etc are taken into account.
in fact, the /etc/default/ebusd settings are simply passed as arguments to ebusd when started as a service. So if you like to use the same for testing in foreground or whatever, you just need to copy them as arguments.

closed 2020-02-02 by john30

tttomson, 2020-02-02:
Thanks John. One step closer to solve my Problem. :-)