john30/ebusd #335: Vaillant eBus finding additional datapoints

joergensen70, 2020-02-06:
Hi, I know this topic has been discussed a lot but I didn't find a good direction yet. 
I'm using ebusd with the standard csv files successfully since a couple of month - thanks for the great effort from John30. Awesome Job.
I still miss some datapoints which are important to troubleshoot my heating system. Most of them are coming from the HMU (Address 08). I can view these values under ...

My understanding is that find and grab won't help since I have to ask for the values actively.
I run the script readallvaillantregisters.sh -a 08 but I get only = 00 for all 512 values. Same for -a 15.

I just used the bash to execute the script. What do I miss here? What is the way to poll all possible addresses from the HMU and try to identify the values I need?

I need:
Sole-Temperatur (Aus / Ein) (HMU)
Durchfluss Gebäudepumpe (HMU)
Puffertemperatur Oben/Unten (VR700)
Pumpenleistung (HMU)
Umweltkeris Druck (HMU)

I can see the values on the HMU under the Menu Fachhandwerkerebene->Testmenü->Sensor/Aktor Test

T.01 Leistung Gebäudepumpe
T.83 Durchfluss Gebäudekreis
T.97 Sole Temperatur Ein
T.98 Sole Temperatur Aus
T.101 Druck Umweltkreis

It seems many people are looking for some of these values and I would like to try getting them.
Any suggestions in a bit detailed manner (still in the learning curve - sorry about that)

john30, 2020-03-10:
well if the reading all registers revealed nothing, then you probably can't get the values. You might want to extend the script to really all possible registers (65536). Currently it only looks at the first 512 (just replace "512" with "65536").
if you can see the values in the menu, you might want to check the bus traffix at the time you're walking to the menu, but since this is the HMU integrated UI, I guess it wont send anything on the bus.

joergensen70, 2020-03-30:
I tried now another way to find additional data points for the Vaillant HMU 08. 
- Added all possible addresses to the csv file 00-FF after entry *r,,,,,,"B51A","05FF32",,,IGN:3,,,,
- Polled all possible addresses first with UIN datatype
- Added changing values to my Volkszaehler and record them

So far I was successful and found following channels and added them (see below):

I finally found the source in- and out temperature which is requested by many people, flow pressure, source pressure, Integral, flow temperature,...
But I have also a couple of values where I didn't find a correlation. See the channels ch_?.

I thought adding this to my post will help people who are still looking for the source temp espeacially. 

One question which came up was about the flow pressure and source pressure. I identified D2C as most appropriate data type (divisor 16) but I had to divide by 4 again to get the right value. The required divisor seems to be 64 but didn't find an appropriate data type for it.

# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment
*r,,,,,,"B511",,,,,,,
r,,State,,,,,07,energy,,UCH,,,,,,energy,,kWh,,onoff,,UCH,0=off;1=on,,,state,,UCH,0x01=ready;0x0b=error;0x09=heating;0x11=cooling;0x81=heating_water,,
# Heatpump statistics
*r,,,,,,"B51A","05FF32",,,IGN:3,,,,
r,,YieldHcDay,,,,,00,,,energy,,,,,,,,,,,,,,,
r,,ch_1,,,,,01,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_2,,,,,02,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_3,,,,,03,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_4,,,,,04,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_5,,,,,05,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_6,,,,,06,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_7,,,,,07,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,YieldHcMonth,,,,,0e,,,UIN,10,,,,,,,,,,,,,,,,,,,,
r,,CopHcMonth,,,,,0f,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,YieldHc,,,,,10,,,energy,,,,,,,,,,,,,,,
r,,CopHc,,,,,11,,,energy,,,,,,,,,,,,,,,
r,,YieldHwcMonth,,,,,12,,,energy,10,,,,,,,,,,,,,,
r,,CopHwcMonth,,,,,13,,,energy,,,,,,,,,,,,,,,
r,,YieldHwc,,,,,16,,,energy,,,,,,,,,,,,,,,
r,,CopHwc,,,,,17,,,energy,,,,,,,,,,,,,,,
r,,ch_18,,,,,18,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_1e,,,,,1e,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_1f,,,,,1f,,,D2C,,,,,,,,,,,,,,,,,,,,,
r,,FlowTemp,,,,,20,,,D2C,,,,,,,,,,,,,,,,,,,,,
r,,Integral,,,,,21,,,SIN,,,,,,,,,,,,,,,,,,,,,
r,,SourceTempInput,,,,,22,,,D2C,,,,,,,,,,,,,,,,,,,,,
r,,CurrentYieldPower,,,,,23,,,D1B,10,,,,,,,,,,,,,,,,,,,,
r,,CurrentConsumedPower,,,,,24,,,D1B,10,,,,,,,,,,,,,,,,,,,,
r,,CurrentCompressorUtil,,,,,25,,,D1B,,,,,,,,,,,,,,,,,,,,,
r,,SourceTempOutput,,,,,27,,,D2C,,,,,,,,,,,,,,,,,,,,,
r,,FlowPressure,,,,,3d,,,D2C,4,,,,,,,,,,,,,,,,,,,,
r,,SourcePressure,,,,,3e,,,D2C,4,,,,,,,,,,,,,,,,,,,,
r,,ch_3f,,,,,3f,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,Hours,,,,,40,,,energy,,,,,,,,,,,,,,,
r,,HoursHc,,,,,41,,,energy,,,,,,,,,,,,,,,
r,,HoursCool,,,,,43,,,energy,,,,,,,,,,,,,,,
r,,HoursHwc,,,,,44,,,energy,,,,,,,,,,,,,,,
r,,ch_46,,,,,46,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_49,,,,,49,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_4a,,,,,4a,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_4b,,,,,4b,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_4c,,,,,4c,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,ch_63,,,,,63,,,UIN,,,,,,,,,,,,,,,,,,,,,
# Total heating yields
*r,,,,,,"B516","18",,,IGN:1,,,
# Total electric power consumption
r,,ConsumptionTotal,,,,,02,,,energy,,,,,,,,,,,,,,,
!include,hcmode.inc,,,,,,,,,,,,
!include,errors.inc,,,,,,,,,,,,

john30, 2020-04-11:
good approach! for the extra divisor you can just add another template type in the _templates.csv

mwildbolz, 2020-10-11:
@joergensen70 Do you probably use this data also in combination with openHAB?
I already use your datapoints - working well :-) on ebusd side - but I'm stuck at implementing these in my openHAB environment?!?

joergensen70, 2020-10-11:
No, I'm using a Redmatic implementation. Never used open HAB

Holen Sie sich Outlook für Android

pagaille, 2020-11-02:
Hi @joergensen70 

Would you mind elaborating  a bit more your method ?

> Added all possible addresses to the csv file 00-FF after entry *r,,,,,,"B51A","05FF32",,,IGN:3,,,,
> Polled all possible addresses first with UIN datatype
> Added changing values to my Volkszaehler and record them

Why would your method reveal other data that @john30's script ?

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30

plasma99new, 2022-01-06:
Für jeden, der Werte für eine 08.hmu.scv sucht, diese funktionieren einwandfrei an meiner.

For everyone who is looking for values ​​for an 08.hmu.scv, these work perfectly on mine.

address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0305;HW=0403", loaded "vaillant/08.hmu.csv"
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0305;HW=0403"

Paradoxa, 2022-07-21:
Hallo @joergensen70 and @plasma99new, 

Thanks for providing the data points. 
Have you found any other settings for the hmu?

kjumybit, 2025-01-12:
Hi, 
I just add the "SourceTempOutput" definition  mentioned above to the "vaillant/08.hmu.csv" file. That is missing in the current official [repo](https://github.com/eBUS/ebus.github.io).

```
r,,,SourceTempOutput,default *r,,,b51a,05ff3227,value,,IGN:3,,,,value,,D2C,,°C,
```

It works for me!.

```
$ ebusctl r -f -vv -c hmu SourceTempInput
hmu SourceTempInput value=3.75 °C

$ ebusctl r -f -vv -c hmu SourceTempOutput
hmu SourceTempOutput value=-0.19 °C
```

Thanx!