john30/ebusd #337: connection but no connection?

minitrue-nl, 2020-03-06:
Hi,
For more than a year I had ebusd running on my linux server trough eseras usb coppler without problems, but now (after adding another usb-device) I encountered problems.

Ebusd seems to run just fine as a program, but doesn't seem connected with the ebusd anymore. I used to have a full set up which could read all comunnication from vrc700, heatpump and everything.

> ubuntu@homeserver:~$` ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
signal: acquired
symbol rate: 59
max symbol rate: 147
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

if I run `sudo ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0` I see that there is a connection and data is coming in, but lots of the messages are >ff in stead of >aa as described in the docs. Could this be a hint to a solution? I rember from my first set up I actually saw the >aa's

john30, 2020-03-10:
in most cases this is due to another device connected to the UART or another software using it for different purposes. I'd recommend changing from /dev/ttyUSB* to /dev/serial/by-id/... to prevent a reboot or plug in of another USB device to ruin your configuration

minitrue-nl, 2020-03-14:
Thanks, that pointed me in the right direction. There was still another (modbus) program running in the background, messing with the /dev/ttyUSB0.

closed 2020-03-14 by minitrue-nl