john30/ebusd #338: MQTT

Dinth, 2020-03-08:
My ebusd is communicating with Home Assistant using MQTT. 
Whenever i restart Mosquitto server, ebusd doesnt autoconnect back to it and i also need to restart ebusd daemon to get ebusd talk with my Home Assistant again. 
Im also using same Mosquitto daemon for multiple other services, like Zwave2MQTT, Tasmota, Valetudo and all those services automatically re-establish connection with MQTT after Mosquitto restart. I had a look in ebusd log but couldnt find anything worrisome there. Can someone help me with ideas how to troubleshoot this further?

john30, 2020-03-10:
which ebusd version do you use?

Dinth, 2020-03-10:
version: ebusd 3.4.v3.3-51-g57eae05

fmeies, 2020-03-12:
Hi all, I can confirm this, I'm using ebusd-3.4_armhf-stretch_mqtt1.deb. 
I can find this in my ebusd.log:
`[mqtt error] communication error: errno 22=Invalid argument`
Restarting ebusd resolves the issue.

fmeies, 2020-04-09:
Hm, I did some more testing with the current HEAD and cannot reproduce the reconnect problems anymore.

closed 2020-04-11 by john30