john30/ebusd #339: Cannot find other devices anymore

minitrue-nl, 2020-03-16:
A couple of months ago I had running ebusd like a charm on my ubuntu server, it would find (all) devices on my network: hmu (heat pump), vr_70 (mixer) and 700 (vrc control device). However since a new install ebusd only seems to find data on the hmu. 

When I try to probe the installation either through http (localhost:8082/data) or ebusctl I have no success because the page won't load or the command hangs. What could be a solution?

`ubuntu@homeserver:~$ sudo ebusd --scanconfig -d /dev/ttyUSB0 --httpport=8082 -f
2020-03-16 22:12:10.107 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with auto scan
2020-03-16 22:12:10.367 [bus notice] bus started with own address 31/36
2020-03-16 22:12:10.367 [bus notice] signal acquired
2020-03-16 22:12:11.697 [bus notice] new master 71, master count 2
2020-03-16 22:12:11.741 [bus notice] new master 03, master count 3
2020-03-16 22:12:11.741 [update notice] received unknown MS cmd: 7108b5110107 / 0500d9000101
2020-03-16 22:12:13.229 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 00
2020-03-16 22:12:17.746 [update notice] received unknown MS cmd: 7108b507030bffff / 0100
2020-03-16 22:12:20.486 [bus notice] scan 08: ;Vaillant;HMU01;0304;8802
2020-03-16 22:12:20.486 [update notice] store 08 ident: done
2020-03-16 22:12:20.487 [update notice] sent scan-read scan.08  QQ=31: Vaillant;HMU01;0304;8802
2020-03-16 22:12:20.487 [bus notice] scan 08: ;Vaillant;HMU01;0304;8802
2020-03-16 22:12:20.755 [main notice] read common config file vaillant/scan.csv
2020-03-16 22:12:20.817 [main notice] read common config file vaillant/general.csv
2020-03-16 22:12:20.886 [main notice] read common config file vaillant/broadcast.csv
2020-03-16 22:12:20.966 [main notice] read scan config file vaillant/08.hmu.csv for ID "hmu01", SW0304, HW8802
2020-03-16 22:12:21.094 [main notice] found messages: 59 (0 conditional on 0 conditions, 0 poll, 9 update)
2020-03-16 22:12:23.305 [bus notice] scan 76: ;Vaillant;VWZIO;0111;0103
2020-03-16 22:12:23.306 [update notice] store 76 ident: done
2020-03-16 22:12:23.306 [update notice] sent scan-read scan.76  QQ=31: Vaillant;VWZIO;0111;0103
2020-03-16 22:12:23.306 [bus notice] scan 76: ;Vaillant;VWZIO;0111;0103
2020-03-16 22:12:23.563 [update notice] sent unknown MS cmd: 3176b5090124 / 09003231313731303030
2020-03-16 22:12:23.737 [update notice] sent scan-read scan.76 id QQ=31:
2020-03-16 22:12:23.912 [update notice] sent scan-read scan.76 id QQ=31:
2020-03-16 22:12:24.007 [bus notice] max. symbols per second: 120
2020-03-16 22:12:24.086 [update notice] sent scan-read scan.76 id QQ=31: 21;17;10;0020180703;1610;005051;N9
2020-03-16 22:12:24.086 [bus notice] scan 76: ;21;17;10;0020180703;1610;005051;N9
2020-03-16 22:12:24.161 [main error] unable to load scan config 76: no file from vaillant with prefix 76 found
2020-03-16 22:12:24.161 [main error] scan config 76: ERR: element not found
2020-03-16 22:12:26.248 [main error] scan config e8: ERR: ACK error
2020-03-16 22:12:27.845 [update notice] received unknown MS cmd: 7108b507030bffff / 0100
2020-03-16 22:12:28.097 [update notice] received read hmu State QQ=71: 0;217;off;ready
2020-03-16 22:12:33.550 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 00
2020-03-16 22:12:35.939 [update notice] received read hmu State QQ=71: 0;217;off;ready
2020-03-16 22:12:37.939 [update notice] received unknown MS cmd: 7108b507030bffff / 0100
2020-03-16 22:12:38.343 [main error] scan config e8: ERR: ACK error
2020-03-16 22:12:43.690 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 00
2020-03-16 22:12:43.978 [update notice] received read hmu State QQ=71: 0;217;off;ready
2020-03-16 22:12:48.035 [update notice] received unknown MS cmd: 7108b507030bffff / 0100
2020-03-16 22:12:50.432 [main error] scan config e8: ERR: read timeout
2020-03-16 22:12:52.028 [update notice] received read hmu State QQ=71: 0;217;off;ready
2020-03-16 22:12:53.890 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 00
2020-03-16 22:13:00.102 [update notice] received read hmu State QQ=71: 0;217;off;ready
2020-03-16 22:13:03.047 [update notice] received unknown MS cmd: 7108b507030bffff / 0100
2020-03-16 22:13:04.034 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 00
^C2020-03-16 22:13:06.338 [main notice] SIGINT received
`

john30, 2020-04-11:
it detected the VWZ as well, but there is no CSV for it. Other than that, there are some ACK errors in the log, so you might want to check your hardware interface.

minitrue-nl, 2020-04-11:
Thanks for your comment. But I'm wondering, if ebusd detects the communication from the HMU and VWZ, could it be a hardware interface problem that it is not reading vr_70 and 700?

And can you explain what an ACK error is?

john30, 2020-04-18:
yes could be. you might want to try it with extended [latency](https://github.com/john30/ebusd/wiki/2.-Run#device-options).
the ACK error is logged when an illegal byte is received in that state or when the device responded with "not acknowledged"

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30