john30/ebusd #340: Ergänzung von fehlenden Parametern in den CSV Dateien

RESDE, 2020-03-26:
Lieber John30,

sehr guter Adapter bis jetzt, sammelt auch fleissig Daten auf dem Bus.
Leider aber meldet das logfites fehlende Objekte/Adressen, habe die schon mit grab etc abgefangen.
Kannst Du nochmals ein wenig mehr Anleitung geben, wo welche Datei wie ergänzt werden muss? Konkret die Vaillant Dateien, denn es liegen sehr viele bei Dir im Github, aber ich kann leider nicht erkennen welche sich der EbusD auf meinen Raspberry Pi "gezogen" hat.
Grüße
Reiner

john30, 2020-03-29:
ebusd lädt die passenden für die gescannten Geräte vom Webservice runter. dort liegen die Files, die im Link "latest" auf ebusd liegen, also [diese hier](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/de).
Wenn du an den Files Ergänzungen vornehmen willst, dann wie im [readme erläutert](https://github.com/john30/ebusd-configuration#installation-on-up-to-date-ebusd-20-or-higher) vorgehen zum auschecken und ebusd entsprechend konfigurieren.

RESDE, 2020-03-30:
Hi,
O.K. Das heisst konkret die relevanten Files sind bereits lokal bei mir abgespeichert, die habe ich gefunden und kann diese jetzt ergänzen....
Aber: wieso sind noch nicht alle Bus Objekte "entschlüsselt"? Es sind doch so viele Vaillant Nutzt auf EbusD unterwegs und Du sammelst immer wieder fleissig ein. Soll ich Dir dann meine ergänzten Vaillant Files auch wieder zusehen?
Hast Du Tipps wie ich herausfinde, welches Bus D Objekt sprich Steuerungsgröße der Anlage wirklich es ist? Sprich Vorlauftemperatur hast Du ja in den Files schon aufgenommen, welche anderen Parametern können das sein?- Hattest Du Gelegenheit die Grau Ergebnisse zu analysieren? Hast Du konkret ne Idee, welche Emus Objekte dass sind, dann pflege ich mir die CSV selbständig weiter?

- Frage zum Bus Adapter: ich verwende Essers USB Adapter. Wieso tritt ein Fehler in meiner Heizungssteuerung auf, wenn ich meinen Pi mit angeschlossenem USB Adapter neu starte? Die WP Steuerung geht dann auf Fehler und schaltet sogar auf Notbetrieb?
- Wieso hast Du kein Tipps für die Einbindung in IOBroker auf Deinem Wiki? Er IO Broker Adapter Author weiß nicht wirklich, was "Dein" EBUS Damon kann und macht...;-) Könntest DU mir evtl. einen oder 2 Experten "vermitteln", mit denen ich aus Deinem Adapter die Objekte dann in den IOBroker einwachst tiefe?Hast Du einen oder zwei Experten der den EBUSd via MQTT faktisch in Iosbroker aufnimmt?Ich würde dann anbieten, dass ich diesen Teil des HowTo verfassen werde und den kannst Du dann in Dein Wiki aufnehmen? Schliesslich möchte ich nicht nur "nehmen" sondern auch gerne "geben".
Btw:Darf ich Deine Telefonnummer erfahren, damit ich Dich mal sprechen kann um mich zu bedanken?
GrüßeReiner

-

john30, 2020-04-11:
es sind immer noch weiße Flecken in der Liste der Message Definitionen und ich komme nicht dazu, die Lücken zu füllen, selbst mit den ganzen issues auf ebusd-configuration, weil das enorm Zeit in Anspruch nimmt, die ich nicht habe.
Also am besten entschlüsselte CSVs als Pull Request unter ebusd-configuration einstellen und dabei auf die englische Übersetzung achten. Dann hat es eine Chance, übernommen zu werden.
"Welchs Bus D Objekt..." verstehe ich nicht. was meinst du denn, also von welcher "Steuerungsgröße" ist die Rede? da gibt es hunderte.
Der Fehler in der Heizungssteuerung tritt vermutlich deshalb auf, weil der Adapter am RPi nach Neustart vermutlich nonense auf den Bus schreibt und/oder weil die Spannungslevel sowie die Stromaufnahme nicht eBUS Spezifikations-konform sind.
Den IOBroker Experten musst du schon selbst finden (was soll ich denn noch alles?).
Bitte hab Verständnis dafür, dass ich nicht meine Telefonnummer publiziere. Ein bisschen Privatsphäre hätte ich schon gern.

RESDE, 2020-04-12:
Hi John,

Erst einmal frohe Ostern.
Sehr nett dass Du nochmals Tipps gibst.
Das mit einem Tipp ob Du noch andere Experten kennst, war gedacht dass die Last der Issues/fragen auf mehrere Schulter verteilt werden sprich Entlastung für Dich😎😎

Soll ich dann besser meine Fragen in das github Forum einstellen, dann könntest Du dort antworten und es wäre gleich so ne q&a session als ergebnis, gut für die Community...
Beste grüsse
Reiner

john30, 2020-05-10:
ja das ist die Idee. aber dann bitte in englisch

closed 2021-02-04 by RESDE

KarLOs964cabrio, 2023-10-10:
Hallo John, 
ich versuche mit der Homeassistant Integration sinnvolle Daten von meiner Auromatic 620 zu erhalten.
 leider bekomme ich nur den Broadcast und SumFlow, alle anderen Daten sind "nicht verfügbar".
bei der Hardware handelt es sich um einen Raspi3 an dem per USB ein Adapter von ESERA angeschlossen ist:

[image]

Meine configuration .yaml beinhaltet nur die drei Zeilen:  

ebusd:
  host: 127.0.0.1
  circuit: "SOLSY"

und im MQTT Explorer erhalte ich folgendes:

ebusd
global
version = "ebusd 23.2.p20230716"
running = true
scan = "running"
signal = true
updatecheck = "revision 23.2 available"
uptime = 47259
hc
SumFlowSensor = { "temp": {"value": 33.81}, "sensor": {"value": "ok"}}
DateTime = { "dcfstate": {"value": "valid"}, "btime": {"value": "11:51:35"}, "bdate": {"value": "10.10.2023"}, "temp2": {"value": 14.938}}
currenterror = { "0": {"name": "error", "value": null}, "1": {"name": "error", "value": null}, "2": {"name": "error", "value": null}, "3": {"name": "error", "value": null}, "4": {"name": "error", "value": null}}
scan.08
= { "MF": {"value": "Vaillant"}, "ID": {"value": "BAI00"}, "SW": {"value": "0703"}, "HW": {"value": "7401"}}
broadcast
vdatetime = { "time": {"value": "11:51:02"}, "date": {"value": "10.10.2023"}}
datetime = { "outsidetemp": {"value": 14.750}, "time": {"value": "11:51:02"}, "date": {"value": "10.10.2023"}}
outsidetemp = { "temp2": {"value": 14.750}}
scan.26
= { "MF": {"value": "Vaillant"}, "ID": {"value": "SOLSY"}, "SW": {"value": "0500"}, "HW": {"value": "6301"}}
scan.50
= { "MF": {"value": "Vaillant"}, "ID": {"value": "SOLSY"}, "SW": {"value": "0500"}, "HW": {"value": "6301"}}
scan.23
= { "MF": {"value": "Vaillant"}, "ID": {"value": "SOLSY"}, "SW": {"value": "0500"}, "HW": {"value": "6301"}}
scan.25
= { "MF": {"value": "Vaillant"}, "ID": {"value": "SOLSY"}, "SW": {"value": "0500"}, "HW": {"value": "6301"}}
scan.ec
= { "MF": {"value": "Vaillant"}, "ID": {"value": "SOLSY"}, "SW": {"value": "0500"}, "HW": {"value": "6301"}}

was fehlt , oder kann die 620/3 nicht mehr Infos herausgeben, oder ist der ESERA  Adapter Mist ?

Ich fummle da jetzt schon ein paar Monate dran rum , komme aber nicht weiter.