john30/ebusd #341: Unable to parse poll-read sc YieldThisYear - massive number of log entries

mwildbolz, 2020-03-30:
Hi there!

I have an Vaillant heating system connected via an Esera ebus-coupler (USB) to an raspberry pi running ebusd. In the logs I get a massive number of the following entries:

```2020-03-30 17:14:38.104 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f00 / 00: ERR: invalid position
2020-03-30 17:14:38.227 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f08 / 00: ERR: invalid position
2020-03-30 17:14:38.351 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f10 / 00: ERR: invalid position
2020-03-30 17:14:38.474 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f18 / 00: ERR: invalid position
2020-03-30 17:14:38.596 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f20 / 00: ERR: invalid position
```

What's the problem here?
A configuration issue, or from openHAB (ebus 2.0 binding)?

ebusctl info
```
Version: ebusd 3.4.v3.3-51-g57eae05
signal: acquired
symbol rate: 48
max symbol rate: 159
min arbitration micros: 11
max arbitration micros: 564
min symbol latency: 0
max symbol latency: 31
reconnects: 0
masters: 4
messages: 549
conditional: 0
poll: 1
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0613;HW=6903", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0613;HW=6903", loaded "vaillant/ec.sol.sc.csv"
```

ebusctl scan report
```08;Vaillant;HMU00;0307;0403;21;19;44;0010016422;0001;005692;N8
15;Vaillant;70000;0613;6903;21;19;44;0020266797;0082;067959;N0
26;Vaillant;VR_71;0104;0503;21;19;39;0020184846;0082;009563;N2
76;Vaillant;VWZ00;0307;0403
ec;Vaillant;SOL00;0613;6903;21;19;44;0020266797;0082;067959;N0
```

Input appreciated, thanks...

john30, 2020-04-11:
the answer of the slave to the message is too short: "00" is the length byte and consequently there is nothing left to read from. I guess the device does not support reading YieldThisYear, at least in the version used (there are several in the CSV, just look out for yield*.inc).

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30

peterfido, 2023-05-18:
Diese Meldung füllt auch mein Log:
`2023-05-18 09:42:55.069 [mqtt error] decode sc YieldThisYear: ERR: invalid position
2023-05-18 09:42:58.364 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f00 / 00: ERR: invalid position
2023-05-18 09:42:58.717 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f08 / 00: ERR: invalid position
2023-05-18 09:42:59.086 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f10 / 00: ERR: invalid position
2023-05-18 09:42:59.448 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f18 / 00: ERR: invalid position
2023-05-18 09:42:59.570 [update notice] received read bai Status01 QQ=10: 44.5;44.0;8.625;55.0;61.0;off
2023-05-18 09:42:59.600 [bus error] poll sc YieldThisYear failed: ERR: wrong symbol received
2023-05-18 09:43:00.357 [update notice] received update-write bai SetMode QQ=10: auto;63.0;-;-;0;0;0;0;0;0
2023-05-18 09:43:05.081 [mqtt error] decode sc YieldThisYear: ERR: invalid position
2023-05-18 09:43:09.318 [update notice] received read bai Status01 QQ=10: 44.0;43.5;8.625;55.0;61.0;off
2023-05-18 09:43:09.587 [update notice] received update-write bai SetMode QQ=10: auto;63.0;-;-;0;0;0;0;0;0
2023-05-18 09:43:09.849 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;65.0
2023-05-18 09:43:10.344 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f00 / 00: ERR: invalid position
2023-05-18 09:43:10.697 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f08 / 00: ERR: invalid position
2023-05-18 09:43:11.066 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f10 / 00: ERR: invalid position
2023-05-18 09:43:11.433 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f18 / 00: ERR: invalid position
2023-05-18 09:43:11.787 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f20 / 00: ERR: invalid position
2023-05-18 09:43:12.155 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f28 / 00: ERR: invalid position
2023-05-18 09:43:12.523 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f30 / 00: ERR: invalid position
2023-05-18 09:43:12.876 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f38 / 00: ERR: invalid position
2023-05-18 09:43:13.245 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f40 / 00: ERR: invalid position
2023-05-18 09:43:13.614 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f48 / 00: ERR: invalid position
2023-05-18 09:43:13.982 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f50 / 00: ERR: invalid position
2023-05-18 09:43:14.335 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f58 / 00: ERR: invalid position
`

Heiha9, 2024-03-24:
@peterfido, hast Du ne Lösung gefunden?

peterfido, 2024-03-25:
Nein.