john30/ebusd #344: configlang ignored

guillaumezin, 2020-04-23:
Hello,

How do you use the configlang option? Whatever I do, I got german definition:

I tried many different options:

sudo ebusd --scanconfig --httpport 8889 --configlang=en-GB --log="all error" --dumpconfig
sudo ebusd --scanconfig --httpport 8889 --configlang=en --log="all error" --dumpconfig
sudo ebusd --scanconfig --httpport 8889 --configlang="en_US.UTF-8" --log="all error" --dumpconfig
...

And it gives

```
type,circuit,name,comment,qq,zz,pbsb,id,*name,part,type,divisor/values,unit,comment
u,broadcast,datetime,Datum/Uhrzeit,,fe,0700,,outsidetemp,m,D2B,,°C,Temperatur,time,m,BTI,,,Uhrzeit,date,m,BDA,,,Datum
u,broadcast,error,Fehlernachricht,,fe,fe01,,error,m,STR:10,,,
u,broadcast,id,Identifikation,,fe,0704,,manufacturer,m,UCH,6=Dungs;15=FH Ostfalia;16=TEM;17=Lamberti;20=CEB;21=Landis-Staefa;22=FERRO;23=MONDIAL;24=Wikon;25=Wolf;32=RAWE;48=Satronic;64=ENCON;80=Kromschröder;96=Eberle;101=EBV;117=Grässlin;133=ebm-papst;149=SIG;165=Theben;167=Thermowatt;181=Vaillant;192=Toby;197=Weishaupt;253=ebusd.eu,,Geräte-Hersteller,id,m,STR:5,,,Geräte-ID,software,m,PIN,,,Software-Version,hardware,m,PIN,,,Hardware-Version
r,broadcast,id,Identifikation,,,0704,,manufacturer,s,UCH,6=Dungs;15=FH Ostfalia;16=TEM;17=Lamberti;20=CEB;21=Landis-Staefa;22=FERRO;23=MONDIAL;24=Wikon;25=Wolf;32=RAWE;48=Satronic;64=ENCON;80=Kromschröder;96=Eberle;101=EBV;117=Grässlin;133=ebm-papst;149=SIG;165=Theben;167=Thermowatt;181=Vaillant;192=Toby;197=Weishaupt;253=ebusd.eu,,Geräte-Hersteller,id,s,STR:5,,,Geräte-ID,software,s,PIN,,,Software-Version,hardware,s,PIN,,,Hardware-Version
w,broadcast,id,Identifikation,,fe,0704,,
w,broadcast,queryexistence,Existenzanfrage,,fe,07fe,,
u,broadcast,signoflife,Lebenszeichen,,fe,07ff,,
r,memory,eeprom,EEPROM Daten lesen,,,0902,,address,m,UIN,,,Startadresse,count,m,UCH,,,number of bytes,data,s,HEX:*,,,Datenbytes
w,memory,eeprom,EEPROM Daten schreiben,,,0903,,address,m,UIN,,,Startadresse,data,m,HEX:*,,,Datenbytes
r,memory,ram,RAM Daten lesen,,,0900,,address,m,UIN,,,Startadresse,count,m,UCH,,,number of bytes,data,s,HEX:*,,,Datenbytes
w,memory,ram,RAM Daten schreiben,,,0901,,address,m,UIN,,,Startadresse,data,m,HEX:*,,,Datenbytes
```

version: ebusd 3.4.v3.4-13-g7b8ceb7

Thank you

john30, 2020-04-24:
this feature was added in preparation of using 3.x configuration files, but those are not yet available

john30, 2023-09-23:
solved with 4263c98f04da3244e0f71640af8d5e70c3858495

closed 2023-09-23 by john30