john30/ebusd #345: UART to eBUS

meatalhead, 2020-04-30:
Does the + and - eBUS connections match TX and RX UART? Just with different voltages?

guillaumezin, 2020-05-01:
No, eBUS is asynchronous UART with power supply on the same line.

"-" is ground
"+" is RX or TX modulation (depending on the direction) plus approximately 12V

meatalhead, 2020-05-01:
Daft question then, how do you connect eBus to Pi UART pins if they're Rx and TX?

guillaumezin, 2020-05-01:
You don't, you need an hardware interface in between, cf. https://github.com/john30/ebusd/wiki/6.-Hardware

guillaumezin, 2020-05-01:
By the way, I still have 3 PCB of this board: https://www.mikrocontroller.net/topic/346833 if you want to buy components and solder them yourself, I can send you back a PCB if you send me a stamped enveloppe (20g, I'm in France).

closed 2020-09-20 by john30