john30/ebusd #347: Can ebusd act as master ?

pagaille, 2020-05-06:
Hi,

I'd like to connect to a Vaillant solar controller (VMS8/D). That device is currently not connected to anything (it works autonomously) and is slave (no power supply).

I plan to power the bus myself, but will ebusd be able to act as a bus master and generate all the needed signals ?

Thanks

pagaille, 2020-05-09:
Anybody ?

john30, 2020-05-10:
yes, you can use [--generatesyn ](https://github.com/john30/ebusd/wiki/2.-Run#ebus-options) to let ebusd act as SYN generator. I use this mode myself for a solar station that only has a kind of passive connector. Fortunately, that already has the necessary power supply.

pagaille, 2020-05-10:
Oh cool ! I was unaware of the use of that setting. Thanks, I’ll try. 

Matthieu G.  (en mode mobile)

closed 2020-05-16 by john30

pagaille, 2020-10-31:
Hi @john30 

I finally built my interface. I'm trying to connect to the VMS but until now no success.

Here is the ebusd output when starting and running `ebusctl scan full`  : 

```
pi@raspberrypi:~ $ docker logs ebusd -f
2020-10-31 14:47:49.559 [main notice] ebusd 3.4.p20201029 started with auto scan
2020-10-31 14:47:49.857 [bus notice] bus started with own address 31/36
2020-10-31 14:47:50.273 [bus notice] acting as AUTO-SYN generator
2020-10-31 14:47:50.275 [bus notice] signal acquired
2020-10-31 14:48:21.606 [main error] scan: empty
2020-10-31 14:49:55.543 [main notice] update check: revision v3.4 available
```
```
ebusctl info
localhost: info
version: ebusd 3.4.p20201029
update check: revision v3.4 available
signal: acquired
symbol rate: 9
max symbol rate: 72
min arbitration micros: 93
max arbitration micros: 264
min symbol latency: 3
max symbol latency: 15
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

The TX led is blinking fast on the adapter but I can't see any RX packet. Not sure that my adapter is perfectly working (I don't have a scope) but static measures shows that it should work correctly.

Any idea ?

andig, 2020-10-31:
I‘d say id RX is silent your adapter ist broken- you should at least see broadcast and other master‘s messages on the bus.

pagaille, 2020-10-31:
Thanks @andig I should have mentioned again that the VMS is alone, there is nothing except ebusd on the bus. Therefore I use the --generatesyn feature.

pagaille, 2020-11-01:
BTW my RX led was reversed, I fixed it and now it works : RX and TX are blinking together rapidly because of --generatesyn.. But the VMS is still quiet. @john30 I remember you explained me you had roughly the same setup. Is it a Vaillant solar controller ? Isn't it waiting for a particular packet to get awake ?

pagaille, 2020-11-01:
I did some progress. I tried to install the[ ttyebus kernel extension](https://github.com/eBUS/ttyebus) because I'm using a Pi Zero W and that I know that it is using the PL011 for the primary UART, which is known to not being friendly with the ebus protocol. 
My VMS is now detected ! 

```
signal: acquired
symbol rate: 10
max symbol rate: 79
min arbitration micros: 223
max arbitration micros: 383
min symbol latency: 3
max symbol latency: 41
reconnects: 0
masters: 2
messages: 13
conditional: 0
poll: 0
update: 4
address 01: master #6
address 06: slave #6, scanned "MF=Vaillant;ID=VMD01;SW=0108;HW=1703"                                                                   
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ed: slave, scanned "MF=Vaillant;ID=VMD01;SW=0108;HW=1703"
```

Now I still didn't saw any message on the bus but hey, that's something !

john30, 2020-11-01:
since you now know the slave address of the device, you could go ahead and use the [readallvaillantregisters.sh](https://github.com/john30/ebusd/blob/master/contrib/scripts/readallvaillantregisters.sh) against it like this with `nc` (netcat) available on the host:
`readallvaillantregisters.sh -a 06`

pagaille, 2020-11-01:
Oh thanks !!

```
pi@raspberrypi:~$ ./readvms.sh -a 06
0 = 0100
1 = 0100
2 = 0100
3 = 0100
4 = 0100
5 = 03600100
6 = 0385fdaa
7 = 032b0200
8 = 0100
9 = 0100
10 = 0385fdaa
11 = 0100
12 = 03920300
13 = 0100
14 = 0100
15 = 0100
16 = 0100
17 = 0100
18 = 0100
19 = 0100
20 = 0100
21 = 0100
22 = 0100
23 = 0100
24 = 0100
25 = 043a300000
26 = 0100
27 = 04050e0000
28 = 0100
29 = 0470070000
30 = 0100
31 = 0400000000
32 = 0100
33 = 0100
34 = 020000
35 = 03261711
36 = 03010b14
37 = 0101
38 = 0101
39 = 0107
40 = 0101
41 = 0100
42 = 0100
43 = 0100
44 = 0100
45 = 020000
46 = 02c003
47 = 0490010000
48 = 0100
```

Now something I must understand is how to use of the CSV that are supposed to help parse the hex flow. Let's continue reading the doc 😊 Next step : node-red !

Thanks for your help.

pagaille, 2020-11-02:
@john30 is there a guide to build or at least correct a CSV parsing file ?

john30, 2020-11-16:
[see here](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files)