john30/ebusd #348: Setup on Synology with esera eBus Ethernet?

skyynet, 2020-05-09:
I installed ebusd 3.4 on my Synology RS818+ as a docker image. This is the first time I use docker so excuse my probably stupid questions.

I know the ip address of my esera ebus connector. I assume I have to set it up on the port-settings tab of the  docker container but how? Currently local port is set to auto, the container port to 8888 and the type to tcp.

Is there any step by step howto for Synology?

I want to use the eBus connector with my Valilant geoTHERM plus with PHP. Any information on how to get the data out of the ebusd after having it setup highly appreciated.

john30, 2020-05-10:
you don't have to set up anything special in the docker run for connecting to a network eBUS interface, just set the [--device](https://github.com/john30/ebusd/wiki/2.-Run#device-options) appropriately.
You can retrieve the data easily using MQTT or the HTTP interface of ebusd or even by opening a socket to the ebusd client port.

skyynet, 2020-05-10:
Thanks for your fast response. Unfortunately I don't know how to set the --device. Currently, I only looked at the Docker GUI provided by Synology. I assume I have to ssd on my Synology and edit the prefs file there to change the --device parameter. I searched my drive and can't find a config file. I created one as /etc/ebusd but it isn't used. Where do I place the config file so it is used by ebusd? Would this be a good start?

BUSD_OPTS="--configpath=/etc/ebusd -d /dev/ttyebus -p 8888  --latency=0 --recei
vetimeout=100000 -l /var/log/ebusd.log  --scanconfig=full  --httpport=8889 --htm
lpath=/var/www/"

I use an esera ebus connector with ethernet which sits at 192.168.1.xx.

john30, 2020-05-16:
see details for using the docker image [here](https://github.com/john30/ebusd/blob/master/contrib/docker/README.md)

skyynet, 2020-05-16:
I was aware of the docker information page but had difficulties for the Synology Docker GUI.
I now found out that one has to define the launch command when creating the container under advanced settings. The launch command can only be defined while creating the container and not changed afterwards. 

[image]

Now I'm  a bit further. When I get everything working I'll document it at skyynet.de/heatpump

closed 2020-05-16 by skyynet