john30/ebusd #349: No signal

tinkerdudeno1, 2020-05-13:
Hi John,
first of all thanks for sharing this awesome project. By now I think I must have read every single forum thread, wiki, and whatnot about ebusd but I can't seem to figure out where my problem is. Hope you or a member of the community can help me:

### Setup:

- A Vaillant auroCompact with a screw terminal named "Bus" which I figured to be an eBus (could not find any confirmation for that, though)
- To see if this is even working, I made a read-only eBus converter based on an Arduino nano (with and without a separate FTDI) like in this [project](https://github.com/dpavlin/Arduino-projects/blob/master/arduino_ebus/).
- A Raspberry 3 running ebusd (had it in a docker image at first but dropped that for better debugging)

### The oberservations:

- I see the FTDI (or the Arduino, respectively) blink like crazy when I connect the eBus lines from the heating system. Did some Serial.println() for testing to see that there is indeed a more or less constant 20 V arriving at the Arduino which drops to somewhere below 12 occasionally. Indicates to me the bus signal is fine and there is some action to see for ebusd.
- Now I connect the thing to the RPi via USB and run `$ebusd -f -c /tmp --logareas all --readonly --loglevel debug --lograwdata=bytes -d /dev/ttyUSB1` which gives me 

`2020-05-13 19:59:28.213 [main notice] ebusd 3.4.v3.3-51-g57eae05 started
2020-05-13 19:59:28.214 [main info] loading configuration files from /tmp
2020-05-13 19:59:28.214 [main info] reading dir  .XIM-unix
2020-05-13 19:59:28.214 [main info] successfully read dir .XIM-unix
2020-05-13 19:59:28.214 [main info] reading dir  .font-unix
2020-05-13 19:59:28.214 [main info] successfully read dir .font-unix
2020-05-13 19:59:28.214 [main info] reading dir  .Test-unix
2020-05-13 19:59:28.214 [main info] successfully read dir .Test-unix
2020-05-13 19:59:28.214 [main info] reading dir  .ICE-unix
2020-05-13 19:59:28.214 [main info] successfully read dir .ICE-unix
2020-05-13 19:59:28.215 [main info] reading dir  .X11-unix
2020-05-13 19:59:28.215 [main info] successfully read dir .X11-unix
2020-05-13 19:59:28.215 [main info] reading dir  systemd-private-aaefcd86b33b45c4aa0e4d3d1ef0a3ec-systemd-timesyncd.service-QcOJHo
2020-05-13 19:59:28.215 [main info] reading dir  systemd-private-aaefcd86b33b45c4aa0e4d3d1ef0a3ec-systemd-timesyncd.service-QcOJHo/tmp
2020-05-13 19:59:28.215 [main info] successfully read dir systemd-private-aaefcd86b33b45c4aa0e4d3d1ef0a3ec-systemd-timesyncd.service-QcOJHo/tmp
2020-05-13 19:59:28.215 [main info] successfully read dir systemd-private-aaefcd86b33b45c4aa0e4d3d1ef0a3ec-systemd-timesyncd.service-QcOJHo
2020-05-13 19:59:28.215 [main info] read config files
2020-05-13 19:59:28.218 [bus notice] bus started with own address 31/36
2020-05-13 19:59:28.218 [main info] registering data handlers
2020-05-13 19:59:28.218 [main info] registered data handlers
2020-05-13 19:59:38.219 [main debug] performing regular tasks
2020-05-13 19:59:48.219 [main debug] performing regular tasks`

- `ebusctl info` gives me 

`version: ebusd 3.4.v3.3-51-g57eae05
signal: no signal
reconnects: 0
masters: 0
messages: 0
conditional: 0
poll: 0
update: 0`

### The questions:
- Obviously: What am I doing wrong? 
- More specifically: I read that even for the Arduino version, people had to fiddle with the resistors. The way I understand it, these just act as voltage dividers on the Arduino input pins (4k7 vs. 1k or 47k vs. 10k as suggested by the above mentioned project (tried both actually)) to make sure we don't fry the 5V Arduino. The sketch is simple: If the 20 V arrive at the A0 pin (which results in more or less 5V on the A0 pin), we write a 1 on the serial RX pin. If the voltage drops to below 12 V (again something like 2-3 V on the A0), we write a 0 on the RX pin. So all we need is to decide at which A0-reading (range 0 (0V) to 1023 (5V)) we have 1 or 0 on the serial. From the Serial.println() I see that the signal fluctuates between roughly 820 and 320, so setting the threshold to 600 should be sufficient. Don't see how tinkering with the resistor values could help me here... It's seems so simple, that can't possibly be the issue, can it? Am I overlooking something?

### The bonus:
I read on one of the issues here that there was supposed to be a mass order of adapter PCBs last fall. Are there any left? Alternatively: Would you consider sharing the gerber files? Having the schematics is awesome enough, but judging from the past, there is quite a lot that can go wrong from there (especially with my lousy Eagle skills)...

Thanks for any assistance in advance.
Dan

kalledausb, 2020-05-13:
Hi Dan,
could You specify more precisely what auroCompact You have (VSC ...)? I have an older one and it has no ebus, rather the older 7-8-9 bus of Vaillant (to connect other Vaillant controllers for example). 

Cheers
Kalle

tinkerdudeno1, 2020-05-13:
Hi Kalle,
now that's what I call a quick reply, thanks so much! It's a VSC S 206/4-5 150. Thanks for pointing me to have a closer look at the exact type, now I did find the [confirmation](https://www.vaillant.de/heizung/produkte/solar-gas-brennwertsystem-aurocompact-1025.html), that it indeed does have an eBus.
Just had a look at the signal with my [Arduino-Oscilloscope](https://sourceforge.net/projects/scopino/) and could provide a screenshot if that is of any help. But it just shows that there is something going on on that bus line. Apparently, now we know that it's most probably an eBus signal.
Best,
Dan

kalledausb, 2020-05-13:
Hi Dan,
yes You are right, that has an eBus. I am unfortunately not the right one to answer any details regarding the self-made Arduino adapter.
However two things I know:
- Gerber files are not available
- Yes, there was a mass order last fall, but all sold out. Another one seems to be planned for next fall (at least that is how a message from Reinhart from fhem forum can be interpreted).

Basically I feel it would be best for You to get a "verified" adapter if possible. Or someone with deeper technical insight could provide more hints for You here.

Cheers
Kalle

tinkerdudeno1, 2020-05-14:
Hi Kalle,
thanks for your help. So the fhem forum is where to look for the adapter pcb, good to know. 

> Basically I feel it would be best for You to get a "verified" adapter if possible. Or someone with deeper technical insight could provide more hints for You here.

I'm afraid you're right. I'm also afraid the "read only" isn't going to cut it since the Vaillant systems are supposed to be not very talkative by default. But still from an academic point of view, I'm kind of at a "challenge accepted" point at which I really want to know what is going on. The Arduino stuff is -- in theory -- a very cheap and easy way to get started with ebusd. That is, if it works. Maybe with this issue we figure it out and can provide some help for others out there that are facing the same issue. 
Best,
Dan

tinkerdudeno1, 2020-05-15:
So I got it to work using two Nodemcu and John's ebusd-esp Firmware. Used the same sketch as on the Arduino before on the first Nodemcu and connected the Output pin of the first to the RX pin of the Nodemcu running ebusd-esp. On the Pi, I used -d <ip>:<port> and voila got a signal right away. Still bugs me that it didn't work with just one Arduino.
Now I'm facing a much bigger problem: Since my setup is a readonly setup, I get a lot of "received unknown" but I guess that's because I can't get the right config. I might open another issue for this or just hit the FHEM forum to see if there is a way to get a decent adapter.

closed 2020-09-20 by john30