john30/ebusd #352: Raspberry 4 - Signal / Acquire toggle

Gyrinx, 2020-06-18:
Hi John,

I have a problem since I switch to a raspberrypi 4.

I have newest ttyebus compiled and the ebusd version: ebusd 3.4.v3.3-51-g57eae05.
After a reboot the ebusd service start and toggle between Signal & Acquire. 

The log shows:
2020-05-30 00:06:05.349 [main error] scan config 15: ERR: no signal
2020-05-30 00:06:35.350 [bus info] scan 08 cmd: 3108070400

After a service restart, everything is ok. So I think, it has to do something with the serviced and sequence of boot. The serviced is your default.

Do you habe an Idea, how to fix that little task?

Thanks and kind regards,
Dirk

john30, 2020-09-20:
no idea, sorry

john30, 2021-10-06:
closed due to inactivity