john30/ebusd #353: Couldn't download docker ebusd mqtt version

skyynet, 2020-06-28:
Installing the docker image on my Synology works fine from registry.hub.docker.com (installed amd64 version 3.4).
Now I wanted to use MQTT by adding --mqtthost=192.168.1.21 --mqttport=1883 to the startup command which failed.
My startup options are -f --scanconfig -d tcp:192.168.1.70:5000 --mqtthost=192.168.1.21 --mqttport=1883
Thus, I thought I might need to install an MQTT enabled ebusd version. From the above repository I see none with _mqtt addition.
I found at https://github.com/john30/ebusd/releases that the docker images should be available from https://cloud.docker.com/repository/docker/john30/ebusd which redirects to https://hub.docker.com/r/john30/ebusd/

Where do I get an amd64 version as Docker file with MQTT? Sorry if I missed something.

andig, 2020-06-29:
The docker image is mqtt enabled. What do you mean by „the startup failed“?

skyynet, 2020-07-01:
Thanks for your reply. When I add the mqtt parameters to the startup command I get a syntax error:
[image]
[image]

andig, 2020-07-02:
Working for me:

    ebusd:
      container_name: ebusd
      image: john30/ebusd
      ports:
      - 8888:8888
      command: --enabledefine --enablehex --accesslevel='*' --scanconfig -f -d 192.168.0.46:9999 --latency 20000 --mqtthost=nas.fritz.box --mqttport=1883
      restart: always

I can only suggest to run this on the console instead of the NAS GUI to see what's really going on...

skyynet, 2020-07-02:
Thanks. Your parameters don't work for me when I use them in the Docker config of my Synology.
I'm not very familiar with Linux and running this in the console. I could SSH to my Synology but how do I configure it to run the ebusd incl. restart on reboot?

It would be much easier if the Docker options would work. Don't know why they don't.

andig, 2020-07-02:
Just try them on the console to see what/if error messages there are, correct the errors, then retry in the ui.

mozzarelka, 2020-07-06:
For me what makes the difference was connection part "-d tcp:", try without TCP:.

skyynet, 2020-07-06:
> For me what makes the difference was connection part "-d tcp:", try without TCP:.

I tried
`-f --scanconfig -d 192.168.1.70:5000 --mqtthost=192.168.1.21 --mqttport=1883`

but still get en execution command syntax error (Ausführungsbefehl-Formatfehler).

andig, 2020-07-07:
> but still get en execution command syntax error (Ausführungsbefehl-Formatfehler).

Suggesting again: try them on the console to see what/if error messages there are. If you'd do that we could stop guessing!

skyynet, 2020-07-07:
I SSH to my Synology. When I enter `ebusd -f --scanconfig -d 192.168.1.70:5000 --mqtthost=192.168.1.21 --mqttport=1883` the command is not found. I'm really dumb on Linux. I know, when ebusd is not in the path I can't call it. I just don't know where it is as all I did was instal the Docker image. Once found I can try it on the console and see what error I get.

andig, 2020-07-07:
You need to 

    sudo docker run john30/ebusd .... your options

olli-r, 2020-07-12:
Same here. The reason is some weird handling (aka. bug) of the (not so) special character "=". Just escape it within the configuration dialog, then it works:

ebusd -f --scanconfig -d 192.168.1.70:5000 --mqtthost\\=192.168.1.21 --mqttport\\=1883

skyynet, 2020-07-13:
Thanks olli-r!!!!!

This really solved it. Seems to be a bug of the docker implementation or Synology's way of using it.

closed 2020-07-13 by skyynet