john30/ebusd #354: Aussentemperatur zeitweise -40°

brezelmann01, 2020-07-01:
Hallo zusammen,

nach viel Tüftelei ist es mir vor einigen Wochen Dank diesem Forum gelungen, meinem Ebus einige Daten zu entloggen. Leider macht die Aussentemperatur, welche gebroadcasted wird, immer wieder Probleme. Unregelmäßig springt diese für Augenblicke auf -40°C. Alle anderen Werte funktionieren stabil.
Folgendes Setup habe ich im Einsatz:
Esera Buskoppler USB
Ecotec VC146
VR70 solarstation
Multimatic700
Buskoppler VR32 zur Anbindung der Recovair 260.

Die ebus Logs zeigen zum Zeitpunkt des Fehlers immer kurz vorher unknown messages:

`2020-07-01 17:42:51.112 [update notice] received read vr_70 SensorData QQ=10: 48.31;35.94;-;19.31;37.06;28.56;c8 10 00
2020-07-01 17:42:51.384 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:42:52.017 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:42:52.781 [update notice] received unknown MS cmd: 1038b50903298703 / 0487030080
2020-07-01 17:42:53.127 [update notice] received unknown MS cmd: 1038b50903298803 / 0488030080
2020-07-01 17:42:59.593 [update notice] received read bai Status02 QQ=10: auto;60;45.0;70;55.0
2020-07-01 17:43:00.599 [update notice] received read bai Status01 QQ=10: 20.5;21.0;23.250;-;-;off
2020-07-01 17:43:00.895 [update notice] received read vr_70 SensorData QQ=10: 48.31;35.94;-;19.31;37.06;28.56;c8 10 00
2020-07-01 17:43:01.168 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:43:02.159 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:43:04.625 [update notice] received unknown MS cmd: 1008b5110100 / 084c0112001f000000
2020-07-01 17:43:05.351 [update notice] received unknown MS cmd: 1038b50903298603 / 0486030080
2020-07-01 17:43:05.694 [update notice] received unknown MS cmd: 1038b50903298703 / 0487030080
**2020-07-01 17:43:08.063 [update notice] received update-read broadcast outsidetemp QQ=10: -40.000**
2020-07-01 17:43:10.258 [update notice] received read bai Status01 QQ=10: 20.5;21.0;23.250;-;-;off
2020-07-01 17:43:10.553 [update notice] received read vr_70 SensorData QQ=10: 48.31;35.94;-;19.31;37.06;28.50;c8 10 00
2020-07-01 17:43:10.824 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:43:11.303 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:43:20.309 [update notice] received read bai Status01 QQ=10: 20.5;21.0;23.438;-;-;off
2020-07-01 17:43:20.603 [update notice] received read vr_70 SensorData QQ=10: 48.25;35.94;-;19.31;37.00;28.50;c8 10 00
2020-07-01 17:43:20.878 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:43:21.137 [update notice] received read bai Status02 QQ=10: auto;60;45.0;70;55.0
2020-07-01 17:43:21.416 [update notice] received read bai DateTime QQ=10: valid;17:43:22;01.07.2020;23.438
2020-07-01 17:43:21.653 [update notice] received update-read broadcast vdatetime QQ=10: 17:43:20;01.07.2020
2020-07-01 17:43:21.916 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:43:22.184 [update notice] received unknown MS cmd: 1008b5110100 / 084c0112001f000000
2020-07-01 17:43:22.513 [update notice] received unknown MS cmd: 1038b522050002012800 / 0100
2020-07-01 17:43:23.212 [update notice] received unknown MS cmd: 1038b50903298603 / 0486030080
2020-07-01 17:43:23.554 [update notice] received unknown MS cmd: 1038b50903298703 / 0487030080
**2020-07-01 17:43:25.926 [update notice] received update-read broadcast outsidetemp QQ=10: -40.000**
2020-07-01 17:43:30.344 [update notice] received read bai Status01 QQ=10: 20.5;21.0;23.438;-;-;off
2020-07-01 17:43:30.640 [update notice] received read vr_70 SensorData QQ=10: 48.25;35.94;-;19.25;36.94;28.50;c8 10 00
2020-07-01 17:43:30.911 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:43:31.388 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:43:40.385 [update notice] received read bai Status01 QQ=10: 20.5;21.0;23.438;-;-;off
2020-07-01 17:43:40.682 [update notice] received read vr_70 SensorData QQ=10: 48.31;35.94;-;19.31;37.38;28.50;c8 10 00
2020-07-01 17:43:40.957 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:43:41.433 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:43:50.431 [update notice] received read bai Status01 QQ=10: 20.5;21.0;23.438;-;-;off
2020-07-01 17:43:50.727 [update notice] received read vr_70 SensorData QQ=10: 48.31;35.94;-;19.31;37.56;28.50;c8 10 00
2020-07-01 17:43:50.998 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:43:51.258 [update notice] received read bai Status02 QQ=10: auto;60;45.0;70;55.0
2020-07-01 17:43:51.520 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:44:00.477 [update notice] received read bai Status01 QQ=10: 20.5;21.0;23.438;-;-;off
2020-07-01 17:44:00.772 [update notice] received read vr_70 SensorData QQ=10: 48.31;36.00;-;19.31;37.81;28.50;c8 10 00
2020-07-01 17:44:01.044 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-07-01 17:44:01.479 [update notice] received write vr_70 SetActorState QQ=10: off;off;off;off;off;off;off
2020-07-01 17:44:03.028 [update notice] sent read bai FlowTemp QQ=31: 20.75;ok`

Die Fehlerhäufigkeit ist völlig unterschiedlich. Von garnicht /36h bis zu 2x/12h.

Hat noch jemand die gleichen Sorgen?
Herzlichen Dank für Eure Hilfe!
Dirk

mwildbolz, 2020-07-01:
Ich habe bei mir das selbe beobachten können, teilweise noch viel häufiger (bis zu ca. 40 Mal in 24h).
Setting ist sehr ähnlich: Esera USB EBus Koppler, RPi3 an Vaillant VWF117/4

Was mir aufgefallen ist:
Ich hatte viele Fehlermeldungen im Log beim Abfragen von TotalYield (meiner Wärmepumpe). Seit ich dieses Polling nicht mehr habe, sind auch die Außentemperatur-Fehler weniger geworden. Generell hab ich festgestellt, dass bei intensiver Buslast die Wahrscheinlichkeit höher ist, -40°C AT Werte zu bekommen. Warum das so ist, ist mir selbst nicht klar...

brezelmann01, 2020-07-01:
Totalyield macht bei mir keinen Unterschied. Ich habe in einem anderen Forum gelesen, man solle die Häufigkeit der Abfrage reduzieren, um die Buslast zu reduzieren. Ich vermute aber eher, das dann gar nicht alle Fehler erfasst werden und doch weiter bestehen. Warum betrifft es nur die Aussentemp.? Hast du auch immer unknown messages genau davor?

closed 2020-09-20 by john30

mwildbolz, 2020-10-19:
@brezelmann01 
ein später Nachtrag zu diesem Thema:
Ich hab meinen ESERA eBUS-Koppler aufgrund eines Defektes eingeschickt gehabt und eine Reparatur bekommen. Seit dem habe ich das Phänomen mit den -40°C selten bis gar nicht mehr beobachtet (mehr als 1 Woche ohne Fehler).
Hängt also anscheinend irgendwie mit der Hardware zusammen?

Vielleicht hilft dir das noch weiter...

brezelmann01, 2020-10-19:
Hi mwildbolz!

Ich musste mich jetzt wohl oder übel auch intensiv mit dem Problem beschäftigen.
Meine obige Beobachtung im Log war falsch / unwichtig.
Zwischenzeitlich habe ich meine Peripherie um ein Busnetzteil erweitert, da bei Ausfall des RPIs immer der Bus in Notlauf fiel  (Fehler Aussentemperatursensor) und die Heizung - auch im Sommer- ansprang. Auf die sporadischen Temperaturausfälle hatte das aber keine Auswirkungen.
Dann habe ich die Buslast reduziert und von 30s Abtastung auf 2min erhöht-> immer noch Fehler. 
Dann habe ich **alle** aktiven Abfragen raus genommen. Bisher dachte ich, nur die Aussentemp würde gebroadcastet. Das ist aber falsch!!

Zu meiner Überraschung sendet die VR70 alle Temperaturen rund ums Brauchwasser (Temp oben, unten, Solarpumpe, Kollektortemp) von sich aus. Auch bei der Heizung kommen einige Signale von sich aus. Seitdem ich jetzt keinen von sich aus hebroadcasteten Wert mehr aktiv Abfrage, habe ich keinerlei Ausfälle mehr. Ich steigere jetzt gerade Schritt  für Schritt wieder die Abfragen, die mir noch fehlen....

Vielleicht hilft das ja noch jemanden.

Empfehlung: ALLE AKTIVEN Abfragen mal löschen in Iobroker, gucken was dann noch kommt und dann ggf. gezielt wieder ergänzen.

brezelmann01, 2020-11-02:
Stand 02.11.2020: weiterhin Ausfälle in der Aussentemperatur, allerdings deutlich seltener. Reduziere ich das Abfrageintervall, dann sieht es zwar augenscheinlich so aus als gäbe es keine Ausfälle mehr. Das ist allerdings ein Trugschluss, wie man dann am Verlauf der SOLL- Vorlauftemperaturen und der unnötigen Brennerstarts sehen kann.

Stand aktuell: 4 aktive Abfragen alle 30s, Fehler tritt ca. alle 24h auf.

@john30 Hab nur ich das Problem, oder kannst du das bei dir auch nachvollziehen?! Mich wundert, das nicht viel mehr Nutzer das Phänomen haben.

mwildbolz, 2020-11-02:
Bei mir ist das Problem in den letzten Tagen viel seltener geworden.
Ich kann definitiv einen Zusammenhang mit der Hardware herstellen (siehe auch meinen vorigen Post) und kann die Abhängigkeit mit den Abfragen bestätigen.

Aktuell hab ich ca. 40 Abfragen laufen, mit Intervallen zwischen 2 Minuten (min) und 1x pro Stunde (max)
vieles hab ich umgestellt, so dass ich nur mehr vorhandene Telegramme auswerte und nicht extra polle.

Den letzten Ausfall (Anzeige -40°C) hatte ich vor ca. 16 Tagen

brezelmann01, 2020-11-02:
40 Abfragen? Wie viele davon aktiv? Und wie stellst du unterschiedliche Intervalle ein? Ich habe bei iobroker unter "Adapter" das Intervall festgelegt. Das scheint für alle aktiven Abfragen zu gelten.

john30, 2020-11-16:
nach wie vor ist die Latenz ein Thema´und ich schätze, dass das auch hier das Problem darstellt

brezelmann01, 2020-11-18:
Hm. Okay. Kann ich irgendwelche Einstellungen machen, die helfen?

john30, 2021-01-10:
ja, mit "--latency=XX"

john30, 2021-02-20:
die Latenz hängt ja ganz essentiell von dem Hardware Interface ab und die meisten Adapter laufen hier außerhalb der eBUS Spezifikation. Das lässt sich eigentlich nur mit Arbitrierung direkt in Hardware lösen, was erstmalig unser [neuester Adapter 3](https://adapter.ebusd.eu/) beherrscht.

pokeplayer2, 2023-01-15:
Moin, welche Einstellungen werden für den Esera USB EBus Koppler am Raspi 4 empfohlen? Ebusd bleibt bei mir immer wieder hängen und der Adapter 3 ist zwar bestellt, aber es sind keine auf Lager ...

RotorMotor42, 2023-01-21:
Ich habe das gleiche Problem, gibt es noch Dinge die wir ausprobieren können?