john30/ebusd #356: request more than one parameter (including refresh) with HTTP?

AloisKlingler, 2020-08-03:
Hi,

I have a brink device connected via ebusd (version 3.4 from debian-repository) where I fetch regularly (every 5 Minutes) a set of the by the device offered datapoints to print them in a fancy munin-graph.

regrettably the values are only cached if I try to fetch them via "/data?required" or "/data/kwl?required".
so currently I flood the bus with MQTT messages for each value before fetching the data by HTTP.

as I want to have the data now in a PHP frontend I am searching for a possibility to trigger a update for HTTP without making #count of HTTP requests in a row.

e.g. "data/kwl/LuftmengeStufe1&LuftmengeStufe2?required" does not work - is there any possibility to trigger a reload of #count parameters with one single request and get the json with updated values for further parsing?

Thanks!

Best regards
Alois

AloisKlingler, 2020-09-11:
nobody?

john30, 2020-09-20:
you could just request polling from ebusd once for each parameter you're interested in

AloisKlingler, 2021-03-01:
John,

is it possible to implement that the ?required forces a re-read from bus, for single values and also for values from the complete given circuit (so not only one parameter) or is there a technical reason behind why you did not?

Thanks. :-)

Best regards
Alois

john30, 2021-03-07:
should work by adding maxage=0
reading the whole circuit should work by leaving out the message name in the URL path. it is expensive though

closed 2021-10-06 by john30