john30/ebusd #363: ebusd CPU usage

Mathadon, 2020-09-28:
Is ebusd supposed to use 100% CPU? I ran `./ebusd --scanconfig -d /dev/tty.usbserial-AL2I7EJY --enablehex` on OSX and the process is continuously using 100% CPU. Note that there were some hacks that I had to do to get ebusd compiled, see https://github.com/john30/ebusd/issues/362.

john30, 2020-10-12:
definitely not. suppport for OSX was in there a couple of years ago but I didn't follow that any more due to lack of time. I'm pretty sure it can't work without the rt lib.

Mathadon, 2020-10-12:
Well, I definitely got it running. If you can provide some hints on how to debug the CPU usage and what is going on with rt lib, I can poke around a bit and perhaps get OSX supported properly again?

john30, 2020-10-12:
just checked my old MAC just to realize that I had it on last time three years ago... I really don't remember the details for OSX, but maybe the commit history could help here.

salvq, 2022-03-04:
I am facing also CPU usage high however it is standard x64 in docker

CPU usage is very high constantly, I am running ebusd in docker in QNAP NAS x64 and get constantly ~ 25% (1 full core from 4):
```
Mem: 7509332K used, 469772K free, 336744K shrd, 1000620K buff, 2274252K cached
CPU:  4.9% usr 24.1% sys  4.7% nic 63.8% idle  2.2% io  0.0% irq  0.0% sirq
Load average: 2.78 2.52 2.52 4/1269 17568
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
22073 22045 admin    S N   261m  3.3   0 25.3 ebusd -f --scanconfig --configpath=/ebusd/ebusd-2.1.x/en -d enh:192.168.6.98:9999 -l /var/log/ebusd.log --logareas=all --loglevel=info --httpport=8080 --accesslevel=* --mqtthost=
21992 20722 admin    S N   216m  2.7   0  0.8 container-station/python2.7 docker-monitor.pyc
 3114     1 admin    S    2012m 25.7   0  0.7 /sbin/hal_daemon -f
20070 24665 admin    S N  3217m 41.1   3  0.4 python3 -m homeassistant --config /config

[~] # docker exec -it d1b75f2d2746 uname -a
Linux NASEFC472 5.10.60-qnap #1 SMP Sat Jan 29 09:16:22 CST 2022 x86_64 GNU/Linux
[~] #

```

Also when tried in docker
```
[~] # docker exec -it 1791afda2d2e top
top - 14:37:42 up 6 days,  3:01,  0 users,  load average: 2.43, 2.47, 2.50
Tasks:   3 total,   1 running,   2 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1.2 us, 23.6 sy,  4.9 ni, 70.2 id,  0.0 wa,  0.0 hi,  0.1 si,  0.0 st
KiB Mem :  7979104 total,   460128 free,  3829084 used,  3689892 buff/cache
KiB Swap: 32826160 total, 32395096 free,   431064 used.  3297376 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
    1 root      25   5  267264   1484      0 S 100.3  0.0   2531:35 ebusd
   38 root      25   5   41192    836    380 S   0.0  0.0   0:29.65 top
   97 root      25   5   41020   3148   2692 R   0.0  0.0   0:00.01 top
```

I tried version v21.2 and also v22.2 and same result with ~100% CPU utilization / 1 core:
```
[~] # docker exec -it 1791afda2d2e ebusctl info
version: ebusd 21.2.v21.2
```

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30