john30/ebusd #364: cannot write to burner (reading is ok)

domi-a, 2020-09-29:
i am tikering for about several days and i can't get ebus daemon to work propably.

burner>ebus-adapter>esp8266>wifi>ebusd>mqtt

i can read trough mqtt from the burner. so i suggest the rest of the setting is correct.
send to mqtt "ebusd/bai/Status01/get"
in return the payload is for exampe "39.0;39.0;-;-;30.0;off" what seems to be the current boiler/burner temp. ok.

but i have no clue how to write to the burner.
i researched for hours and i am sure it should be somethong like payoad:"auto;90.0;-;-;0;0;1;0;0;0" with topic "ebusd/bai/SetMode/set"
which is doing nothing.

when i use the deamon more direct like method:
ebusctl write -c bai SetMode "auto;0.0;-;-;1;0;1;0;0;0"
the response is only "ERR: element not found"

while read returns something
ebusctl read -v -c bai SetMode
bai SetMode hcmode=auto;flowtempdesired=90.0;hwctempdesired=-;hwcflowtempdesired=-;disablehc=0;disablehwctapping=0;disablehwcload=1;remoteControlHcPump=0;releaseBackup=0;releaseCooling=0

what am i doing wrong?
is there no correct configuration for my burner (which can write)?
am i using the wrong command to write?

i read in other issues that the configuration does not support writing.
how can i check which configuration is automaticly loaded to my ebusd?

thanks for your work and help,
regards domi
___________________________________________
here some details you may ask for:

hardware:
burner: vaillant atmotec plus
roomthermostat: disconnected (don't wanna use it at all)
ebus adapter with attached esp8266(nodemcu) with ebusd-esp flashed
raspberry pi 3b (with ebusdeamon) with installed ebusd(debian way with mqtt)

ebusd-esp html UI shows:
ebusd connected: yes (inactive)
eBUS signal: acquired
eBUS initial signal: yes

ebusd:
/etc/default/ebusd only line is
EBUSD_OPTS="--scanconfig --device tcp:192.168.3.57:9999 --latency=80000 --mqttport=1883 --mqtthost=192.168.3.27 --accesslevel=*"

"ebusctl info" shows:
version: ebusd 3.4.v3.3-51-g57eae05
access: *
signal: acquired
symbol rate: 22
max symbol rate: 93
min arbitration micros: 51
max arbitration micros: 133
min symbol latency: 6
max symbol latency: 21
reconnects: 0
masters: 2
messages: 210
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

john30, 2020-10-12:
it all depends on the available messages. to find the writable ones, you should start with issuing `ebusctl f -w -c bai` and check if the message in question is in there. If so, ask for the definition by issuing `ebusctl f -f -w -c bai <foundname>`

domi-a, 2020-10-20:
thank you for your response,

"ebusctl f -w -c bai" returns the following:

```
bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
bai APCLegioProtection = no data stored
bai BlockTimeHcMax = no data stored
bai clearerrorhistory = no data stored
bai DSNOffset = no data stored
bai FanSpeedOffsetMax = no data stored
bai FanSpeedOffsetMin = no data stored
bai FlowsetHcMax = no data stored
bai FlowsetHwcMax = no data stored
bai HcPumpMode = no data stored
bai HoursTillService = no data stored
bai HwcPostrunTime = no data stored
bai HwcTempMax = no data stored
bai PartloadHcKW = no data stored
bai PartloadHwcKW = no data stored
bai ReturnRegulation = no data stored
bai SecondPumpMode = no data stored
bai SetFactoryValues = no data stored
bai SolPostHeat = no data stored
bai StorageLoadTimeMax = no data stored
bai ValveMode = no data stored
bai WarmstartOffset = no data stored
bai WPPostrunTime = no data stored
```

--> so there is no SetMode included.

I thougt I had to use the "SetMode" Message, because i found it listening to my room thermostat (which i wanna replace).
Everytime the burner fires up the Message "SetMode" was appearing on the bus (on the logging).
I checked every description from "ebusctl f -w -c bai" above and with the "ebusctl f -f -w -c bai <messagename>" but no one seems the proper message for heatig.

here is the logging of ebusd while listening 1 sec to the bus traffic:

```
2020-10-20 19:50:19.228 [update notice] received read bai Status01 QQ=10: 29.0;28.0;-;-;26.0;off
2020-10-20 19:50:21.220 [update notice] received read bai Status02 QQ=10: auto;60;50.0;70;41.0
2020-10-20 19:50:23.276 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-10-20 19:50:29.323 [update notice] received read bai Status01 QQ=10: 29.0;27.0;-;-;26.0;off
2020-10-20 19:50:31.334 [update notice] received read bai DateTime QQ=10: nosignal;18:18:38;-.-.-;-
2020-10-20 19:50:33.348 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-10-20 19:50:39.419 [update notice] received read bai Status01 QQ=10: 29.0;27.0;-;-;26.0;off
2020-10-20 19:50:43.442 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-10-20 19:50:49.446 [update notice] received read bai Status01 QQ=10: 29.0;27.0;-;-;26.0;off
2020-10-20 19:50:51.362 [update notice] received update-read broadcast vdatetime QQ=10: 19:51:29;20.10.2020
2020-10-20 19:50:51.615 [update notice] received unknown MS cmd: 1008b512020000 / 00
2020-10-20 19:50:53.497 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-10-20 19:50:59.486 [update notice] received read bai Status01 QQ=10: 29.0;27.0;-;-;26.0;off
2020-10-20 19:51:01.505 [update notice] received read bai DateTime QQ=10: nosignal;18:19:10;-.-.-;-
2020-10-20 19:51:01.874 [update notice] received read bai Status02 QQ=10: auto;60;50.0;70;41.0
2020-10-20 19:51:03.497 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
2020-10-20 19:51:09.498 [update notice] received read bai Status01 QQ=10: 29.0;27.0;-;-;26.0;off
2020-10-20 19:51:13.600 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
```

the room thermostat is sending a write command with SetMode. or am i iterpreting it wrong?

the burner always starts the flame and heats after following log line appeared:
[update notice] received update-write bai SetMode QQ=10: auto;90.0;-;-;0;0;1;0;0;0

so i have to send SetMode as well. isn't it so?

domi-a, 2020-11-11:
could anyone help and tell me where to read or research to make it work.
it fells like there is not much work needed to get it work.
basically i just want to turn the heating on or off on my: vaillant atmotec plus

john30, 2020-11-16:
the SetMode message is defined as update-write, i.e. as-is is not writeable by the user of ebusd. reason for that is, that your controller is sending it all the time as is revealed in your log.
so if you start sending it as well, you would have to remove the controller from the bus, which is usually not desired.

domi-a, 2020-11-26:
i want to replace my controller. i just mentioned it because i thought i have to replicate it.

how can i find out wich is the right command/message to control the burner/ or target temp/ or whatever let's me control the flame?

andig, 2020-11-26:
You have SetMode, that should do.

domi-a, 2020-11-28:
yes, except it does not!

as i mentioned in my initial post: 
> ebusctl write -c bai SetMode "auto;0.0;-;-;1;0;1;0;0;0"
> the response is only "ERR: element not found"

and @john30 mentioned: setMode ist not writeable by the user of ebusd.

i searched many hours on websites and forums and cant find the information i'm looking for.
where is the place to find what command is correct for starting my burner(vaillant atmotec plus)?

domi-a, 2020-12-07:
so @andig SetMode seems to be readonly.
since `ebusctl f -w -c bai` does not return SetMode
is it possible to use local Configuration with --configpath and manipulate the file i a way that i can write to SetMode?
and if, in what way?

sahblood, 2021-01-07:
Having the same problem with my vaillant ecotec plus, have you found a solution @domi-a ?

domi-a, 2021-01-07:
hi @sahblood.
no, unfortunately not.
my complete setup is 99% complete. just need this to work.
till now i havent found the motivation to go on with it. the messages that are interpreted for each system is pretty much a blackbox. from projekt https://github.com/john30/ebusd-configuration
the only idea is to tinker with raw data. get it from the reading and send it raw.

john30, 2021-01-10:
guys, it really does not make sense at all to change the SetMode to writable as long as a controller is in the system that frequently uses the message to talk to the heater.
so, of course you can change the entry in the CSV to be writable, but then you'd also have to disconnect your controller from the bus.
I'm pretty sure that's not what you want

domi-a, 2021-01-13:
no, that seems to be exactly what i want.
i'm afraid whe have a missunderstanding.
i think i used some topic specific words the wrong way. 
so i try to condense the important part and write in in some other words:

i do not want to use my vaillant controller!! (thing in my livingroom which controls target and current temp and timers ctc.)
i want to replace the damn thing with the device i built: an **ebusd-esp** with an **ebus adapter** and the **ebus-deamon** running on my server.
i want to control when the burner has to heat and when it has to stop.(yes, i know is is controlled by the target temp and not burn enable/disable)

which i thought are in my case (i think they are from reading the log which contains data from the burner and the controller - which i don't wanna use):
```
ebusctl write -c bai SetMode "auto;0.0;-;-;1;0;1;0;0;0"
ebusctl write -c bai SetMode "auto;90.0;-;-;1;0;1;0;0;0"
```
i do not want to do anything else.

as you wrote in https://github.com/john30/ebusd/issues/317
- i cloned the config to /etc/config
- edited /etc/config/vaillant/hcmode.inc line 10 from 'uw ...' to 'w ...'
- disconnected the vaillant controller
- restarted ebusd (startet with ebusd -f --scanconfig --accesslevel=* -d tcp:192.168.3.57:9999 --latency=80000 --mqttport=1883 --mqtthost=192.168.3.27)
- but cant write to SetBai
`ebusctl write -c bai SetMode "auto;0.0;-;-;1;0;1;0;0;0"` still returns 'ERR: element not found'

`ebusctl f -w -c bai` does not return SetMode

btw: starting ebus is outputting an error:
`[main error] error executing instructions: ERR: missing argument, last error: loaded "vaillant/bai.308523.inc" for "08" ([HW=7401]), error including "vaillant/hcmode.inc" for "08": ERR: missing argument vaillant/hcmode.inc:10: ERR: missing argument, circuit, included "vaillant/errors.inc" for "08"`

i'm not sure how to interpret this

john30, 2021-01-24:
> error executing instructions: ERR: missing argument

this is certainly the problem that needs to be fixed first. You need to add a line before the hcmode line you changed that includes the write default. just copy it from one of the first lines in the vaillant/bai.308523.inc file:
`*w,,,,,,"B509","0E",,,,,,`

domi-a, 2021-01-28:
ja. geht. danke dafür. blinkt auch rot am adapter.
aber hat überhaupt keine auswirkung an der therme. weder heizen stoppen noch starten - nix.
setmode taucht mit den ganzen modifikationen auch im log gar nicht mehr auf.

ich hab keinen bock mehr. hab da jetzt sicher an die 200 stunden arbeit reingesteckt. für genau 0 erfolg. ich werde nicht nochmal 200h reinsteken. ich bin raus.

closed 2021-01-28 by domi-a