john30/ebusd #365: = no data stored ((

zamdr, 2020-10-12:
Hello. Your app is super! But apparently my boiler is not supported, because the state of the sensors is not detected, it outputs "= no data stored". Tell me what I can do to get data from my boiler. All he sees is a bus scan.08  = Vaillant;BAI00;0304;4702
scan.08 id = 21;15;09;0010007995;3100;008143;N5

john30, 2020-10-12:
the product code is not yet known in the [bai.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/08.bai.csv). I would give it a try by adding the product code '0010007995' to line 14 of that file. In parallel I'd check PDFs for one of the other products in that line to check if the burner is at least somewhat similar to the one you own.

zamdr, 2020-10-14:
> the product code is not yet known in the [bai.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/08.bai.csv). I would give it a try by adding the product code [...]

zamdr, 2020-10-14:
Thanks. How do I do this?

john30, 2020-10-14:
clone the ebusd-configuration repo, point ebusd to the 2.1.x/de or /en directory using the "-c" option, and add the product code to line 14 with the other product codes before starting ebusd

john30, 2021-10-06:
closed due to inactivity, wrong area anyway, use https://github.com/john30/ebusd-configuration/ instead