john30/ebusd #367: Help needed for Synology Docker to run ebusd

tanker11, 2020-10-20:
Hello,

I have a Wemos running ebus adapter 2.0 connected via an ebus interface to my heater.
I googled a lot of commands how to run the daemon on Synology, and once it is run the wifi adapter says: ebus connected: yes.
But I could not figure out, what configuration folders and files to be placed and where on Synology NAS in order that ebusd finds them and start reading out. I got "invalid configpath" or "error reading config files" error messages.
Could I receive a little bit of help how/where to place the files on Synology and refer to them?
Also what exactly I should place there from the folder (English version)?

Thanks in advance.

john30, 2020-11-16:
you need to start ebusd with the right options. if you plan to use the webservice on ebusd.eu for getting the config files, you need to add "--scanconfig" to the ebusd start parameters.

pepsonEL, 2021-12-17:
Hi @tanker11 
Can you write step by step how run daemon ebusd on Synology? Thanks

tanker11, 2022-01-08:
Hello,

Sorry I gave up, and took no notes on this, so now I cannot help. :(

pepsonEL ***@***.***> ezt írta (időpont: 2021. dec. 17., P,
21:03):

pepsonEL, 2022-01-08:
Ok no problem. I run it and works ok. Thanks