john30/ebusd #368: Error while trying to run on Raspberry

Wo0Zzz, 2020-10-20:
Hello,
I'm trying to run the software on a Raspberry Pi 3B+, installed a fresh version of Raspberry OS (based on Debian).
Followed Step 1 of the installation, but when i get to starting the programm, I get an error. 
See the attached screenshot.
[image]

I don't know what i'm doing wrong, but can anyone help me? 
Thanks, 
Jelle

salvq, 2020-11-04:
Hi, copy config files to local folders and set it to config path
```
/home/pi/ebusd-configuration/ebusd-2.1.x/en
pi@rpi3-ebus:~/ebusd-configuration/ebusd-2.1.x/en $ ls -la
total 24
drwxr-xr-x 3 pi pi 4096 Oct 24 00:21 .
drwxr-xr-x 4 pi pi 4096 Oct 24 00:21 ..
-rw-r--r-- 1 pi pi 1255 Oct 24 00:21 broadcast.csv
-rw-r--r-- 1 pi pi  744 Oct 24 00:21 memory.csv
-rw-r--r-- 1 pi pi 1229 Oct 24 00:21 _templates.csv
drwxr-xr-x 2 pi pi 4096 Nov  4 23:22 vaillant
pi@rpi3-ebus:~/ebusd-configuration/ebusd-2.1.x/en $

```

Verify your config first, here is my own (just need to adjust broker ip address, userid and password).

```
pi@rpi3-ebus:~/ebusd-configuration/ebusd-2.1.x/en/vaillant $ sudo nano /etc/default/ebusd
......
EBUSD_OPTS="-d /dev/ttyebus -p 8888 -l /var/log/ebusd.log --httpport=8080 --accesslevel=* --scanconfig=full --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en --mqtthost=192.168.X.XXX --mqttport=1883 --mqttuser=LOGIN --mqttpass=PASS --mqttjson --mqtttopic=ebusd/%circuit/%name"
.......

```

john30, 2020-11-16:
you need to configure your path the configuration files using "-c" parameter of ebusd