john30/ebusd #371: Vaillant VRC720f  / BASV0 config

xxxturboxxx, 2020-10-28:
I was using a Vaillant VRC700 controller on my heating system and I was able to log all the ebus data. 2 weeks ago I changed the controller to a VRC720f which sends the data to its radio receiver which is connecet to the ebus. 

The 15.700.csv did not work obviously and ebusd could not find a suitable config. Therefor I adapted the 15.700.csv and renamed it to 15.basv0.csv which worked fine for a while. But now I got some errors (see log). 

What could be the reason for that an ho can I solve that

2020-10-16 11:00:35.563 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with full scan
2020-10-16 11:00:35.575 [bus notice] bus started with own address 31/36
2020-10-16 11:00:35.589 [bus notice] signal acquired
2020-10-16 11:00:38.354 [bus notice] new master 10, master count 2
2020-10-16 11:00:38.381 [update notice] received unknown MS cmd: 1052b5230402010137 / 0201f1
2020-10-16 11:00:42.203 [bus notice] new master 03, master count 3
2020-10-16 11:00:45.577 [main notice] starting initial full scan
2020-10-16 11:00:49.944 [bus notice] scan 08: ;Vaillant;HMU01;0305;8802
2020-10-16 11:00:49.944 [update notice] store 08 ident: done
2020-10-16 11:00:49.944 [update notice] sent scan-read scan.08  QQ=31: Vaillant;HMU01;0305;8802
2020-10-16 11:00:49.944 [bus notice] scan 08: ;Vaillant;HMU01;0305;8802
2020-10-16 11:00:55.581 [main notice] read common config file vaillant/scan.csv
2020-10-16 11:00:55.581 [main notice] read common config file vaillant/broadcast.csv
2020-10-16 11:00:55.581 [main notice] read common config file vaillant/general.csv
2020-10-16 11:00:55.584 [main notice] read scan config file vaillant/08.hmu.csv for ID "hmu01", SW0305, HW8802
2020-10-16 11:00:55.584 [main error] error executing instructions: ERR: element not found, last error: error including "vaillant/hcmode.inc" for "08.hmu": ERR: element not found /etc/ebusd/ebusd-2.1.CK/de/vaillant/hcmode.inc, error including "vaillant/errors.inc" for "08.hmu": ERR: element not found /etc/ebusd/ebusd-2.1.CK/de/vaillant/errors.inc
2020-10-16 11:00:55.584 [main notice] found messages: 50 (0 conditional on 0 conditions, 0 poll, 8 update)
2020-10-16 11:00:59.434 [main error] scan config 15: ERR: wrong symbol received
2020-10-16 11:01:02.007 [main error] scan config 52: ERR: wrong symbol received
2020-10-16 11:01:09.941 [update notice] received unknown MS cmd: 1008b507020937 / 02ad01
2020-10-16 11:01:17.240 [main error] scan config 15: ERR: arbitration lost
2020-10-16 11:01:21.350 [main error] scan config 52: ERR: wrong symbol received
2020-10-16 11:01:28.633 [update notice] received unknown MS cmd: 1052b5230402010137 / 0201f6
2020-10-16 11:01:34.418 [main error] scan config 15: ERR: wrong symbol received
2020-10-16 11:01:36.579 [update notice] received unknown MS cmd: 1008b5100900003affffff060000 / 0101
2020-10-16 11:01:37.546 [main error] scan config 52: ERR: CRC error
2020-10-16 11:01:48.589 [bus notice] new master 71, master count 4
2020-10-16 11:01:50.409 [main error] scan config 15: ERR: CRC error
2020-10-16 11:01:52.409 [update notice] received unknown MS cmd: 1008b5100900003affffff060000 / 0101
2020-10-16 11:01:56.422 [main error] scan config 52: ERR: wrong symbol received
2020-10-16 11:02:00.267 [main error] scan config 76: ERR: wrong symbol received
2020-10-16 11:02:05.884 [update notice] received unknown MS cmd: 1008b5110101 / 0939390080ffff0000ff
2020-10-16 11:02:11.371 [update notice] received unknown MS cmd: 1008b5110100 / 09c801ff000208008000
2020-10-16 11:02:12.939 [update notice] received update-read broadcast outsidetemp QQ=10: 5.824
2020-10-16 11:02:14.364 [main error] scan config 15: ERR: wrong symbol received
2020-10-16 11:02:17.003 [main error] scan config 52: ERR: wrong symbol received
2020-10-16 11:02:19.605 [bus notice] new master 00, master count 5
2020-10-16 11:02:19.649 [update notice] received unknown MS cmd: 0015b52406020002011e00 / 0601021e000100
2020-10-16 11:02:20.167 [main notice] SIGTERM received
2020-10-16 11:02:22.362 [main error] scan config 76: ERR: CRC error
2020-10-16 11:02:22.472 [main notice] ebusd stopped
2020-10-16 11:12:43.248 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with full scan
2020-10-16 11:12:43.260 [bus notice] bus started with own address 31/36
2020-10-16 11:12:43.271 [bus notice] signal acquired
2020-10-16 11:12:47.448 [bus notice] new master 10, master count 2
2020-10-16 11:12:47.507 [bus notice] new master 71, master count 3
2020-10-16 11:12:47.507 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff5d000033
2020-10-16 11:12:48.853 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2020-10-16 11:12:53.260 [main notice] starting initial full scan
2020-10-16 11:13:05.230 [main error] scan config 15: ERR: CRC error
2020-10-16 11:13:11.079 [main error] scan config 52: ERR: arbitration lost
2020-10-16 11:13:13.891 [main error] scan config 76: ERR: CRC error
2020-10-16 11:13:15.121 [bus notice] new master 03, master count 4
2020-10-16 11:13:15.193 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 0b0000000001ffffffffff00
2020-10-16 11:13:16.967 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2020-10-16 11:13:17.133 [bus notice] scan e8: ;Vaillant;FMU00;0202;6502
2020-10-16 11:13:17.133 [update notice] store e8 ident: done
2020-10-16 11:13:17.133 [update notice] sent scan-read scan.e8  QQ=31: Vaillant;FMU00;0202;6502
2020-10-16 11:13:17.133 [bus notice] scan e8: ;Vaillant;FMU00;0202;6502
2020-10-16 11:13:17.133 [main error] unable to load scan config e8: no file from vaillant with prefix e8 found
2020-10-16 11:13:17.134 [main error] scan config e8: ERR: element not found
2020-10-16 11:13:23.277 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff5d000033
2020-10-16 11:13:24.154 [update notice] received unknown MS cmd: 1076b512030f0001 / 07eb02009c011403
2020-10-16 11:13:24.397 [update notice] received unknown BC cmd: 10feb5160301d305
[... 855 lines cut ...]
2020-10-28 12:53:57.169 [main error] scan config 52: ERR: wrong symbol received
2020-10-28 12:53:58.287 [update notice] received unknown MS cmd: 0015b52406020003000500 / 080203050000007041
2020-10-28 12:53:59.450 [main error] scan config 76: ERR: CRC error
2020-10-28 12:54:01.451 [update notice] received unknown MS cmd: 1076b512030f0001 / 07ff020090011403
2020-10-28 12:54:01.681 [update notice] received unknown MS cmd: 0015b52406020003001100 / 050203110001
2020-10-28 12:54:01.982 [update notice] received unknown MS cmd: 0015b52406020003001c00 / 0500031c00ff
2020-10-28 12:54:02.036 [main error] scan config e8: ERR: wrong symbol received
2020-10-28 12:54:02.674 [update notice] received unknown MS cmd: 0015b52406020003002700 / 050003270000
2020-10-28 12:54:02.770 [update notice] received update-write hmu SetMode QQ=10: auto;27.5;-;-;0;1;1;0;0;0
2020-10-28 12:54:02.962 [update notice] received unknown MS cmd: 0015b52406020003000f00 / 0800030f00ffffff7f
2020-10-28 12:54:05.600 [update notice] received unknown MS cmd: 0015b52406020002001e00 / 0600021e000000
2020-10-28 12:54:06.438 [update notice] received unknown MS cmd: 0015b52406020002000600 / 050202060000
2020-10-28 12:54:10.186 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff60000032
2020-10-28 12:54:11.650 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2020-10-28 12:54:14.160 [update notice] sent scan-read scan.15 id QQ=31: 
2020-10-28 12:54:14.673 [main error] scan config 15: ERR: wrong symbol received
2020-10-28 12:54:16.810 [main error] scan config 52: ERR: wrong symbol received
2020-10-28 12:54:18.914 [main error] scan config 76: ERR: wrong symbol received
2020-10-28 12:54:22.404 [update notice] sent scan-read scan.e8 id QQ=31: 
2020-10-28 12:54:22.664 [update notice] sent scan-read scan.e8 id QQ=31: 
2020-10-28 12:54:22.738 [main error] scan config e8: ERR: wrong symbol received
2020-10-28 12:54:33.367 [update notice] received unknown MS cmd: 1052b5230402010133 / 020100
2020-10-28 12:54:33.632 [update notice] received unknown MS cmd: 1008b5110100 / 098a01ff000308008000
2020-10-28 12:54:34.354 [main error] scan config 15: ERR: wrong symbol received
2020-10-28 12:54:37.111 [main error] scan config 52: ERR: wrong symbol received

xxxturboxxx, 2020-11-28:
Nachdem mir leider niemand helfen konnte/wollte bin ich der Sache selbst auf den Grund gegangen. 
Meine Vermutung ist, dass durch den Wechsel des Reglers dies in irgendeiner Form die Einstellung des Pottis am Esera-Koppler beeinflusst hat. Nach erneutem einstellen des Pottis funktioniert jetzt wieder alles wie gehabt.

Man könnte nun die 15.basv0.csv mit in die Online-config-Datenbank mit übernehmen. Diese ist wie gesagt identisch zur 15.700.csv und müsste nur umbenannt werden.

closed 2020-11-28 by xxxturboxxx

john30, 2021-01-10:
@xxxturboxxx bitte ein ticket unter ebusd-configuration öffnen